zephyrproject-rtos/zephyr #67259: imx_usdhc: SD-Card mount on MIMXRT1170 Board fails

skumg, 2024-01-05:
**Describe the bug**
The SD-Card mounting process often fails on the NXP MIMXRT1170 Eval Board. The board is unable to switch to higher frequencies, and hangs with an error `Card never left busy state (-5)`.

 - What target platform are you using? 
   mimxrt1170_evk_cm7 (NXP MIMXRT1170 Eval Board)

 - What have you tried to diagnose or workaround this issue? 
   I have tried to 

 - Is this a regression? If yes, have you been able to "git bisect" it to a specific commit?
   A similar issue had been reported (https://github.com/zephyrproject-rtos/zephyr/issues/39942), which suggests `no-1-8-v` as a workaround. This, however, reduces the read write speeds. I have also tried successive mounts, which works around 75% of the time. Also tried to increase the delay 

**To Reproduce**
Steps to reproduce the behavior:

```
static FATFS fs;
static struct fs_mount_t mount = {
	.type = FS_FATFS,
	.fs_data = &fs,
	.mnt_point = "/SD:",
};

int main(void)
{
	int res = fs_mount(&mount));
	if(res) {
		LOG_ERR("Mounting file system failed, Error %d", -EIO);
	}

	return res;
}
```

Following project variables are enabled: 
```
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_FS_FATFS_EXFAT=y
CONFIG_FS_LOG_LEVEL_INF=y
CONFIG_FS_FATFS_MOUNT_MKFS=n
CONFIG_SD_LOG_LEVEL_INF=y
CONFIG_SDHC_LOG_LEVEL_INF=y
CONFIG_SDMMC_LOG_LEVEL_INF=y
```

**Expected behavior**
The SD-Card mount should be successful, the following logs: 
```
00:00:08.767,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power OFF, voltage 3.3V
[00:00:09.267,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:00:09.788,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 400000Hz, card power ON, voltage 3.3V
[00:00:09.788,000] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 386718
[00:00:09.987,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:00:09.987,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 1.8V
[00:00:10.017,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 400000Hz, card power ON, voltage 1.8V
[00:00:10.017,000] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 386718
[00:00:10.018,000] <inf> sd: Card switched to 1.8V signaling
[00:00:10.021,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 25000000Hz, card power ON, voltage 1.8V
[00:00:10.021,000] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 24750000
[00:00:10.022,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 4, clock 25000000Hz, card power ON, voltage 1.8V
[00:00:10.115,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 4, clock 208000000Hz, card power ON, voltage 1.8V
[00:00:10.115,000] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 198000000
[00:00:10.196,000] <dbg> fs: fs_mount: fs mounted at /SD:
```

**Actual behavior**
The SD-Card mount fails, with the following logs: 
```
[00:00:07.596,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power OFF, voltage 3.3V
[00:00:08.096,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:00:08.617,000] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 400000Hz, card power ON, voltage 3.3V
[00:00:08.617,000] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 386718
[00:00:19.129,000] <err> sd: Card never left busy state
[00:00:19.129,000] <err> sd: Failed to query card OCR
[00:00:19.129,000] <err> fs: fs mount error (-5)
```

**Impact**
Prototype: Annoyance
Series: Showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK with West
 - Version used: 3.5.99

DerekSnell, 2024-01-08:
Hi @skumg ,
Thank you for reporting this.  Since there are two revisions of the RT1170 EVK, can you please confirm if the part number on your board includes `EVKB`?  I see you are building for `mimxrt1170_evk_cm7`, but I just want to confirm if you are using the original EVK or the latest EVKB.  I am not sure yet if that matters for your issue, but I will try to replicate on the same hardware.
Thanks

skumg, 2024-01-08:
> Hi @skumg , Thank you for reporting this. Since there are two revisions of the RT1170 EVK, can you please confirm if the part number on your board includes `EVKB`? I see you are building for [...]

Hi Derek, I have the MIMXRT1170-EVK (first gen) board, not the EVKB (with SCH-32171 REV C4 / 700-32171 REV A1, in case that helps). Also using the default configuration, and have tested this with 4GB and 16 GB cards (both Class 10).

DerekSnell, 2024-01-10:
Hi @skumg ,
I have not been able to replicate this with the MIMXRT1170-EVK and the few uSD cards I have.  However, we are looking into the error in your log and what may cause it.  Best regards

skumg, 2024-01-15:
> Hi @skumg , I have not been able to replicate this with the MIMXRT1170-EVK and the few uSD cards I have. However, we are looking into the error in your log and what may cause it. Best regards

Hi @DerekSnell, thank you for the update and sorry for the delay in my response. Is it possible to also check if the mount works after the reset button (SW4) has been pressed? I have often noticed that to be a factor. If I can test it further and have updates from my end, I shall let you know. Thank you once again!

DerekSnell, 2024-01-16:
Hi @skumg ,
Yes, I reset the EVK many times in my testing, and I did not see this issue mounting the SD cards I tested.

Also, I do not think https://github.com/zephyrproject-rtos/zephyr/issues/38575 applies to this board.  First, in my testing with the RT1170 EVK, I saw from the log that the SDHC driver switched to 1.8 V and was able to mount those cards.  

Second, https://github.com/zephyrproject-rtos/zephyr/issues/38575 is specific to how the voltage level is changed on the other MIMXRT685-EVK.  On that board, the voltage level for the SDHC pins is controlled by the PCA9420 PMIC regulator, specifically the `LDO2_OUT` supply rail.  So https://github.com/zephyrproject-rtos/zephyr/issues/38575 states the SDHC driver needs to request the PMIC regulator driver to change the voltage of that supply.

But on the MIMXRT1170-EVK board, the MCU signal `USDHC1_VSELECT` is used to select this voltage, which is controlled by the uSDHC peripheral.  No regulator driver required.  `USDHC1_VSELECT` is connected to the board signal `SD1_VSELECT`, which controls U36 to supply the selected voltage level.  So https://github.com/zephyrproject-rtos/zephyr/issues/38575 does not apply to this board.

Best regards

skumg, 2024-01-17:
Hi @DerekSnell Thank you for the clarification, I was confused about the PMIC on that board. 

One more important thing that I missed, I have been using the card with `exFAT` file formatting (formatted on Windows), with allocation units 32 Kilobytes large.

danieldegrasse, 2024-01-17:
Hi @skumg
I've tested the `samples/subsys/fs/fs_sample` build with `-DCONFIG_FS_FATFS_EXFAT=y` on the RT1170 EVK (rev A), with the following SD cards (all formatted at exFAT):
- Kingston 2GB SD (SD-C02G, 80X speed)
- Microcenter 16 GB SDHC (Speed Class 10)
- Kingston 16 GB SDHC (Speed Class 10, UHS Speed Grade 1)
- Microcenter 32 GB SDHC (Speed Class 10, UHS Speed Grade 1)
- Sandisk Ultra 64 GB SDXC (Speed Class 10, UHS Speed Grade 1)
- Sandisk Ultra 256 GB SDXC (Speed Class 10, Video Speed Class 10)

I was not able to reproduce this issue on any card. Are you able to reproduce this on multiple cards? A few things to check/try:
- Does increasing CONFIG_SD_OCR_RETRY_COUNT affect anything?
- Will the cards initialize if `no-1-8-v` is set?

skumg, 2024-01-18:
Hi @DerekSnell,
The sample runs through the `disk_access` API initially, and hence seems to somehow work fine. As soon as I am trying to run it without the `disk_access` API, the issue occurs. I have access to only two cards, but we are in the process of ordering some more (I have an Xmore industrial 4GB, and a Sandisk Ultra 16GB). 

Other points: 
1) `CONFIG_SD_OCR_RETRY_COUNT` haven't tried this yet, I will try it out. I do not have the board with me currently, and will be able to test it next week (mostly along with new [micro] SD-Cards)
2) The cards do initialize with `no-1-8-v` set, but as mentioned, the speed of transfer is then reduced. 

Thanks and best.

danieldegrasse, 2024-01-18:
> The cards do initialize with no-1-8-v set, but as mentioned, the speed of transfer is then reduced.

This is very strange to me. The OCR query step occurs before the SD stack switches the card to 1.8V I/O. Could you try locally commenting out this line: https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/sd/sdmmc.c#L583, and see if the card will initialize?

DerekSnell, 2024-02-09:
Closing until we receive more details

skumg, 2024-02-21:
Hello guys, 

Sorry to have kept you waiting, I could not get back to this issue till today. I got a new board, new SD-cards, and I have narrowed down the problem a bit further. The problem occurs as soon as I activate the two Kconfig params below (I tried this on the fs-sample; sorry for the previous chaos!): 

```
CONFIG_NETWORKING=y
CONFIG_NET_CONFIG_SETTINGS=y
```

Without the raw disk IO operations, the following error is seen: 

```
*** Booting Zephyr OS build zephyr-v3.4.0-10921-g592a7cd6a750 ***
[00:00:01.504,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:12.959,000] <err> sd: Card never left busy state
[00:00:12.959,000] <err> sd: Failed to query card OCR
[00:00:12.959,000] <err> fs: fs mount error (-5)
Error mounting disk.
[00:00:12.959,000] <err> fs: fs not mounted (mp == 0x80001260)
```

With raw disk IO operations, the disk is mounted, but the following error is seen: 

```
*** Booting Zephyr OS build zephyr-v3.4.0-10921-g592a7cd6a750 ***
[00:00:01.503,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:12.910,000] <err> sd: Card never left busy state
[00:00:12.910,000] <err> sd: Failed to query card OCR
[00:00:12.910,000] <err> main: Storage init ERROR!
[00:00:13.989,000] <inf> sd: Card switched to 1.8V signaling
Disk mounted.
```

---

One more thing, as soon as I enable `CONFIG_SPEED_OPTIMIZATIONS=y`, the following array-out-of-bounds warning can be seen in the `fsl_usdhc.c` (although this is / might be a separate issue): 

```
[244/294] Building C object modules/hal_nxp/hal_nxp/CMakeFiles/..__modules__hal__nxp.dir/mcux/mcux-sdk/drivers/usdhc/fsl_usdhc.c.obj
/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/usdhc/fsl_usdhc.c: In function 'USDHC_TransferCreateHandle':
/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/usdhc/fsl_usdhc.c:2401:18: warning: array subscript 3 is above array bounds of 'usdhc_handle_t *[3]' {aka 'struct _usdhc_handle *[3]'} [-Warray-bounds]
 2401 |     s_usdhcHandle[USDHC_GetInstance(base)] = handle;
      |     ~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/usdhc/fsl_usdhc.c:209:24: note: while referencing 's_usdhcHandle'
  209 | static usdhc_handle_t *s_usdhcHandle[ARRAY_SIZE(s_usdhcBase)] = {0};
      |                        ^~~~~~~~~~~~~

```

DerekSnell, 2024-02-26:
Hi @skumg ,
Thank you for sharing these details.  It seems there is some interaction during initialization causing the issue mounting the SD card.  I am glad you found the `CONFIG_NET_CONFIG_AUTO_INIT=n` workaround.

Best regards

dleach02, 2024-06-07:
@danieldegrasse is the AUTO_INIT workaround enough or should we continue to debug more?

skumg, 2024-06-10:
> @danieldegrasse is the AUTO_INIT workaround enough or should we continue to debug more?

The following pull request had actually solved the issue for us: https://github.com/zephyrproject-rtos/zephyr/pull/70311. I shall close this issue as resolved.

closed 2024-06-10 by skumg

dleach02, 2024-06-10:
Thank you @skumg