zephyrproject-rtos/zephyr #67263: The lvgl sample has screen glitching

rivendell1984, 2024-01-06:
**Describe the bug**
I'm following this page:
https://docs.zephyrproject.org/latest/boards/shields/waveshare_epaper/doc/index.html
to configure waveshare_epaper_gdeh029a1 & run the lvgl sample. 
I found when I invoke the lv_label_set_text(count_label, count_str) in the while loop, there is screen glitching on the top of the screen every two times, and the "Hello World" text will have position offset.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. copy lvgl sample from: https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/display/lvgl
2. add board nrf52840dk: https://github.com/zephyrproject-rtos/zephyr/tree/main/boards/arm/nrf52840dk_nrf52840
3. add set(SHIELD waveshare_epaper_gdeh029a1) in CMakeLists.txt, which will add devicetree overlay: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/shields/waveshare_epaper/waveshare_epaper_gdeh029a1.overlay
4. make
5. Flash to the nrf52840dk
6. Insert the WAVESHARE e-Paper Raw Panel Shield to the nrf52840dk, and epaper_gdeh029a1 to the shield
7. The epaper_gdeh029a1 will show glitch and the "Hello World" text will have position offset.

**Expected behavior**
The text stay in the same position, and there is no glitch

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain ncs 2.5.1, Zephyr version: 3.4.99

faxe1008, 2024-01-09:
I looks like there is an issue due to partial refresh. Can you maybe try the following:
Set `CONFIG_LV_Z_VBD_SIZE=100` and `CONFIG_LV_Z_FULL_REFRESH=y` to force a full update of the screen. If this drastically increases the refresh time you might also need to change this line `(count % 100)` to something bigger, so the screen is not invalidated as often.

rivendell1984, 2024-01-11:
Thanks for your comment. I'm using nordic connect sdk v2.5.1 with zephyr 3.4.99. 
For the "CONFIG_LV_Z_VBD_SIZE=100", I got this error:
_warning: attempt to assign the value '100' to the undefined symbol LV_Z_VBD_SIZE
error: Aborting due to Kconfig warnings_

And after adding "CONFIG_LV_Z_FULL_REFRESH=y", there is the same issue.

nashif, 2024-03-12:
@faxe1008  can you please followup?

faxe1008, 2024-03-12:
@rivendell1984 Sorry for the missing feedback on the issue. The Kconfig option was not introduced for that ncs version apparently. The issue is that I don't have a EPD with that controller currently with me.

I did some digging and I found this issue: [https://github.com/lvgl/lvgl/issues/1427](https://github.com/lvgl/lvgl/issues/1427)
And this one:
[https://github.com/lvgl/lvgl/issues/2974](https://github.com/lvgl/lvgl/issues/2974)

Does this controller have requirements for the the coordinates you can write to maybe? Can you please try to patch the rounder callback for the mono display like this:

```
--- a/modules/lvgl/lvgl_display_mono.c
+++ b/modules/lvgl/lvgl_display_mono.c
@@ -94,16 +94,8 @@ void lvgl_rounder_cb_mono(lv_disp_drv_t *disp_drv, lv_area_t *area)
 {
        struct lvgl_disp_data *data = (struct lvgl_disp_data *)disp_drv->user_data;

-       if (data->cap.screen_info & SCREEN_INFO_X_ALIGNMENT_WIDTH) {
-               area->x1 = 0;
-               area->x2 = data->cap.x_resolution - 1;
-       } else {
-               if (data->cap.screen_info & SCREEN_INFO_MONO_VTILED) {
-                       area->y1 &= ~0x7;
-                       area->y2 |= 0x7;
-               } else {
-                       area->x1 &= ~0x7;
-                       area->x2 |= 0x7;
-               }
-       }
+       area->x1 = 0;
+       area->x2 = data->cap.x_resolution - 1;
+       area->y1 = 0;
+       area->y2 = data->cap.y_resolution - 1;
 ```
 This should now force a full refresh.

closed 2024-05-27 by github-actions[bot]