zephyrproject-rtos/zephyr #67268: File linker-defs.h does not have a path to file offsets.h.

paultino, 2024-01-06:
**Describe the bug**

When I tried to build an application using the T2 topology, I received an error. 
When I specified the full path to the offsets.h file, the application was built successfully.

**To Reproduce**
4. See error
[image]

**Expected behavior**
specify the path as indicated in the lines above in the linker-defs.h file.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK 0.16.4
 - Zephyr version: 3.5.99, build: zephyr-v3.5.0-3682-gd7af6f371034

MaureenHelm, 2024-01-09:
Will you please provide more details about how to reproduce?

hellesvik-nordic, 2024-01-12:
Several people reporting the same error over at DevZone: https://devzone.nordicsemi.com/f/nordic-q-a/106046/offsets-h-no-such-file-or-directory. I have not yet been able to figure out what causes this.
A mentioned workaround is to tell west to use make instead of ninja.
Also some people mention that the error is apparent in the first build, but that it works again on a subsequent builds.

hellesvik-nordic, 2024-01-12:
Update: One of the users in the devzone ticket fixed this by including "#include <zephyr/drivers/i2c.h>" into the .cpp file instead of header file. Other users confirm that this fix worked.

paultino, 2024-01-13:
I got the same thing yesterday on a fresh ncs installation.

C:/ncs/v2.5.1/zephyr/include/zephyr/linker/linker-defs.h:26:10: fatal error: offsets.h: No such file or directory
   26 | #include <offsets.h>
      |          ^~~~~~~~~~~
compilation terminated.

I fixed it in both cases with this line:
#include **<../subsys/testsuite/ztest/include/zephyr/offsets.h>**

Building a free-standing (T3) application.

```
C:/ncs/v2.5.1/zephyr/include/zephyr/linker/linker-defs.h:26:10: fatal error: offsets.h: No such file or directory
   26 | #include <offsets.h>
      |          ^~~~~~~~~~~
compilation terminated.
[8/369] Building C object drivers/display/CMakeFiles/..__..__..__Projects__Chips__test-origin-tft__drivers__display.dir/display_ili9xxx.c.obj
FAILED: drivers/display/CMakeFiles/..__..__..__Projects__Chips__test-origin-tft__drivers__display.dir/display_ili9xxx.c.obj 
C:\ncs\toolchains\c57af46cb7\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -DKERNEL -DLV_CONF_INCLUDE_SIMPLE=1 -DLV_CONF_PATH=C:/ncs/v2.5.1/zephyr/modules/lvgl/lv_conf.h -DNRF52840_XXAA -DPICOLIBC_INTEGER_PRINTF_SCANF -D_FORTIFY_SOURCE=1 -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -IC:/ncs/v2.5.1/zephyr/include -Izephyr/include/generated -IC:/ncs/v2.5.1/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/ncs/v2.5.1/zephyr/lib/posix/getopt/. -IC:/ncs/v2.5.1/zephyr/soc/arm/nordic_nrf/common/. -IC:/ncs/v2.5.1/nrf/include -IC:/ncs/v2.5.1/nrf/tests/include -IC:/ncs/v2.5.1/modules/hal/cmsis/CMSIS/Core/Include -IC:/ncs/v2.5.1/zephyr/modules/cmsis/. -IC:/ncs/v2.5.1/modules/hal/nordic/nrfx -IC:/ncs/v2.5.1/modules/hal/nordic/nrfx/drivers/include -IC:/ncs/v2.5.1/modules/hal/nordic/nrfx/mdk -IC:/ncs/v2.5.1/zephyr/modules/hal_nordic/nrfx/. -IC:/ncs/v2.5.1/modules/lib/gui/lvgl/src -IC:/ncs/v2.5.1/zephyr/modules/lvgl/. -IC:/ncs/v2.5.1/modules/debug/segger/SEGGER -IC:/ncs/v2.5.1/modules/debug/segger/Config -isystem C:/ncs/v2.5.1/nrfxlib/crypto/nrf_cc310_platform/include -fno-strict-aliasing -Os -imacros C:/Projects/Chips/test-origin-tft/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=C:/ncs/toolchains/c57af46cb7/opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi -imacros C:/ncs/v2.5.1/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=C:/Projects/Chips/test-origin-tft=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/ncs/v2.5.1/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/ncs/v2.5.1=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -D_POSIX_THREADS -std=c99 -MD -MT drivers/display/CMakeFiles/..__..__..__Projects__Chips__test-origin-tft__drivers__display.dir/display_ili9xxx.c.obj -MF drivers\display\CMakeFiles\..__..__..__Projects__Chips__test-origin-tft__drivers__display.dir\display_ili9xxx.c.obj.d -o drivers/display/CMakeFiles/..__..__..__Projects__Chips__test-origin-tft__drivers__display.dir/display_ili9xxx.c.obj -c ../drivers/display/display_ili9xxx.c
In file included from C:/ncs/v2.5.1/zephyr/include/zephyr/app_memory/app_memdomain.h:9,
                 from C:/ncs/v2.5.1/zephyr/include/zephyr/rtio/rtio.h:31,
                 from C:/ncs/v2.5.1/zephyr/include/zephyr/drivers/spi.h:29,
                 from ../drivers/display/display_ili9xxx.h:13,
                 from ../drivers/display/display_ili9xxx.c:9:
C:/ncs/v2.5.1/zephyr/include/zephyr/linker/linker-defs.h:26:10: fatal error: offsets.h: No such file or directory
   26 | #include <offsets.h>
      |          ^~~~~~~~~~~
compilation terminated.
```

henrikbrixandersen, 2024-01-16:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

de-nordic, 2024-01-22:
Eh... command line or Kconfig list to reproduce would be nice, huh?

I have tried building Hello World adding offsets.h to main and flash_shell adding the offsets.h to soc_nrf_flash.c. Both built for nrf52840dk_nrf52840 with the Zephyr at a9779eca400e67e2eebfc1f774eb4186efdc7a96.
Both built with -DCONFIG_I2C=y, and looks ok.
The example, that is supposed to not build on sdk-nrf, but is fine to build with the Zephyr:
https://github.com/robertpaananen/mre_bug_offsets_h
```
west build -d builds/nrf -b nrf52840dk_nrf52840 ../ncs1/mre_bug_offsets_h/ -DCONFIG_I2C=y
```
fails, but when you re-try build it builds fine.
Seems that the offsets.h gets generated after the file, triggering the problem, is compiled.

eriksage, 2024-03-06:
I see the same issue on SDK 0.16.9, Zephyr v3.6.0 with board nucleo_f767zi. (Using SPI and CAN)
However it is not consistent. Sometimes it builds from a clean build folder without issues, other times it complains about the missing <offsets.h>. About 50/50. Building the second time always succeeds. 

Because of this random behavior and that some report that it helps to add includes, could it be that this is caused by the parallel nature of ninja and that it sometimes manage to create offsets.h in time?

josuah, 2024-03-17:
Would there be a missing dependency somewhere in some `CMakeLists.txt`? If the build is parallel, one file could be built before another, and depending on the order, there could be a missing dependency or not?

A manual workaround is to generate the `offsets.h` explicitly:

```
~/zephyrproject$ cmake --build build --target offsets_h
[1/1] Generating include/generated/offsets.h
~/zephyrproject$
```

htcrane, 2024-05-07:
I am also seeing this with the following:

- Cmake 3.29.2
- Zephyr Build: build: v3.6.0-2750-g6a497ac3735c
- Zephyr SDK: 0.16.5

It occurs anytime you perform a pristine build or have otherwise nuked the build directory. If you build again after the failure (without the pristine flag), the process will continue as normal. _I believe `offsets.h` is generated by the Zephyr SDK, and therefore is not built early enough in the compiling process._ Hence, the failure upon first clean builds but success after continuing the failed build.

Looking a the `offsets.h` file (in your build dir/zephyr/include/generated), there are no external includes so ideally, it should be one of the first built with a gated build mechanism (@eriksage, I agree it is the parallelism causing issues).

henrikbrixandersen, 2024-05-08:
> It occurs anytime you perform a pristine build or have otherwise nuked the build directory.

Please provide concise steps for how to reproduce this (list the exact commands used along with full error output). I am not able to reproduce this just by doing a pristine build.

hraftery, 2024-05-15:
I looked into this because it's been causing us strife for quite a while. The build process is confounding enough without worrying about intermittent tooling issues like this.

I found that according to the [build system docs](https://docs.zephyrproject.org/latest/build/cmake/index.html#pre-build) the `offsets.h` file is created during **pre-build**. Critically that comes after the "Configuration Phase" as the first step during the "Build Phase". So this is under the control of either `make` or `ninja`.

In my case it is `ninja`, so I tracked down my `build.ninja` "makefile". Of course, it is 4.3MB and ninja run builds without inter-dependencies in parallel, so what could possibly go wrong?

I ran `ninja -t deps` and `ninja -t graph` to get a view into the dependency tree. The relevant build target was easy to spot in the graph:

[image]

So to understand why it's not getting prioritised we simply need to look at it in context within the overall build. I've drawn a big red box around the previous screenshot to make it easy to spot.

[image]

Yeah. So I'm still looking into it...

hraftery, 2024-05-15:
Oh man, that was easy.

Everything that was included using `target_sources` had dependencies like `zephyr_generated_headers`. So did drivers I'd added using `add_subdirectory`, but which have a path already known to Zephyr, like input drivers. They get a line in the `build.ninja` file like:

`cmake_object_order_depends_target_drivers__input: phony || [...] zephyr/zephyr_generated_headers`

However, drivers in an unknown path, which are included in my `CMakeLists.txt` file using a combination of `add_subdirectory` **and** `target_link_library`, get no such dependency. I found an [old thread](https://lists.zephyrproject.org/g/devel/topic/cmake_application_project/28375071) which got me on the right track.

### **All I had to do**, was add `add_dependencies(LIBRARY_NAME, offsets_h)` to my `CMakeLists.txt` file, and clean builds are now fast and successful.

If you're following the Zephyr library naming convention, as I am, that would be something like `add_dependencies(drivers__fuel_gauge__max17043 offsets_h)`, or in the OP's case, maybe `add_dependencies(drivers__display__display_ili9488 offsets_h)`.

I'm not sure why this dependency isn't included transitively already...

henrikbrixandersen, 2024-05-15:
CC: @tejlmand

sidharth-prabukumar, 2024-05-29:
@hraftery Using `zephyr_library_add_dependencies(offsets_h)` worked for me

closed 2025-01-28 by github-actions[bot]