zephyrproject-rtos/zephyr #67289: subsys: llext: app fails if CONFIG_LOG is enabled

kinsamanka, 2024-01-08:
I've created a [test app](https://gist.github.com/kinsamanka/d514287468ddf9dcb7aa5facb6a340e1) based on the [llext test code](https://github.com/zephyrproject-rtos/zephyr/tree/zephyr-v3.5.0/tests/subsys/llext).

Enabling `CONFIG_LOG` causes the app to fail.

**To reproduce**
```shell
mkdir tmp
cd tmp
wget https://gist.github.com/kinsamanka/d514287468ddf9dcb7aa5facb6a340e1/archive/df4f7b9feadddd4824e029833383cacebbf7ba22.tar.gz -O - | tar xzf -
west build -b qemu_cortex_m3 -t run -p auto d5142*
```

**Logs**

`CONFIG_LOG=y` and `CONFIG_LOG_MODE_IMMEDIATE=y`
```shell
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build zephyr-v3.5.0 ***
[00:00:00.000,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:00:00.000,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, size 42
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240
[00:00:00.000,000] <dbg> llext: llext_find_tables: symtab at 8
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, fla[00:00:00.010,000] <err> os: ***** USAGE FAULT *****
[00:00:00.010,000] <err> os:   Illegal load of EXC_RETURN into PC
[00:00:00.010,000] <err> os: r0/a1:  0x20008ad0  r1/a2:  0x00004324  r2/a3:  0x00000033
[00:00:00.010,000] <err> os: r3/a4:  0x000012e9 r12/ip:  0x20008ac8 r14/lr:  0xfffffffd
[00:00:00.010,000] <err> os:  xpsr:  0x00000000
[00:00:00.010,000] <err> os: Faulting instruction address (r15/pc): 0x00004324
[00:00:00.010,000] <err> os: >>> ZEPHYR FATAL ERROR 34: Unknown error on CPU 0
[00:00:00.010,000] <err> os: Current thread: 0x200000c8 (unknown)
[00:00:00.010,000] <err> os: Halting system
qemu-system-arm: terminating on signal 2
ninja: build stopped: interrupted by user.
```

`# CONFIG_LOG is not set`
``` shell
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build zephyr-v3.5.0 ***
load result 0
hello_world fn at 0x20000b95
hello world
A number is 42
qemu-system-arm: terminating on signal 2
ninja: build stopped: interrupted by user.
```

`CONFIG_LOG=y` and `CONFIG_LOG_MODE_MINIMAL=y`
```shell
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build zephyr-v3.5.0 ***
D: Loading relocatable or shared elf
D: Finding ELF tables...
D: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0
D: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56
D: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24
D: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0
D: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0
D: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37
D: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33
D: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, size 42
D: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240
D: symtab at 8
D: section 9 at 3d0: name 9, type 3, flags 0, addr 0, size 47
D: strtab at 9
D: section 10 at 3f8: name 17, type 3, flags 0, addr 0, size 81
D: shstrtab at 10
D: Mapping ELF sections...
D: section 0 name 
D: Not copied section 
D: section 1 name .text
D: section 2 name .rel.text
D: Not copied section .rel.text
D: section 3 name .data
D: section 4 name .bss
D: section 5 name .rodata
D: section 6 name .comment
D: Not copied section .comment
D: section 7 name .ARM.attributes
D: Not copied section .ARM.attributes
D: section 8 name .symtab
D: Not copied section .symtab
D: section 9 name .strtab
D: Not copied section .strtab
D: section 10 name .shstrtab
D: Not copied section .shstrtab
D: Allocation memory for ELF sections...
D: Copying sections...
D: Counting exported symbols...
D: symbol count 15
D: unhandled symbol 0, name , type tag 0, bind 0, sect 0
D: unhandled symbol 1, name hello_world.c, type tag 4, bind 0, sect 65521
D: unhandled symbol 2, name , type tag 3, bind 0, sect 1
D: unhandled symbol 3, name , type tag 3, bind 0, sect 3
D: unhandled symbol 4, name , type tag 3, bind 0, sect 4
D: unhandled symbol 5, name , type tag 3, bind 0, sect 5
D: unhandled symbol 6, name $d, type tag 0, bind 0, sect 5
D: unhandled symbol 7, name number, type tag 1, bind 0, sect 5
D: unhandled symbol 8, name $t, type tag 0, bind 0, sect 1
D: unhandled symbol 9, name $d, type tag 0, bind 0, sect 1
D: unhandled symbol 10, name $t, type tag 0, bind 0, sect 1
D: unhandled symbol 11, name , type tag 3, bind 0, sect 6
D: unhandled symbol 12, name , type tag 3, bind 0, sect 7
D: function symbol 13, name hello_world, type tag 2, bind 1, sect 1
D: unhandled symbol 14, name printk, type tag 0, bind 1, sect 0
D: Allocating memory for symbol table...
D: Copying symbols...
D: function symbol 0 name hello_world addr 0x2000035d
D: Linking ELF...
D: relocation section .rel.text (2) linked to section 8 has 3 relocations
D: relocation 2:0 info 502 (type 2, sym 5) offset 40 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: found section symbol  addr 0x200003a0
I: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 40 link section 8
I: writing relocation symbol  type 2 sym 5 at addr 0x20000384 addr 0x200003a0
D: relocation 2:1 info e02 (type 2, sym 14) offset 44 sym_name printk sym_type 0 sym_bind 1 sym_ndx 0
I: found symbol printk at 0x273f, updating op code 0x20000388
I: relocating (linking) symbol printk type 0 binding 1 ndx 0 offset 44 link section 8
I: writing relocation symbol printk type 2 sym 14 at addr 0x20000388 addr 0x273f
D: relocation 2:2 info 502 (type 2, sym 5) offset 48 sym_name  sym_type 3 sym_bind 0 sym_ndx 5
I: found section symbol  addr 0x200003b0
I: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 48 link section 8
I: writing relocation symbol  type 2 sym 5 at addr 0x2000038c addr 0x200003b0
D: loaded module, .text at 0x2000035c, .rodata at 0x2000039c
I: Loaded extension hello
load result 0
hello_world fn at 0x2000035d
hello world
A number is 42
D: freeing memory region 0
D: freeing memory region 2
qemu-system-arm: terminating on signal 2
ninja: build stopped: interrupted by user.
```
**Environment:**
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Version: 3.5.0

teburd, 2024-01-09:
Can you provide an object dump with the relocations and asm associated with it? @laurenmurphyx64 is working on a more elaborate sample app at this moment and noticed some new relocation instructions that weren't currently supported for arm and its quite possible this is related.

kinsamanka, 2024-01-09:
The test code is a modified version of the llext twister test that works well:

```shell
~/new_zephyrproject/workspaces/v3.5.0$ zephyr/scripts/twister -p qemu_cortex_m3 -T zephyr/tests/subsys/llext -v
ZEPHYR_BASE unset, using "/home/test/new_zephyrproject/workspaces/v3.5.0/zephyr"
Renaming output directory to /home/test/new_zephyrproject/workspaces/v3.5.0/twister-out.2
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.5.0
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/test/new_zephyrproject/workspaces/v3.5.0/twister-out/testplan.json
INFO    - JOBS: 12
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - 1/1 qemu_cortex_m3            tests/subsys/llext/llext.simple.arm                PASSED (qemu 2.094s)

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 11.07 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 626 platforms (0.16%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/test/new_zephyrproject/workspaces/v3.5.0/twister-out/twister.json
INFO    - Writing xunit report /home/test/new_zephyrproject/workspaces/v3.5.0/twister-out/twister.xml...
INFO    - Writing xunit report /home/test/new_zephyrproject/workspaces/v3.5.0/twister-out/twister_report.xml...
INFO    - Run completed
```

arm-zephyr-eabi-objdump -d hello_world.llext:
```
hello_world.llext:     file format elf32-littlearm

Disassembly of section .text:

00000000 <hello_world>:
   0:   b580            push    {r7, lr}
   2:   af00            add     r7, sp, #0
   4:   4b08            ldr     r3, [pc, #32]   ; (28 <hello_world+0x28>)
   6:   0018            movs    r0, r3
   8:   4b08            ldr     r3, [pc, #32]   ; (2c <hello_world+0x2c>)
   a:   f000 f813       bl      34 <hello_world+0x34>
   e:   222a            movs    r2, #42 ; 0x2a
  10:   4b07            ldr     r3, [pc, #28]   ; (30 <hello_world+0x30>)
  12:   0011            movs    r1, r2
  14:   0018            movs    r0, r3
  16:   4b05            ldr     r3, [pc, #20]   ; (2c <hello_world+0x2c>)
  18:   f000 f80c       bl      34 <hello_world+0x34>
  1c:   46c0            nop                     ; (mov r8, r8)
  1e:   46bd            mov     sp, r7
  20:   bc80            pop     {r7}
  22:   bc01            pop     {r0}
  24:   4700            bx      r0
  26:   46c0            nop                     ; (mov r8, r8)
  28:   00000004        .word   0x00000004
  2c:   00000000        .word   0x00000000
  30:   00000014        .word   0x00000014
  34:   4718            bx      r3
  36:   46c0            nop                     ; (mov r8, r8)
```

arm-zephyr-eabi-objdump -x hello_world.llext:
```
hello_world.llext:     file format elf32-littlearm
hello_world.llext
architecture: armv4t, flags 0x00000011:
HAS_RELOC, HAS_SYMS
start address 0x00000000
private flags = 0x5000000: [Version5 EABI]

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         00000038  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE
  1 .data         00000000  00000000  00000000  0000006c  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  00000000  00000000  0000006c  2**0
                  ALLOC
  3 .rodata       00000025  00000000  00000000  0000006c  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  4 .comment      00000021  00000000  00000000  00000091  2**0
                  CONTENTS, READONLY
  5 .ARM.attributes 0000002a  00000000  00000000  000000b2  2**0
                  CONTENTS, READONLY
SYMBOL TABLE:
00000000 l    df *ABS*  00000000 hello_world.c
00000000 l    d  .text  00000000 .text
00000000 l    d  .data  00000000 .data
00000000 l    d  .bss   00000000 .bss
00000000 l    d  .rodata        00000000 .rodata
00000000 l     O .rodata        00000004 number
00000000 l    d  .comment       00000000 .comment
00000000 l    d  .ARM.attributes        00000000 .ARM.attributes
00000000 g     F .text  00000034 hello_world
00000000         *UND*  00000000 printk

RELOCATION RECORDS FOR [.text]:
OFFSET   TYPE              VALUE 
00000028 R_ARM_ABS32       .rodata
0000002c R_ARM_ABS32       printk
00000030 R_ARM_ABS32       .rodata
```

teburd, 2024-01-10:
please add it with the relocations, -r

teburd, 2024-01-10:
Thank you, I'll take a look here tomorrow

teburd, 2024-01-12:
@laurenmurphyx64 please take a look when you get a chance, I really think this might be related to what you are already working on but if not we can figure it out.

kinsamanka, 2024-01-15:
The problem was caused by a stack overflow (found using `CONFIG_STACK_SENTINEL=y`):
```shell
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build zephyr-v3.5.0 ***
[00:00:00.000,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:00:00.000,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, size 42
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240
[00:00:00.000,000] <dbg> llext: llext_find_tables: symtab at 8
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, fl[00:00:00.010,000] <err> os: r0/a1:  0x00000002  r1/a2:  0x200000d8  r2/a3:  0xf0f0f0f0
[00:00:00.010,000] <err> os: r3/a4:  0x20008c08 r12/ip:  0x200001c8 r14/lr:  0x00000d8d
[00:00:00.010,000] <err> os:  xpsr:  0x4100000f
[00:00:00.010,000] <err> os: Faulting instruction address (r15/pc): 0x000028d2
[00:00:00.010,000] <err> os: >>> ZEPHYR FATAL ERROR 2: Stack overflow on CPU 0
[00:00:00.010,000] <err> os: Fault during interrupt handling

[00:00:00.010,000] <err> os: Current thread: 0x200000d8 (unknown)
[00:00:00.010,000] <err> os: Halting system
```

Using `CONFIG_MAIN_STACK_SIZE=2048` fixed the problem:
```shell
qemu-system-arm: warning: nic stellaris_enet.0 has no peer
Timer with period zero, disabling
*** Booting Zephyr OS build zephyr-v3.5.0 ***
[00:00:00.000,000] <dbg> llext: llext_load: Loading relocatable or shared elf
[00:00:00.000,000] <dbg> llext: do_llext_load: Finding ELF tables...
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 0 at 268: name 0, type 0, flags 0, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 1 at 290: name 31, type 1, flags 6, addr 0, size 56
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 2 at 2b8: name 27, type 9, flags 40, addr 0, size 24
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 3 at 2e0: name 37, type 1, flags 3, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 4 at 308: name 43, type 8, flags 3, addr 0, size 0
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 5 at 330: name 48, type 1, flags 2, addr 0, size 37
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 6 at 358: name 56, type 1, flags 30, addr 0, size 33
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 7 at 380: name 65, type 1879048195, flags 0, addr 0, size 42
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 8 at 3a8: name 1, type 2, flags 0, addr 0, size 240
[00:00:00.000,000] <dbg> llext: llext_find_tables: symtab at 8
[00:00:00.000,000] <dbg> llext: llext_find_tables: section 9 at 3d0: name 9, type 3, flags 0, addr 0, size 47
[00:00:00.010,000] <dbg> llext: llext_find_tables: strtab at 9
[00:00:00.010,000] <dbg> llext: llext_find_tables: section 10 at 3f8: name 17, type 3, flags 0, addr 0, size 81
[00:00:00.010,000] <dbg> llext: llext_find_tables: shstrtab at 10
[00:00:00.010,000] <dbg> llext: do_llext_load: Mapping ELF sections...
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 0 name 
[00:00:00.010,000] <dbg> llext: llext_map_sections: Not copied section 
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 1 name .text
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 2 name .rel.text
[00:00:00.010,000] <dbg> llext: llext_map_sections: Not copied section .rel.text
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 3 name .data
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 4 name .bss
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 5 name .rodata
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 6 name .comment
[00:00:00.010,000] <dbg> llext: llext_map_sections: Not copied section .comment
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 7 name .ARM.attributes
[00:00:00.010,000] <dbg> llext: llext_map_sections: Not copied section .ARM.attributes
[00:00:00.010,000] <dbg> llext: llext_map_sections: section 8 name .symtab
[00:00:00.010,000] <dbg> llext: llext_map_sections: Not copied section .symtab
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 9 name .strtab
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .strtab
[00:00:00.020,000] <dbg> llext: llext_map_sections: section 10 name .shstrtab
[00:00:00.020,000] <dbg> llext: llext_map_sections: Not copied section .shstrtab
[00:00:00.020,000] <dbg> llext: do_llext_load: Allocation memory for ELF sections...
[00:00:00.020,000] <dbg> llext: do_llext_load: Copying sections...
[... 18 lines cut ...]
[... cut ...]
[00:00:00.040,000] <inf> llext: relocating (linking) symbol  type 3 binding 0 ndx 5 offset 48 link section 8
[00:00:00.040,000] <inf> llext: writing relocation symbol  type 2 sym 5 at addr 0x200003c4 addr 0x200003e8
[00:00:00.040,000] <dbg> llext: do_llext_load: loaded module, .text at 0x20000394, .rodata at 0x200003d4
[00:00:00.050,000] <inf> llext: Loaded extension hello
load result 0
hello_world fn at 0x20000395
hello world
A number is 42
[00:00:00.050,000] <dbg> llext: llext_unload: freeing memory region 0
[00:00:00.050,000] <dbg> llext: llext_unload: freeing memory region 2
```

closed 2024-01-15 by kinsamanka