zephyrproject-rtos/zephyr #67327: RTIO Support: SPI

teburd, 2024-01-08:
**Is your feature request related to a problem? Please describe.**

RTIO is a feature that enables lock-free queue like requests to hardware peripherals, allowing for things like SPI transactions to be started in an ISR context. This is very useful for sensors and is part of the newer sensor API. Peripheral drivers should support this API in a broad way in Zephyr

**Describe the solution you'd like**

SPI bus drivers supporting RTIO broadly.

**Additional context**

Status of SPI bus driver RTIO support

* [x] sam
* [x] nxp lpspi
* [ ] nxp flexcomm
* [ ] nrf spi
* [ ] nrf spim
* [ ] st spi
* [ ] dw spi (intel and arc use this)

teburd, 2024-01-08:
@henrikbrixandersen @yperess

FlorianWeber1018, 2025-01-27:
Should we consider to add the rpi pio spi driver to the list as well?