zephyrproject-rtos/zephyr #67331: Retained memory subsystem only support taking blocks of 1kB multiples on ram

Laczen, 2024-01-08:
When using the retained memory subsystem to share data with another application or to maintain data after reboot the sram node needs to be reduced to account for this. This reduction can only be done in 1kB multiples as the sram size is retrieved from the dts in kB.

This leeds to wasted ram and makes the subsystem unusable on devices with limited ram.

nordicjm, 2024-01-08:
That seems more like a configuration/build system issue?

Laczen, 2024-01-08:
> That seems more like a configuration/build system issue?

It is, as you are both a build system maintainer as the author of the subsystem you seemed like the ideal assignee.

kartben, 2024-01-08:
@Laczen can you provide an example of what you mean, or point to the code that you think is "retrieving sram size from dts in kilobytes"? Nothing prevents one to size in bytes in devicetree instead of using DT_SIZE_K macro? There is even a sample in the documentation that specifically shows how to have just 1 bytes of retained mem, I believe.

Laczen, 2024-01-08:
> @Laczen can you provide an example of what you mean, or point to the code that you think is "retrieving sram size from dts in kilobytes"? Nothing prevents one to size in bytes in devicetree instead [...]

When you reduce the sram in dts with 1 byte the sram size reported at the end of compilation is reduced with 1kB. I think it is caused by the linker scripts taking the size in kB.

kartben, 2024-01-08:
I see. Would be good to update the title and labels of the issue then, as this doesn't seem to be specific to retention subsystem at all. PR bot can then automatically assign based on this. Thanks!

Laczen, 2024-01-09:
@kartben, the retained_mem subsystem on ram is AFAIK the only subsystem using this construct. The solution for this bug wil probably need changes not to use this construct.

kartben, 2024-01-09:
Not sure why assignee bot is [confused](https://github.com/zephyrproject-rtos/zephyr/actions/runs/7458160491/job/20291606297) so manually assigning...