zephyrproject-rtos/zephyr #67339: I2C broken on SAM E70 3.5 vs 3.3

gwillden, 2024-01-08:
On Zephyr v3.3, I2C transfers work correctly on the SAM E70.
On Zephyr v3.5, the same I2C transfers are failing.

```
const struct device *const i2c_dev = DEVICE_DT_GET(DT_NODELABEL(twihs0));
int main(void)
{
    while(1) {
        uint8_t wbuf[1] = {IIC_TEMP_INTERNAL_ADDR};
        uint8_t rbuf[2] = {0,0};
        i2c_write_read(i2c_dev, LOC_IIC_TEMP1_ADDR, wbuf, sizeof(wbuf), rbuf, sizeof(rbuf));
        uint16_t result = (((uint16_t) rbuf[0] << 8) | ((uint16_t) rbuf[1] ));
        k_sleep(K_MSEC(100));
    }
    return 0;
}
```
This is a logic analyzer trace of the transfer under v3.3
[image]
This is a trace of the transfer under v3.5.  The controller stops after the intial write.
[image]

There were some changes to i2c_sam_twihs.c, same70.dtsi and atmel,sam-i2c-twihs.yaml relating to the clock control but selectively reverting those changes doesn't seem to fix the issue on v3.5.

In my full application, other clocks for UDP, UARTS, etc work fine.

Is there a config or devicetree setting that I missed adding during the upgrade?

gwillden, 2024-01-09:
I was able to bisect the problem to:

commit ea24dd40eb61065942f9577a69ea8b6defca209b
Author: Gerson Fernando Budke <nandojve@gmail.com>
Date:   Thu Aug 3 12:57:38 2023 +0200

My defconfig had 
CONFIG_ICACHE=y
CONFIG_DCACHE=y
But not CONFIG_CACHE_MANAGEMENT

I recommend that a change be made to require CONFIG_CACHE_MANAGEMENT if either CONFIG_ICACHE or CONFIG_DCACHE is set so that this unexpected behaviour doesn't manifest itself.

nandojve, 2024-01-09:
Hi @gwillden ,

In the #61005 was fixed a lot of details about code initialization. The change allows Zephyr users to enable/disable cache based on application needs. The #61005 adjust all in tree boards accordingly and that is the recommendation since then.

Zephyr is improving constantly and the Atmel drivers were not exactly designed with cache in mind. All the code is maintained only by community members and we are happy to review contributions to improve in general.

So, if add the CONFIG_CACHE_MANAGEMENT=y in your board and that solve your problem I believe that everything is fine and there is no issues here.

If the CONFIG_CACHE_MANAGEMENT=y not solve your problem fell free to reopen the issue because it could then be a problem in the driver.

closed 2024-01-09 by nandojve

gwillden, 2024-01-09:
The problem was the surprise and silent disabling of the caching that had side effects in the I2C code.  The defconfig system should have a warning to tell you that CONFIG_ICACHE and CONFIG_DCACHE are effectively turned off if you don't also have CONFIG_CACHE_MANAGEMENT=y.

nandojve, 2024-01-09:
I understand you comment but that is not SoC specific since the configuration came from Arch:

https://github.com/zephyrproject-rtos/zephyr/blob/14c68c9438476268e4de25b6a2778883993c955c/arch/Kconfig#L890-L911

Cache is something that have been in discussing for sometime. This is another clean up #66524 and there are issues and bugs _Originally posted by @nandojve in https://github.com/zephyrproject-rtos/zephyr/pull/66524#discussion_r1437114579_. Maybe you got the problem that cache lines are not proper cleaned. If you are using some bootloader it could be worst.

simontheu, 2024-02-18:
I have the same issue, however the CONFIG_CACHE_MANAGEMENT=y does not fix the issue for me. I have not done a bisect yet. To see if the issue arrives with the same commit.

For me, writes are fine. But when doing a write_read, or separate read followed by a write, I get stuck on a sempahore wait, and the scope shows a clock and data line stuck low.
[image]

simontheu, 2024-03-09:
I am still suffering with this, and investigating more.

nandojve, 2024-03-10:
Hi @simontheu ,

I recommend you try to investigate on mainline. Everything should be fixed on mainline and than back-ported if required.
Take in consideration that Zephyr had improved in general and now there is even LTO (Link Time Optimization) available. This means that code is more and more unpredictable and drivers should be careful designed.

This driver is probably from old times and may be inspired by ASF which is for Super Loop applications. Maybe it necessary some redesign including concurrency and cache maintenance if those are influencing the behavior.

My suggestion is make sure you can run your application without D-Cache. Disable LTO option to avoid compiler back-end reorder instructions and then you try to fix the concurrency problem (because this seems to be the real issues here). Once that is fixed you can enable LTO to see if driver still Ok. If not, you still need to deal with concurrency access/states. Only after that try enable D-Cache. If you start to get problems you need to move the problematic variables into a Non-Cache memory region. This solution makes the driver 100% compatible between all devices. A more optimized version with cache can be valid too but I suggest do it on a next round if is that the case.