zephyrproject-rtos/zephyr #67360: twister (ci) warnings in sysbuild images are not treated as errors

nordicjm, 2024-01-09:
**Describe the bug**
Twister in CI runs and sets the treat all warnings as errors Kconfig to prevent having samples/tests in the code base that emit warnings, however when used with sysbuild this only applies to the main image, any other image will not have the Kconfig set and can freely emit warnings which will not fail tests.

**Expected behavior**
Test should fail

**Impact**
Major

**Environment (please complete the following information):**
 - Commit SHA or Version used: 72c6ffb2d04bf1d9c23e0ef71464b3693d96ba98

henrikbrixandersen, 2024-02-10:
As I understand this issue, the problem is that while `CONFIG_COMPILER_WARNINGS_AS_ERRORS` is set to `y` for the main application domain, this is not propagated through to other sysbuild domains, e.g. MCUboot?

This Kconfig setting is Zephyr-specific and as such it cannot be universally set for all sysbuild domains, but only for those domains that are actually a Zephyr build (main application, MCBoot)?

henrikbrixandersen, 2024-02-10:
CC: @tejlmand

marc-hb, 2024-02-10:
> sets the treat all warnings as errors Kconfig

To clarify you mean C/C++ warnings, correct? Please edit the title and description if correct.

I've been looking for ways to make `dtc` and `kconfiglib.py` warnings fatal in CI too but these tools haven't been cooperative.

PerMac, 2024-02-12:
IMO it is related only to the build system and has nothing to do with twister. Twister just calls cmake build once and checks if it exit w/ w/o errors. It is up to the build system to decide how the warnings are treated. I recall having a chat with @nordicjm about this and I think the issue can be reproduced with calling (after fixing the paths):
```
/home/maciej/.pyenv/shims/cmake -B/home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service -DTC_RUNID=4a199cd0afc06bab71dc4b8ba09f0f44 -DCONFIG_COMPILER_WARNINGS_AS_ERRORS=y -DEXTRA_GEN_DEFINES_ARGS=--edtlib-Werror -GNinja -S/home/maciej/zephyrproject/zephyr/share/sysbuild -DAPP_DIR=/home/maciej/zephyrproject/zephyr/samples/subsys/ipc/rpmsg_service -DBOARD=nrf5340dk_nrf5340_cpuapp
```
and then
```
/home/maciej/.pyenv/shims/cmake --build /home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service
```

The output:
```
/home/maciej/.pyenv/shims/cmake --build /home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service
[5/16] Performing build step for 'rpmsg_service_remote'
[1/165] Preparing syscall dependency handling

[2/165] Generating include/generated/version.h
-- Zephyr version: 3.6.0-rc1 (/home/maciej/zephyrproject/zephyr), build: v3.6.0-rc1-46-g3d27258b0cbe
[34/165] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/ipc/rpmsg_service/rpmsg_backend.c.obj
/home/maciej/zephyrproject/zephyr/subsys/ipc/rpmsg_service/rpmsg_backend.c:158:13: warning: 'ipm_callback' defined but not used [-Wunused-function]
  158 | static void ipm_callback(const struct device *dev,
      |             ^~~~~~~~~~~~
[88/165] Building C object zephyr/drivers/ipm/CMakeFiles/drivers__ipm.dir/ipm_nrfx_ipc.c.obj
/home/maciej/zephyrproject/zephyr/drivers/ipm/ipm_nrfx_ipc.c:247:13: warning: 'gipm_send' defined but not used [-Wunused-function]
  247 | static void gipm_send(uint32_t id)
      |             ^~~~~~~~~
/home/maciej/zephyrproject/zephyr/drivers/ipm/ipm_nrfx_ipc.c:147:12: warning: 'vipm_nrf_init' defined but not used [-Wunused-function]
  147 | static int vipm_nrf_init(const struct device *dev)
      |            ^~~~~~~~~~~~~
/home/maciej/zephyrproject/zephyr/drivers/ipm/ipm_nrfx_ipc.c:140:17: warning: 'vipm_nrf_max_id_val_get' defined but not used [-Wunused-function]
  140 | static uint32_t vipm_nrf_max_id_val_get(const struct device *dev)
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/home/maciej/zephyrproject/zephyr/drivers/ipm/ipm_nrfx_ipc.c:135:12: warning: 'vipm_nrf_max_data_size_get' defined but not used [-Wunused-function]
  135 | static int vipm_nrf_max_data_size_get(const struct device *dev)
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~
[164/165] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       31272 B       256 KB     11.93%
             RAM:       13416 B        64 KB     20.47%
           SRAM1:          0 GB        64 KB      0.00%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service/rpmsg_service_remote/zephyr/zephyr.elf for board: nrf5340dk_nrf5340_cpunet
[165/165] cd /home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/su...mples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service/rpmsg_service_remote/zephyr/zephyr.elf
[13/16] Performing build step for 'rpmsg_service'
[1/167] Preparing syscall dependency handling

[2/167] Generating include/generated/version.h
-- Zephyr version: 3.6.0-rc1 (/home/maciej/zephyrproject/zephyr), build: v3.6.0-rc1-46-g3d27258b0cbe
[47/167] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/ipc/rpmsg_service/rpmsg_backend.c.obj
FAILED: zephyr/CMakeFiles/zephyr.dir/subsys/ipc/rpmsg_service/rpmsg_backend.c.obj 
ccache /home/maciej/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=4096 -DNRF5340_XXAA_APPLICATION -DNRF_SKIP_FICR_NS_COPY_TO_RAM -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -I/home/maciej/zephyrproject/zephyr/kernel/include -I/home/maciej/zephyrproject/zephyr/arch/arm/include -I/home/maciej/zephyrproject/zephyr/include -I/home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service/rpmsg_service/zephyr/include/generated -I/home/maciej/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf53 -I/home/maciej/zephyrproject/zephyr/soc/common/nordic_nrf/. -I/home/maciej/zephyrproject/zephyr/soc/arm/nordic_nrf/common/. -I/home/maciej/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/maciej/zephyrproject/zephyr/modules/cmsis/. -I/home/maciej/zephyrproject/modules/hal/nordic/nrfx -I/home/maciej/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/maciej/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/maciej/zephyrproject/zephyr/modules/hal_nordic/nrfx/. -I/home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service/rpmsg_service/modules/libmetal/libmetal/lib/include -I/home/maciej/zephyrproject/modules/lib/open-amp/open-amp/lib/include -isystem /home/maciej/zephyrproject/zephyr/lib/libc/common/include -fno-strict-aliasing -Werror -Os -imac [... cut ...] src/rpmsg_service-stamp/rpmsg_service-build 
cd /home/maciej/zephyrproject/zephyr/twister-out/nrf5340dk_nrf5340_cpuapp/samples/subsys/ipc/rpmsg_service/sample.ipc.rpmsg_service/rpmsg_service && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
```

The issue is that the first build didn't care about warnings ( ```warning: 'ipm_callback' defined but not used [-Wunused-function]```) in `rpmsg_service_remote` and the build system failed only on the consecutive build `rpmsg_service`.

nordicjm, 2024-02-12:
> As I understand this issue, the problem is that while `CONFIG_COMPILER_WARNINGS_AS_ERRORS` is set to `y` for the main application domain, this is not propagated through to other sysbuild domains, e.g. MCUboot?

Yes, but also applies to TF-M which could be considered or or multiple applications

> This Kconfig setting is Zephyr-specific and as such it cannot be universally set for all sysbuild domains, but only for those domains that are actually a Zephyr build (main application, MCBoot)?

Sysbuild at present only supports adding a zephyr project as an image.

The ideal fix would be to add the same flag to sysbuild which, if set, would force it onto images which wouldn't take long to add, then twister would need to set that in a sysbuild-context for `sysbuild: true` tests, the hard part is deciding what to do about the numerous tests that would then start failing in CI (mostly ipc tests, possibly TF-M)

> To clarify you mean C/C++ warnings, correct?

Yes

> I've been looking for ways to make dtc and kconfiglib.py warnings fatal in CI too but these tools haven't been cooperative.

That would probably cause a lot of boards to no longer successfully build in CI since lots have warnings about e.g. multiple peripherals having overlapped IO addresses

marc-hb, 2024-02-12:
> That would probably cause a lot of boards to no longer successfully build in CI

I meant my CI, not Zephyr's CI. Edited.

commit 5af21409 (2024-03-18): twister: Use proper warnings-to-errors command for sysbuild

If sysbuild is used a flag SB_CONFIG_COMPILER_WARNINGS_AS_ERRORS
has to be used in order to turn warings to errors on all images.
Align pytest tests with the change.

Fixes: #67360
closed 2024-03-18 by carlescufi