zephyrproject-rtos/zephyr #67365: Bluetooth Controller: Not able to find extended advertising until after a few minutes after no BLE devices in the area for ~30 minutes

jakkra, 2024-01-09:
**Describe the bug**
If we leave the scanner in an area without BLE devices for about 30 minutes and then turn on an advertiser with extended advertising it takes about 2-4 minutes until the extended advertiser is seen. Even adding 20+ advertisers, d

The applications here use periodic advertising and syncing. Periodic advertising may not be needed, as the issue seems to be with extended scanning, however it could be caused due to periodic syncing.

It's not enough to leave the scanner without BLE devices for just a few minutes, about 30 minutes it has to be left idle/without BLE devices.

**Please also mention any information which could help others to understand
the problem you're facing:**
 - What target platform are you using?
   - nRF52833 
 - What have you tried to diagnose or workaround this issue:
   - Been debugging and looking around a bit in the BLE controller when it happens and it seems as the scanner finds the `ADV_EXT_IND` but fails moving to and looking for the `AUX_ADV_IND`.
 - Is this a regression?
   - Issue both in Zephyr 3.3 and main branch

**To Reproduce**
Right now I can only reproduce in our BLE AoA Locator application only. It's based on Zephyr 3.3, however I have successfully replaced the BLE Controller from Zephyr main and can still reproduce the issue. So I think it's safe to say the issue still exists in Zephyr main.

I'm in progress of trying to create a sample application based on `samples/bluetooth/direction_finding_connectionless_rx` to reproduce the issue but so far have not succeeded.

**Below is the steps to reproduce if I manage to reproduce it with a modified sample.**
1. Put a scanner and an advertiser inside a shielded box.
2. Turn on advertiser.
3. Wait for it it sync.
4. Turn off advertiser
5. Wait ~30 minutes
6. Turn on advertiser
7. Now it will take 3-4 minutes until the scanner prints out scan info and manages to sync with the advertiser.

**Expected behavior**
Finding extended advertising without large delay.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper):
- We have customer in field with this issue so for us important to find a solution as soon as possible.

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA: main as of day of issue creation 72c6ffb2d04bf1d9c23e0ef71464b3693d96ba98

**Other**
Initially discussed here with @cvinayak: https://discord.com/channels/720317445772017664/733036837576376341/1187715776690524190

cvinayak, 2024-01-10:
@jakkra Could you check if this helps? https://github.com/zephyrproject-rtos/zephyr/pull/64910

This implementation was added in v3.4.0 and removed in v3.5.0 (may explain why you do not see your issue fixed from v.3.3.0 to latest upstream, or maybe not) . But I am of the realization that this fix is needed (as I have seen OVERHEAD_START related assertion when in ISO controller usecases without these code changes). The fix is not optimal, ideally the lower link layer prepare queue should have been implemented as a ordered-list, something for improvement in the near future.

jakkra, 2024-01-11:
Hi @cvinayak, amazing, I _think_ this fixed the issue, but I want to do some more testing to be sure. Will let you know!

jakkra, 2024-01-16:
Hi again @cvinayak I can still reproduce the issue and it's not fixed by https://github.com/zephyrproject-rtos/zephyr/pull/64910 unfortunately. Can this issue be re-opened please.

I have implemented a workaround in our main application to restart scanning after 30s of no scan receive data (i.e. no BLE advertisers in the area) and this seems to make the issue never appear. 

Unfortunately I have been busy with other stuff and not had time to try to adjust the below application to reproduce the issue.

But here is the attached the application I'm trying to reproduce with. 
[aoa_receiver_multiple.zip](https://github.com/zephyrproject-rtos/zephyr/files/13947387/aoa_receiver_multiple.zip)

You need a radio shielded box and it has to be well shielded, if it's not and BLE advertisements leak in it's not ok. I suggest using a board with external antenna and just unplug the antenna in addition.

After ~30+ minutes make sure you have a AoA Transmitter close to the shielded box (use the direction_finding_tx_sample).

You can also have both the receiver and transmitter always in the shielded box and remotely turn on/off the transmitter, this way it's possibly to automate testing.

jakkra, 2024-01-16:
Good new @cvinayak I can reproduce now using 
[aoa_receiver_multiple_reproduce_67365.zip](https://github.com/zephyrproject-rtos/zephyr/files/13949033/aoa_receiver_multiple_reproduce_67365.zip)
The trick missing before was: 
[image]

Build with latest Zephyr main.

jakkra, 2024-01-16:
After starting a tag after about an hour of idle time inside the shielded box:

```
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan_aux.c:569
        prepare_cb: Actual EVENT_OVERHEAD_START_US = 138944946
```

cvinayak, 2024-01-17:
> After starting a tag after about an hour of idle time inside the shielded box:
> 
> ```
> ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan_aux.c:569
>         prepare_cb: Actual EVENT_OVERHEAD_START_US = 138944946
> ```

do you have this commit: https://github.com/zephyrproject-rtos/zephyr/commit/13c29940bf69a30e6d3285859ebd9d00708cb2d7  ?

jakkra, 2024-01-17:
> > After starting a tag after about an hour of idle time inside the shielded box:
> > ```
> > ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan_aux.c:569
> >         prepare_cb: Actual EVENT_OVERHEAD_START_US = 138944946
> > ```
> 
> do you have this commit: [13c2994](https://github.com/zephyrproject-rtos/zephyr/commit/13c29940bf69a30e6d3285859ebd9d00708cb2d7) ?

This is on latest main. But checked also and it's there.

jakkra, 2024-01-17:
No, but I did not try the sample in it's current state with more than one tag. I did a bunch of changes to this smaple application in order to replicate the flow in our main aplication, but did not verify it all with many tags, so I cannot rule out it's caused by an issue in the application.

cvinayak, 2024-01-17:
> After starting a tag after about an hour of idle time inside the shielded box:
> 
> ```
> ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan_aux.c:569
>         prepare_cb: Actual EVENT_OVERHEAD_START_US = 138944946
> ```

I think i found something that can explain this, but the issue is a pattern all over the code, will try to send a PR to address this first. Then lets try to reproduce the issue of not receiving adv reports after 30 mins of idle.

cvinayak, 2024-01-17:
Can you check if helps avoid the LL_ASSERT_OVERHEAD you saw:  https://github.com/zephyrproject-rtos/zephyr/pull/67725
Top commit only fixes issue with assert check and does not impact or fix controller functionality. Second commit possibly helps fix the reception after 30 minutes issue.

Correct me, after #64910 merge, the reception after 30 minutes of being idle, this issue still existed right?

jakkra, 2024-01-17:
@cvinayak got the assert even with https://github.com/zephyrproject-rtos/zephyr/pull/67725 now

```
TAGS:-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,, 0
TAGS:-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,-,, 0
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_scan_aux.c:571
        prepare_cb: Actual EVENT_OVERHEAD_START_US = 81632507
[02:18:37.427,215] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000003  r2/a3:  0x00000002
[02:18:37.427,246] <err> os: r3/a4:  0x00000000 r12/ip:  0x00006500 r14/lr:  0x00002779
[02:18:37.427,246] <err> os:  xpsr:  0x41000011
[02:18:37.427,276] <err> os: s[ 0]:  0x8e89bed6  s[ 1]:  0x01000000  s[ 2]:  0x00000000  s[ 3]:  0x200009e4
[02:18:37.427,307] <err> os: s[ 4]:  0x200009e4  s[ 5]:  0x00002601  s[ 6]:  0x000143b1  s[ 7]:  0x000137cb
[02:18:37.427,307] <err> os: s[ 8]:  0x00000000  s[ 9]:  0x00013f4f  s[10]:  0x00000000  s[11]:  0x020009e4
[02:18:37.427,337] <err> os: s[12]:  0x00002601  s[13]:  0x20000808  s[14]:  0x00000000  s[15]:  0x00000000
[02:18:37.427,368] <err> os: fpscr:  0x00000000
[02:18:37.427,398] <err> os: r4/v1:  0x20005de0  r5/v2:  0x0028d0f6  r6/v3:  0x00000007
[02:18:37.427,398] <err> os: r7/v4:  0x20005dc4  r8/v5:  0x00014429  r9/v6:  0x00000000
[02:18:37.427,429] <err> os: r10/v7: 0x00000000  r11/v8: 0x00000000    psp:  0x2000ed50
[02:18:37.427,429] <err> os: EXC_RETURN: 0x0
[02:18:37.427,429] <err> os: Faulting instruction address (r15/pc): 0x00002784
[02:18:37.427,490] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[02:18:37.427,490] <err> os: Fault during interrupt handling

[02:18:37.427,551] <err> os: Current thread: 0x20001fe8 (idle)
[02:18:37.606,262] <err> os: Halting system
*** Booting Zephyr OS build zephyr-v3.5.0-4151-gf97526ef9c01 **
```

cvinayak, 2024-01-17:
@jakkra this is very weird to get a `diff` value that is `81632507` when the inputs are mask to 24-bits using `HAL_TICKER_CNTR_MASK`.

Could you help me review this function? do you see anything obvious that I am not able to understand here:
https://github.com/zephyrproject-rtos/zephyr/blob/7f99677aa6b5f622dad0d89faf9df3b178f3b47e/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c#L521-L545

cvinayak, 2024-01-18:
Me thinking aloud.

For my reference, recent PRs that are impacting extended scanning implementation:
- https://github.com/zephyrproject-rtos/zephyr/pull/62189
  - I believe this helped pass some conformance tests back again
- https://github.com/zephyrproject-rtos/zephyr/pull/64681
  - This fixed stalled extended adv reports
  - Could `CONFIG_BT_CTLR_EARLY_ABORT_PREVIOUS_PREPARE=y` introduced back after it was reverted in #62189 be the problem?

Issues reproducible, in focus of investigation:
- https://github.com/zephyrproject-rtos/zephyr/issues/67365#issuecomment-1895627735
- https://github.com/zephyrproject-rtos/zephyr/issues/67365#issuecomment-1896036728

cvinayak, 2024-01-18:
> I get this one, is this also something seen by you?
> 
> ![image](https://private-user-images.githubusercontent.com/6350656/297365103-865bb5e7-9841-4d24-b988-d864201c5f30.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1NzU2MzEsIm5iZiI6MTcwNTU3NTMzMSwicGF0aCI6Ii82MzUwNjU2LzI5NzM2NTEwMy04NjViYjVlNy05ODQxLTRkMjQtYjk4OC1kODY0MjAxYzVmMzAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDExOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMThUMTA1NTMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MTM1MmU0OTIyMTlmMGU3ZGRkNjdjMTc3Y2YxYzMwYWY2MjVhZWJkM2M0YTA1ODNmYjA2ZjczNmVkMTc3N2UyOCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Pc-GcGN_LoSe99Va7dHqu2hwPVboMXqQXaluUjFk34Y)

Ok this is now fixed by https://github.com/zephyrproject-rtos/zephyr/pull/67778

I now have the application running without the above assertion (though not in radio chamber, but no periodic advertisers reported by the application, there are other legacy advertisers around):
[image]

@jakkra sorry to bother but while I am now running your application at my end, could you use the fix in #67778 and check if it helps in anyway?

jakkra, 2024-01-18:
@cvinayak Running it now

> I now have the application running without the above assertion (though not in radio chamber, but no periodic advertisers reported by the application, there are other legacy advertisers around):

I'm not 100% sure if "radio silence" is needed or if it's enough to not have any extended advertisers around.

I have no simple way to test as there are so much BLE stuff in the office, many doing extended adv. But I'll bring a setup home today and test if it's possible without a shielded box.

cvinayak, 2024-01-18:
@jakkra managed to reproduce:
- put my recever dk inside few anti-static bags
- turned on few periodic advertisers and a tag transmitter after ~30 mins
- no sync for upto a minute or so
- some syncs happened
- turned off all transmitting boards
- wait few seconds
- turned on only the tag transmitter
- then assert seen

issue confirmed from my side, now I need to analyse reasons this can happen 🤔

[image]

jakkra, 2024-01-18:
Okey that's great you can reproduce, just let me know if you want me to test something :)

cvinayak, 2024-01-18:
> Good new @cvinayak I can reproduce now using [aoa_receiver_multiple_reproduce_67365.zip](https://github.com/zephyrproject-rtos/zephyr/files/13949033/aoa_receiver_multiple_reproduce_67365.zip) The [...]

Is the issue only with continuous scanning? i.e. your original application used continuous scanning?

jakkra, 2024-01-18:
> Is the issue only with continuous scanning? i.e. your original application used continuous scanning?

Yes it's using continuous scanning, but with options to not in order to save power. Also what made me able to reproduce with this sample application was to switch to continuous scanning.

cvinayak, 2024-01-19:
@jakkra that narrow down my search vector, please try this fix: https://github.com/zephyrproject-rtos/zephyr/pull/67817

jakkra, 2024-01-19:
@cvinayak If I either cherry-pick that commit or check out your branch both gives me compile errors 
[image]

cvinayak, 2024-01-19:
@jakkra sorry, I submitted the PR from airport and did not compile for AoA sample... the PR updated now.

jakkra, 2024-01-19:
@cvinayak with https://github.com/zephyrproject-rtos/zephyr/pull/67817 I have so far not been able to reproduce any issues with this fix, fantastic!

cvinayak, 2024-01-19:
> @cvinayak with #67817 I have so far not been able to reproduce any issues with this fix, fantastic!

Same here, periodically have turning on transmitters occasionally and they application has been stable.

cvinayak, 2024-01-22:
@jakkra hope you continued testing and are satisfied that the fix #67817 is good. Are moving to latest upstream or if not which version do you want the fix back ported?

jakkra, 2024-01-23:
Hi @cvinayak I have run tests where I just cherry-picked this patch into 3.3 release and it seems to work as expected.
It would be nice if you just quickly review if this fix could cause other being used with the 3.3 release, or if it is fine as is.

cvinayak, 2024-01-23:
@jakkra #67817 is as-is good for v3.3 release (i have not tested if the cherry-pick is clean tho). You also need to cherry-pick #67778, as I have observed assertion related to bug in ULL versus LLL scheduling of auxiliary PDU reception when primary PDU and auxiliary PDU are close.

jakkra, 2024-01-23:
https://github.com/zephyrproject-rtos/zephyr/pull/67817 applies without issue.
https://github.com/zephyrproject-rtos/zephyr/pull/67778 required merging, but it looks safe. Will bring this one in also, thanks!
[image]

cvinayak, 2024-01-23:
@jakkra looks like you do not need this https://github.com/zephyrproject-rtos/zephyr/pull/67778, as related changes are not present in v3.3. There is no bug related to the assertion I see.

cvinayak, 2024-01-23:
@jakkra Correction for #67778, you need https://github.com/zephyrproject-rtos/zephyr/pull/60581

jakkra, 2024-01-24:
@cvinayak thanks, brought that one in too!

PR #64910 (2024-01-12): Bluetooth: Controller: Fix ull_prepare_dequeue for skipped events
Fix ull_prepare_dequeue to not skip events when preempt does not match the event in the head of the prepare queue. The head of the prepare queue does not match when
ull_prepare_dequeue has put the head of the queue to the last of the queue when it is calling lll_resume interface. This happens when there is already an active event which needs a preempt timeout to be setup and there is another non-resume event in the pipeline which now becomes the head. The fix ensure to restore the order of the events if a preempt timeout was to  [... cut]
PR #68045 (2024-01-29): [Backport v3.4-branch] Bluetooth: Controller: Fix extended scanning assertion
Backport d1e4c17e6dd43a623ba9396f1e427fb7959b3a44~2..d1e4c17e6dd43a623ba9396f1e427fb7959b3a44 from #67817.

Fixes #67365
PR #67921 (2024-02-01): [Backport v3.5-branch] Bluetooth: Controller: Fix extended scanning data length assertion
Backport 4158d9523ea98c3fb92fbd2396a340017381e6f0 from #67778.

Fixes #67365
closed 2024-01-24 by carlescufi