zephyrproject-rtos/zephyr #67376: doc: Documentation issue in 'develop/west/basics'

TorstenRobitzki, 2024-01-09:
**Describe the bug**

I'm still confused and unsure, what a "Workspace" in West/Zephyr means, after reading the documentation. Assume, I'm working on a local PC, working for different customers, developing Zephyr based firmware for the different customers.

Intuitively, I would create a git repository for every customer-project (using this term to not confuse this with west-project) and would add Zephyr as git submodule. So, every customer-project could have its different version of Zephyr.

https://docs.zephyrproject.org/latest/develop/west/workspaces.html#west-workspaces claims to describe the workspace concept in more details, but then starts to describe implementation details about projects.

So, what is a Workspace? When do I need one? How do I collaborate with others, using Workspaces? How are workspaces and projects related? What part of a workspace belongs into a git repository?

Maybe, this questions could be answered, if we are looking at build artifacts and versions? Assumes, I have a customer project with 3 different hex files, containing the firmware for 3 different microcontrollers. Each µC firmware uses Zephyr, but not necessary the very same version. There are a few source code libraries used by all firmwares and unit tests for that libraries running on the build PC. One additional artifact could be a zip file containing the 3 hex files.

And, of cause, I would use a CI for the official build and for that, I would need a minimum command line, that would build and run the unit tests and the zip file.

How would that map to West/Zephyr workspaces and projects?

Thanks in advance,

Torsten

**Environment**

* Page: `develop/west/basics`
* Version: main
* SHA-1: 2cfdfad2e238f61aad0608ad35c2a6bb710f957e

TorstenRobitzki, 2024-03-10:
Well, I've reported my Issue with the page. As a newbie with Zephyr, but plenty of experience in embedded SW development, I had really problems to gasp the basic concepts. Terms like "Workspace", "Project" etc. are very generic and are overloaded thousand times with different meanings and concepts. Describing these terms in a Zephyr context would probably had helped. I would have offered help to improve the documentation, but I still don't have a clear view as how to answer my questions from above.

paultino, 2024-05-10:
There is no need to close it under any circumstances. This is a very convenient position - to write a complex product, “everything works for me, you all (insert definition).” I would like to receive an answer to the question posed.

carlescufi, 2024-06-21:
The stale bot runs automatically, I will try to address this soon.

PR #74962 (2024-06-28): doc: west: Clarify basic west terms
Terms like the west workspace or even west projects and modules were not documented prominently enough, prompting users to ask for additional clarification. This patch attempts to resolve this by adding terms and explanation of concepts to the glossary and west doc pages.

Fixes #67376.
commit 45799481 (2024-06-28): doc: west: Clarify basic west terms

Terms like the west workspace or even west projects and modules were not
documented prominently enough, prompting users to ask for additional
clarification. This patch attempts to resolve this by adding terms and
explanation of concepts to the glossary and west doc pages.

Fixes #67376.
closed 2024-06-28 by nashif