zephyrproject-rtos/zephyr #67394: Incorrect Arguments for align_up Function in MCUBoot Build Process (Zephyr 3.5.0)

khrystynaO, 2024-01-09:
Description:

The Zephyr MCUBoot bootloader build process faces an issue following the addition of commit 4c7942e (Author: Jamie McCrae, Date: Mon Oct 30 11:59:57 2023 +0000). The error specifically arises in the alignment process, resulting in a build failure.

During the Zephyr MCUBoot bootloader build (version: 3.5.0), an error arises concerning the align_up function in the CMakeLists.txt file. The error message highlights that the function is being invoked with incorrect arguments, leading to the build failure.

Steps to Reproduce:

-  Clone the Zephyr repository (version: 3.5.0).
-  Use Zephyr SDK version 0.16.4.
-  Navigate to the MCUBoot bootloader directory (commit: e9131ee8b8ab9306d8f6ad410fd1e3d8a820ce10).
-  Configure the build using the following West build command:
```
west build -p -b blackpill_f411ce --sysbuild . \
  -- -Dmcuboot_DTC_OVERLAY_FILE=$(pwd)/boards/blackpill_f411ce.overlay \
  -Dmcuboot_EXTRA_DTC_OVERLAY_FILE=${ZEPHYR_BASE}/../bootloader/mcuboot/boot/zephyr/app.overlay \
  -DEXTRA_CONF_FILE=$(pwd)/boards/witty_cloud.conf
```
Expected Behavior:

The build process should complete successfully without any errors related to the align_up function.

Actual Behavior:

The build process fails, and the following error message is displayed:
```
user@pc:~/project/hello_world/hello_world$ west build -p -b ${BOARD} --sysbuild . --             -Dmcuboot_DTC_OVERLAY_FILE=$(pwd)/boards/${BOARD}.overlay             -Dmcuboot_EXTRA_DTC_OVERLAY_FILE=${ZEPHYR_BASE}/../bootloader/mcuboot/boot/zephyr/app.overlay             -DEXTRA_CONF_FILE=$(pwd)/boards/witty_cloud.conf
-- west build: generating a build system
Loading Zephyr module(s) (Zephyr base): extensions sysbuild_extensions python west root zephyr_module boards shields sysbuild_kconfig native_simulator_sb_extensions
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: blackpill_f411ce
Parsing /home/user/zephyr_project/zephyr/share/sysbuild/Kconfig
Loaded configuration '/home/user/project/hello_world/hello_world/build/empty.conf'
Merged configuration '/home/user/project/hello_world/hello_world/sysbuild.conf'
Merged configuration '/home/user/project/hello_world/hello_world/build/empty.conf'
Configuration saved to '/home/user/project/hello_world/hello_world/build/zephyr/.config'
Kconfig header saved to '/home/user/project/hello_world/hello_world/build/autoconf.h'
-- 
   *****************************
   * Running CMake for mcuboot *
   *****************************

Loading Zephyr default modules (Zephyr base).
-- Application: /home/user/zephyr_project/bootloader/mcuboot/boot/zephyr
-- CMake version: 3.28.1
-- Found Python3: /usr/bin/python (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/user/zephyr_project/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: blackpill_f411ce
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.4 (/home/user/project/actions-runner/_work/github-actions-demo/github-actions-demo/zephyr-sdk)
-- Found toolchain: zephyr 0.16.4 (/home/user/project/actions-runner/_work/github-actions-demo/github-actions-demo/zephyr-sdk)
-- Found Dtc: /home/user/project/actions-runner/_work/github-actions-demo/github-actions-demo/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/user/zephyr_project/zephyr/boards/arm/blackpill_f411ce/blackpill_f411ce.dts
-- Found devicetree overlay: /home/user/project/hello_world/hello_world/boards/blackpill_f411ce.overlay
-- Found devicetree overlay: /home/user/zephyr_project/zephyr/../bootloader/mcuboot/boot/zephyr/app.overlay
-- Generated zephyr.dts: /home/user/project/hello_world/hello_world/build/mcuboot/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/user/project/hello_world/hello_world/build/mcuboot/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/project/hello_world/hello_world/build/mcuboot/zephyr/dts.cmake

warning: UART_INTERRUPT_DRIVEN (defined at boards/shields/sparkfun_sara_r4/Kconfig.defconfig:24,
boards/shields/wnc_m14a2a/Kconfig.defconfig:17, soc/arm/quicklogic_eos_s3/Kconfig.defconfig:17,
drivers/serial/Kconfig:72) was assigned the value 'y' but got the value 'n'. Check these unsatisfied
dependencies: ((NETWORKING && SHIELD_SPARKFUN_SARA_R4) || (NETWORKING && SHIELD_WNC_M14A2A) ||
(SERIAL && SOC_EOS_S3) || (SERIAL_SUPPORT_INTERRUPT && SERIAL)) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_UART_INTERRUPT_DRIVEN and/or look up
UART_INTERRUPT_DRIVEN in the menuconfig/guiconfig interface. The Application Development Primer,
[... 57 lines cut ...]
CMake Error at CMakeLists.txt:476 (align_up):
  align_up Function invoked with incorrect arguments for function named:
  align_up

-- Configuring incomplete, errors occurred!
CMake Error at cmake/modules/sysbuild_extensions.cmake:495 (message):
  CMake configure failed for Zephyr project: mcuboot

  Location: /home/user/zephyr_project/bootloader/mcuboot/boot/zephyr
Call Stack (most recent call first):
  CMakeLists.txt:42 (ExternalZephyrProject_Cmake)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/home/user/project/hello_world/hello_world/build -GNinja -DBOARD=blackpill_f411ce -Dmcuboot_DTC_OVERLAY_FILE=/home/user/project/hello_world/hello_world/boards/blackpill_f411ce.overlay -Dmcuboot_EXTRA_DTC_OVERLAY_FILE=/home/user/zephyr_project/zephyr//../bootloader/mcuboot/boot/zephyr/app.overlay -DEXTRA_CONF_FILE=/home/user/project/hello_world/hello_world/boards/witty_cloud.conf -S/home/user/zephyr_project/zephyr/share/sysbuild -DAPP_DIR:PATH=/home/user/project/hello_world/hello_world

```
Environment:
- Zephyr version: 3.5.0
- Zephyr SDK version: 0.16.4
- MCUBoot version: latest commit in manifest-rev branch, commit e9131ee8b8ab9306d8f6ad410fd1e3d8a820ce10
- Host OS: Ubuntu 22.04.3 LTS
    
 [Repo with this project](https://github.com/khrystynaO/hello-world)

Thank you for your attention to this matter. Please let me know if any additional information is required to address this bug.

nordicjm, 2024-01-09:
Could you try with https://github.com/zephyrproject-rtos/zephyr/pull/67323 ?

nordicjm, 2024-01-09:
Though that will allow it to build with a warning, it won't actually fix the issue which is an stm32 issue because the flash controller does not declare the erase size

nordicjm, 2024-01-09:
@erwango

vshymanskyy, 2024-01-10:
May I kindly ask you NOT to assign a `priority: low` to this?
This issue is quite annoying because it breaks MCUboot examples for many boards where it worked before.

nordicjm, 2024-01-10:
> May I kindly ask you NOT to assign a `priority: low` to this? This issue is quite annoying because it breaks MCUboot examples for many boards where it worked before.

This is already fixed in zephyr main for the above error, the remaining issue here is an stm32f4 issue whereby the flash controller does not specify the erase size which is for @erwango

vshymanskyy, 2024-01-10:
Ah, I see now that #67323 got merged. Thanks!

erwango, 2024-01-10:
> Though that will allow it to build with a warning, it won't actually fix the issue which is an stm32 issue because the flash controller does not declare the erase size

I'm fine to solve this, but would it be possible for clarity to close current issue and reopen a new one about the actual issue seen once #67323 is used?

erwango, 2024-01-11:
Closing. See previous comment

closed 2024-01-11 by erwango