zephyrproject-rtos/zephyr #67396: Error compiling C++ with flag (-Wmissing-field-initializers)

yperess, 2024-01-09:
**Describe the bug**
When using `-Wmissing-field-initializers` to compile Zephyr we get and error for SYS_INIT since it doesn't initialize the `dev` field

**To Reproduce**
Steps to reproduce the behavior:
1. Add `-Wmissing-field-initializers` to the compile options and build any app with SYS_INIT

**Expected behavior**
Should build

**Impact**
Breaks build

yperess, 2024-01-31:
Fixed in #68125

marc-hb, 2024-02-25:
Re-opening because this flag  broke again:
- #69419

If this policy is not formally agreed and regularly enforced in CI then it will keep breaking. From https://github.com/zephyrproject-rtos/zephyr/pull/68118#issuecomment-1910476735

> @barnas-michal not related to this being the right way to "fix" or not, but we should not be fixing issues of compiler warnings only enabled outside of the project, if this compiler flag is useful we should enable it in zephyr as well and fix issues everywhere and deal with it within the project for all toolchains and support this accordingly.

More relevant discussion in #68118

marc-hb, 2024-06-16:
init.h fixed for good in a portable way in:
- #69486 

Solution thoroughly documented in:
- #74047

Note that new documentation does NOT say anything about `-Wmissing-field-initializers`, which is not required by any C or C++ version. Both languages guarantee that missing fields will be initialized to zero (or equivalent).

@ceolin is `-Wmissing-field-initializers` something MISRA requires?

closed 2024-06-16 by marc-hb