zephyrproject-rtos/zephyr #67405: cannot flash hello_world on stm32f303k8

spectrum70, 2024-01-09:
At least as the involved mcu, issue seems the same of  https://github.com/zephyrproject-rtos/zephyr/issues/54117

I tested succesfully blinky on a small custom board with stm32f3030k8
I can flash it properly by "west flash", using stlink v2 pod.

As soon as i enable serial console in the config, i cannot flash anymore

**Logs and console output**
`
west build -p always -b ks_zion samples/hello_world
...
[95/96] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       12848 B        64 KB     19.60%
             RAM:        4088 B        12 KB     33.27%
             CCM:          0 GB         8 KB      0.00%
        IDT_LIST:          0 GB         2 KB      0.00%
Generating files from /home/angelo/dev-baylibre/zephyrproject/zephyr/build/zephyr/zephyr.elf for board: ks_zion
[96/96] cd /home/angelo/dev-baylibre/zephyrproject/zephyr/.../dev-baylibre/zephyrproject/zephyr/build/zephyr/zephyr.elf
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/angelo/dev-baylibre/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00726-ge2b6f5655 (2023-11-15-11:07)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : clock speed 1000 kHz
Info : STLINK V2J43S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.223669
Info : [stm32f3x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f3x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f3x.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32f3x.cpu       hla_target little stm32f3x.cpu       halted

Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : device id = 0x10016438
Info : flash size = 64kbytes
Warn : Adding extra erase range, 0x08003230 .. 0x080037ff
Error: error writing to flash at address 0x08000000 at offset 0x00000000
auto erase enabled
FATAL ERROR: command exited with status 1: /opt/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/angelo/dev-baylibre/zephyrproject/zephyr/boards/arm/ks_zion/support -s /opt/zephyr-sdk-0.16.4/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/angelo/dev-baylibre/zephyrproject/zephyr/boards/arm/ks_zion/support/openocd.cfg '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase /home/angelo/dev-baylibre/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
`

 - ...

**To Reproduce**
west flash

**Expected behavior**
complete flash write

**Environment (please complete the following information):**
 - Linux
 - Toolchain: Zephyr SDK, 
 - Commit SHA or Version used, master 8f6be9661e64987404f2356b536e1918f173d08f

**Additional context**
If i disable UART from board defconfig, i can flash properly
Upgraded the stlink firmware to last available, same issue
if i copy the zephyr.bin away, and flesh it by 
`
#!/bin/sh
openocd -f interface/stlink-v2.cfg -f ./openocd.cfg -c "program zephyr.bin 0x08000000 verify reset exit"
`
It works, i can flash the binary, and can see the console working.

erwango, 2024-01-11:
@ajarmouni-st ^^

spectrum70, 2024-01-11:
Thanks, i solved for now as done in the above 54117, using pyocd runner, it works fine.

erwango, 2024-01-16:
@ajarmouni-st Can you check we can use `source [find target/stm32f3x.cfg]` instead of `source [find board/st_nucleo_f3.cfg]` could work on nucleo_f303k8 on our side as well ?

FRASTM, 2024-02-06:
@spectrum70 is this issue closed from your POV ?