zephyrproject-rtos/zephyr #67406: doc: Documentation issue in 'develop/getting_started/index', macOS requires wget from homebrew

thompsa, 2024-01-09:
**Describe the bug**

When installing the Zephyr SDK in the second part of the getting started doc on macOS it requires wget to be installed for setup.sh, add this to the earlier homebrew install as a dependancy.

```
diff --git a/doc/develop/getting_started/index.rst b/doc/develop/getting_started/index.rst
index d95651121b..c2974509ca 100644
--- a/doc/develop/getting_started/index.rst
+++ b/doc/develop/getting_started/index.rst
@@ -132,7 +132,7 @@ The current minimum required version for the main dependencies are:
 
          .. code-block:: bash
 
-            brew install cmake ninja gperf python3 ccache qemu dtc libmagic
+            brew install cmake ninja gperf python3 ccache qemu dtc libmagic wget
 
       #. Add the Homebrew Python folder to the path, in order to be able to
          execute ``python`` and ``pip`` as well ``python3`` and ``pip3``.
```

**Environment**

* Page: `develop/getting_started/index`
* Version: main
* SHA-1: e91040917c4ea23008df0b544f3837705086985b

kartben, 2024-01-09:
Thanks for taking the time to report this! Guess we haven't tested these instructions on a vanilla macOS install in a long time :)

Could you please submit the fix as an actuall pull request? You may be interested in first checking out the [project's contribution guidelines](https://docs.zephyrproject.org/latest/contribute/guidelines.html) 

Thanks again!

thompsa, 2024-01-10:
I have added it as https://github.com/zephyrproject-rtos/zephyr/pull/67411 @kartben, thanks.

PR #67411 (2024-02-28): doc: wget is required for installing Zephyr SDK on macOS
When installing the Zephyr SDK in the second part of the getting started doc on macOS it requires wget to be installed for setup.sh, add this to the earlier homebrew install as a dependency.

Fixes #67406
closed 2024-02-28 by nashif