zephyrproject-rtos/zephyr #67420: Sample example for zperf application is not working

selvakumarm-eaton, 2024-01-10:
Test zperf sample application with nucleo_h743zi board.
Follow the steps specified this link:  https://github.com/zephyrproject-rtos/zephyr/issues/24770

Set IP Address in zephyrproject/zephyr/samples/net/zperf/prj.conf in my case:

CONFIG_NET_CONFIG_MY_IPV4="192.168.76.20"
CONFIG_NET_CONFIG_PEER_IPV4="192.168.76.11"

On the host (IP 192.168.76.11) run iperf:
iperf -s -p 5001 -B 192.168.76.11

No response on the Zephyr target.

henrikbrixandersen, 2024-01-10:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

jukkar, 2024-01-10:
You could also set the gateway address `CONFIG_NET_CONFIG_MY_IPV4_GW="192.168.76.xx"` just in case although as both the device and peer are in the same subnet, setting it should not matter.

jukkar, 2024-01-10:
> iperf -s -p 5001 -B 192.168.76.11

You are running iperf in server mode, you need to initiate the test from zephyr side to continue.

jukkar, 2024-01-12:
This looks like a configuration issue. @selvakumarm-eaton if you run Linux iperf in server mode (`-s` option), then you need to start the communication in zephyr side manually from zperf-shell using the `upload` command.
Closing this, reopen if needed.

closed 2024-01-12 by jukkar