zephyrproject-rtos/zephyr #67425: GD32 timer don't work with GD32V

soburi, 2024-01-10:
This driver depends NVIC_SetPendingIRQ in ARM architecture,
but Nuclei's RISC-V core behave not same.

Several improvement strategies are possible.

1. Use CLIC's irq-pending function.

From what I've tried, RISC-V (Nuclei core) also has a pending function, but it doesn't seem to be the same as ARM.
I think it will be difficult to implement this way.

2. Implementation that does not depend on pending

I made this driver based on another SoC's driver, so I followed the idea of using irq-pending, but it may be possible to do it without doing that.
Irq-pending is used here to generate an interrupt triggered by software, but it may be possible to use a software event generation register instead.

henrikbrixandersen, 2024-01-16:
@gmarull Was this fixed in #67147?

gmarull, 2024-01-16:
No

closed 2024-03-31 by github-actions[bot]

cameled, 2024-04-02:
refer https://people.eecs.berkeley.edu/~krste/papers/riscv-privileged-v1.9.1.pdf

3.1.13 Machine Interrupt Registers (mip and mie)

```
The UTIP, STIP and
HTIP bits may be written by M-mode software to deliver timer interrupts to lower privilege levels.
User, supervisor and hypervisor software may clear the UTIP, STIP and HTIP bits with calls to
the AEE, SEE, or HEE, respectively.
```