zephyrproject-rtos/zephyr #67438: benchamrks: latency_measure: benchmark.user.latency.* fails

katgiadla, 2024-01-10:
**Describe the bug**
The tests:
- `benchmark.user.latency`
- `benchmark.user.latency.objcore.stats`
- `benchmark.user.latency.timeslicing`

fail.

Observed for:
- `nrf52dk_nrf52832`
- `nrf52840dk_nrf52840`
- `nrf5340dk_nrf5340_cpuapp`
- `nrf5340dk_nrf5340_cpuapp_ns`
- `nrf5340dk_nrf5340_cpunet`
- `nrf9160dk_nrf9160`
- `nrf9160dk_nrf9160_ns`

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf52840dk_nrf52840` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/benchmarks/latency_measure -p nrf52840dk_nrf52840 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-3895-ga27ae686d3b7 ***
THREAD yield.preemptive.ctx.(K -> K)                :    1317 cycles ,    20578 ns : 
THREAD yield.preemptive.ctx.(U -> U)                :    1768 cycles ,    27638 ns : 
THREAD yield.preemptive.ctx.(K -> U)                :    1637 cycles ,    25584 ns : 
THREAD yield.preemptive.ctx.(U -> K)                :    1448 cycles ,    22632 ns : 
THREAD yield.cooperative.ctx.(K -> K)               :    1317 cycles ,    20578 ns : 
THREAD yield.cooperative.ctx.(U -> U)               :    1768 cycles ,    27638 ns : 
THREAD yield.cooperative.ctx.(K -> U)               :    1637 cycles ,    25584 ns : 
THREAD yield.cooperative.ctx.(U -> K)               :    1448 cycles ,    22632 ns : 
ISR resume.interrupted.thread.kernel                :     355 cycles ,     5546 ns : 
ISR resume.different.thread.kernel                  :    1311 cycles ,    20499 ns : 
ISR resume.different.thread.user                    :    1642 cycles ,    25663 ns : 
THREAD create.kernel.from.kernel                    :    1371 cycles ,    21437 ns : 
THREAD start.kernel.from.kernel                     :    1392 cycles ,    21765 ns : 
THREAD suspend.kernel.from.kernel                   :    1384 cycles ,    21640 ns : 
THREAD resume.kernel.from.kernel                    :    1313 cycles ,    20531 ns : 
THREAD abort.kernel.from.kernel                     :    2708 cycles ,    42328 ns : 
THREAD create.user.from.kernel                      :    1267 cycles ,    19804 ns : 
THREAD start.user.from.kernel                       :   10693 cycles ,   167085 ns : 
THREAD suspend.user.from.kernel                     :    1788 cycles ,    27945 ns : 
THREAD resume.user.from.kernel                      :    1634 cycles ,    25538 ns : 
THREAD abort.user.from.kernel                       :    2605 cycles ,    40710 ns : 
THREAD create.user.from.user                        :    2445 cycles ,    38218 ns : 
THREAD start.user.from.user                         :   11075 cycles ,   173062 ns : 
THREAD suspend.user.from.user                       :    2080 cycles ,    32515 ns : 
THREAD resume.user.from.user                        :    1984 cycles ,    31015 ns : 
THREAD abort.user.from.user                         :    3219 cycles ,    50312 ns : 
THREAD start.kernel.from.user                       :    1816 cycles ,    28382 ns : 
THREAD suspend.kernel.from.user                     :    1717 cycles ,    26835 ns : 
THREAD resume.kernel.from.user                      :    1697 cycles ,    26523 ns : 
THREAD abort.kernel.from.user                       :    3354 cycles ,    52413 ns : 
FIFO put.immediate.kernel                           :     253 cycles ,     3953 ns : 
FIFO get.immediate.kernel                           :     195 cycles ,     3062 ns : 
FIFO put.alloc.immediate.kernel                     :    1052 cycles ,    16437 ns : 
FIFO get.free.immediate.kernel                      :     702 cycles ,    10968 ns : 
FIFO put.alloc.immediate.user                       :    1766 cycles ,    27594 ns : 
FIFO get.free.immediate.user                        :    1406 cycles ,    21968 ns : 
FIFO get.blocking.(K -> K)                          :    1582 cycles ,    24718 ns : 
FIFO put.wake+ctx.(K -> K)                          :    1654 cycles ,    25843 ns : 
FIFO get.free.blocking.(K -> K)                     :    1576 cycles ,    24625 ns : 
FIFO put.alloc.wake+ctx.(K -> K)                    :    1691 cycles ,    26421 ns : 
FIFO get.free.blocking.(U -> K)                     :    2050 cycles ,    32031 ns : 
FIFO put.alloc.wake+ctx.(K -> U)                    :    2153 cycles ,    33640 ns : 
FIFO get.free.blocking.(K -> U)                     :    2035 cycles ,    31796 ns : 
FIFO put.alloc.wake+ctx.(U -> K)                    :    2175 cycles ,    33984 ns : 
FIFO get.free.blocking.(U -> U)                     :    2490 cycles ,    38906 ns : 
FIFO put.alloc.wake+ctx.(U -> U)                    :    2620 cycles ,    40937 ns : 
LIFO put.immediate.kernel                           :     253 cycles ,     3953 ns : 
LIFO get.immediate.kernel                           :     195 cycles ,     3062 ns : 
LIFO put.alloc.immediate.kernel                     :    1049 cycles ,    16390 ns : 
LIFO get.free.immediate.kernel                      :     701 cycles ,    10953 ns : 
LIFO put.alloc.immediate.user                       :    1786 cycles ,    27906 ns : 
LIFO get.free.immediate.user                        :    1408 cycles ,    22000 ns : 
LIFO get.blocking.(K -> K)                          :    1577 cycles ,    24640 ns : 
LIFO put.wake+ctx.(K -> K)                          :    1646 cycles ,    25718 ns : 
LIFO get.free.blocking.(K -> K)                     :    1598 cycles ,    24984 ns : 
LIFO put.alloc.wake+ctx.(K -> K)                    :    1695 cycles ,    26499 ns : 
LIFO get.free.blocking.(U -> K)                     :    2068 cycles ,    32323 ns : 
LIFO put.alloc.wake+ctx.(K -> U)                    :    2149 cycles ,    33591 ns : 
LIFO get.free.blocking.(K -> U)                     :    2047 cycles ,    31984 ns : 
LIFO put.alloc.wake+ctx.(U -> K)                    :    2194 cycles ,    34281 ns : 
LIFO get.free.blocking.(U -> U)                     :    2530 cycles ,    39531 ns : 
LIFO put.alloc.wake+ctx.(U -> U)                    :    2658 cycles ,    41531 ns : 
EVENTS post.immediate.kernel                        :     277 cycles ,     4328 ns : 
EVENTS set.immediate.kernel                         :     275 cycles ,     4297 ns : 
EVENTS wait.immediate.kernel                        :     215 cycles ,     3359 ns : 
EVENTS wait_all.immediate.kernel                    :     212 cycles ,     3312 ns : 
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x40005558
E: r0/a1:  0x006db747  r1/a2:  0x00000000  r2/a3:  0x00002710
E: r3/a4:  0x40005000 r12/ip:  0x00000000 r14/lr:  0x00054ad3
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x0005689a
E: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0
E: Current thread: 0x20040270 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA or Version used: zephyr-v3.5.0-3769-ga006ad5399c4

golowanow, 2024-01-18:
also observed on `npcx9m6f_evb` for `tests/benchmarks/latency_measure/benchmark.user.latency.timeslicing`
```
EVENTS wait_all.immediate.kernel                    :     149 cycles ,     9933 ns :
E: ***** MPU FAULT *****
E:   Data Access Violation
E:   MMFAR Address: 0x200c0b18
E: r0/a1:  0x004aec72  r1/a2:  0x00000000  r2/a3:  0x00002710
E: r3/a4:  0x200c2f4f r12/ip:  0x00000000 r14/lr:  0x10094237
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x10094170
E: >>> ZEPHYR FATAL ERROR 19: Unknown error on CPU 0
E: Current thread: 0x200c0a50 (unknown)
E: Halting system
```

golowanow, 2024-01-18:
@katgiadla please try on your setup with reversed commit  a8914c56

peter-mitsis, 2024-01-19:
I think I figured why this one is occurring.

The exception occured inside the routine SystemCoreClockUpdate(), which is called from z_arm_dwt_freq_get(), which is called from arch_timing_freq_get(), which is called from PRINT_STATS_AVG() which is in a user thread. Unfortunately, this is an MMIO access and the user thread does not have permissions to for it.

The solution is to re-jigger the test a little bit so that the PRINT_STATS_AVG() is invoked by another thread that is running at  kernel/supervisor level.

PR #67850 (2024-01-23): tests: latency_measure: Fix MPU fault
Fixes an MPU fault that was occurring on some ARM platforms when executing the events portion of the latency_measure benchmark. The fault was tracked to an attempt by a user thread to read an MMIO register that it did not have permissions to access.

The solution was to move the printing of the stats averages out of the test thread and into the main thread as the main thread always executes at kernel/supervisor level and will always have permissions to read that MMIO register.

Fixes #67438
commit 7c5a57b8 (2024-01-23): tests: latency_measure: Fix MPU fault

Fixes an MPU fault that was occurring on some ARM platforms when
executing the events portion of the latency_measure benchmark. The
fault was tracked to an attempt by a user thread to read an MMIO
register that it did not have permissions to access.

The solution was to move the printing of the stats averages out of
the test thread and into the main thread as the main thread always
executes at kernel/supervisor level and will always have permissions
to read that MMIO register.

Fixes #67438
closed 2024-01-23 by carlescufi