zephyrproject-rtos/zephyr #67460: West should default to no-tags and initial shallow fetch

fabiobaltieri, 2024-01-10:
## Introduction

West defaults to fetching full history and tags for all modules, which is slow and results in a big repository. The proposal is to change the default to do a shallow fetch with no tags.

### Problem description

A default upstream Zephyr checkout (`west init; west update`) at this point in time is about 5GB, and includes all history of all modules. Practically though, I suspect that a random user does not need the full module history, someone working on a module project or vendor may only need the history for some selected modules, and only people working on the core OS may want a full repository.

The difference between doing a shallow or full fetch on a not too powerful machine is significant, especially for the fetch time, for example on my laptop:

- `west update` takes about **4 minutes** and results in a 5GB repository
- `west update -o=--depth=1 -n` takes **2 minutes** and results in a 4.2GB repository

Note that this is just the time for fetching the modules, the main upstream `zephyr` repo is already in place, and it takes about 920MB, but this also means that there would be even more time saved if west was running on a project and doing a shallow fetch of the main zephyr repo as well.

### Proposed change

Change west to:
- do not fetch [tags](https://github.com/zephyrproject-rtos/west/blob/main/src/west/app/project.py#L902) by default (right now it always does it, which does not quite make sense with shallow fetches).
- detect if the module is being fetched for the first time, and default to do a shallow fetch, unless it's requested to initialize new modules with full history (currently fetch uses the same flag regardless, clone is not used).
- if the module has been fetched already, do a normal fetch, so one could just un-shallow a repository using the standard git tooling (`git fetch --unshallow`) and then west would not create any more grafted commits, this also means that updates of all modules are always going to have history from the initial fetch.

### Concerns and Unresolved Questions

- Why does west default to [always request tags](https://github.com/zephyrproject-rtos/west/blob/main/src/west/app/project.py#L1440) right now? git default is to only fetch them if they are part of the history, I'm suspecting that changing the default may break some use case
- Is this even worth it? Changing the default behavior may be confusing for some users, break some use case. The difference in space is not as big as I hoped, the difference in time is significant so it may help the perception of project being heavy, but I'm not too sure it's worth the change in behavior.

fabiobaltieri, 2024-01-10:
cc @mbolivar-ampere @henrikbrixandersen @nashif @carlescufi

nordicjm, 2024-01-11:
> do not fetch [tags](https://github.com/zephyrproject-rtos/west/blob/main/src/west/app/project.py#L902) by default (right now it always does it, which does not quite make sense with shallow fetches).

Not so sure on this, there are manifests that use tags for revisions which should not be broken, and the tag can also be used for versioning i.e. zephyr boot messages

fabiobaltieri, 2024-01-11:
> Not so sure on this, there are manifests that use tags for revisions which should not be broken, and the tag can also be used for versioning i.e. zephyr boot messages

Fetching by tag seems to work regardless (changed a project revision to a tag, deleted the module, fetched with `west update -n`, works fine), guess it's resolved by the remote, git describe would definitely break though, good point.