zephyrproject-rtos/zephyr #67467: power-domain-gpio **must** have configurable init priority

GrantRolls, 2024-01-11:
Current implementation of `power-domain-gpio` does not have a configurable init priority. It is hardcoded to 75.

This makes it incompatible with any device drivers that have a lower priority.

The init priority of `power-domain-gpio` should be configurable via Kconfig

[image]

henrikbrixandersen, 2024-01-16:
@GrantRolls Could you submit a PR for fixing this?

GrantRolls, 2024-01-16:
> @GrantRolls Could you submit a PR for fixing this?

Yes, I can try get around to this sometime in the next week

ceolin, 2024-01-22:
I have created a pr adding option to change init priority in all available domains

GrantRolls, 2024-01-23:
Thanks for that. 

Closing this regardless because I didn't realise it had already been resolved to some degree with the common config option. I'm running an older version.

closed 2024-01-23 by GrantRolls