zephyrproject-rtos/zephyr #67474: it8xxx2_evb: kernel: tests/kernel/timer/timer_api/kernel.timer.tickless Failed

pkunieck, 2024-01-11:
**Describe the bug**
During test run on it8xxx2_evb platform, there is a Fail test case: test_sleep_abs. This is output from git bisect:
7a5fcb8c60e2e6ba26704fabc28fe2d4e0106513 is the first bad commit
commit 7a5fcb8c60e2e6ba26704fabc28fe2d4e0106513
Author: Keith Packard <keithp@keithp.com>
Date:   Tue Sep 12 23:51:33 2023 -0700

    libc/picolibc: Support 'long long' and 'minimal' printf variants

**To Reproduce**
Steps to reproduce the behavior:
1. cd <path/to>zephyrproject/zephyr
2. source zephyr-env.sh
3. twister -p it8xxx2_evb --device-testing --hardware-map <path/to>/map.yaml -T tests/kernel/timer/timer_api/ -v

**Expected behavior**
Test pass

**Impact**
annoyance

**Logs and console output**
START - test_sleep_abs
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_api/src/main.c:807: timer_api_test_sleep_abs: (late >= 0 && late <= MAX(2, k_us_to_ticks_ceil32(250)) is false)
expected wakeup at 157, got 160 (late 3)
FAIL - test_sleep_abs in 0.034 seconds
___________________________________________________
Other test cases pass:
PASS - [timer_api.test_time_conversions] duration = 0.643 seconds
PASS - [timer_api.test_timeout_abs] duration = 0.003 seconds
PASS - [timer_api.test_timer_duration_period] duration = 0.327 seconds
PASS - [timer_api.test_timer_expirefn_null] duration = 0.327 seconds
PASS - [timer_api.test_timer_k_define] duration = 0.703 seconds
PASS - [timer_api.test_timer_period_0] duration = 0.203 seconds
PASS - [timer_api.test_timer_period_k_forever] duration = 0.203 seconds
PASS - [timer_api.test_timer_periodicity] duration = 0.254 seconds
PASS - [timer_api.test_timer_remaining] duration = 0.052 seconds
PASS - [timer_api.test_timer_restart] duration = 0.277 seconds
PASS - [timer_api.test_timer_status_get] duration = 0.002 seconds
PASS - [timer_api.test_timer_status_get_anytime] duration = 0.327 seconds
PASS - [timer_api.test_timer_status_sync] duration = 0.355 seconds
PASS - [timer_api.test_timer_user_data] duration = 0.303 seconds

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 16.4

henrikbrixandersen, 2024-01-16:
@keith-packard Any ideas?

Dino-Li, 2024-01-19:
@Chenhongren Can you take a look at this issue?

Chenhongren, 2024-01-22:
hi @pkunieck,

The it8xxx2 chip needs more time to resume from the low-power mode and the default ```CONFIG_PM``` for it8xxx2_evb is disabled. However, it's enabled in the ```prj_tickless.conf```. We would disable PM for the specific platform(it8xxx2_evb) in ```testcase.yml``` to address this issue.([disable-pm-for-it8xxx2-evb](https://github.com/Chenhongren/zephyr/tree/pr-disable-pm-for-timer-api))
Is this okay for you?

pkunieck, 2024-03-15:
This PR: #69891 seems to be working, tests pass with those changes

PR #69891 (2024-03-15): drivers: timer: it8xxx2: fix the timer drift issue
This commit enhances the accuracy of hardware cycle calculation before setting the IT8xxx2 event timer. The next target cycle is calculated by the last, elapsed, and expected timeout ticks. And then, the difference in hardware cycles between the target cycle and the current cycle is set into the event timer. This increased accuracy effectively resolves the clock drift issue.

Tested with:
west build -p always -b it8xxx2_evb tests/kernel/timer/timer_api -T kernel.timer.tickless
west build -p always -b it8xxx2_evb tests/kernel/timer/timer_behavi [... cut]
commit 765a4eb0 (2024-03-15): drivers: timer: it8xxx2: enhances the accuracy of hw cycle calculation

This commit enhances the accuracy of hardware cycle calculation before
setting the IT8xxx2 event timer. The next target cycle is calculated by
the last, elapsed, and expected timeout ticks. And then, the difference
in hardware cycles between the target cycle and the current cycle is set
into the event timer. This increased accuracy effectively resolves the
clock drift issue.

Tested with:
west build -p always -b it8xxx2_evb tests/kernel/timer/timer_api
     -T kernel.timer.tickless
west build -p always -b it8xxx2_evb tests [... cut]
closed 2024-03-15 by nashif