zephyrproject-rtos/zephyr #67481: Map device driver instances to power states disabling power to the devices

hubertmis, 2024-01-11:
**The problem**

When a device driver activates a device, it should ensure that the power to the device won't be cut off. To do this, the device driver forbids the power management subsystem from entering the power states which would result in disabling power to the activated device. This can done with the following functions:

https://github.com/zephyrproject-rtos/zephyr/blob/7d7f7aaf03fd0728d298dce22c20cb3098c3d0aa/include/zephyr/pm/policy.h#L114
https://github.com/zephyrproject-rtos/zephyr/blob/7d7f7aaf03fd0728d298dce22c20cb3098c3d0aa/include/zephyr/pm/policy.h#L125

A device driver is expected to be a generic implementation for a given device. A single device may be integrated with different SoCs, by different vendors, or with different boards. Each of them can support other power states, and other power states may result in disabling power for this device. Moreover, in a single SoC, there can be multiple instances of a single device type (using the same driver) and each of these instances can be powered off in other power states.

The device driver implementation should be agnostic to the device integration environment, and because of this it shall not have hardcoded power state and sub-state values which are required in the `void pm_policy_state_lock_get(enum pm_state state, uint8_t substate_id);` function.

**The solution**

I would like to have a mapping between device instances and power states that disable power to a given device. I would like to express it in the device tree reusing already existing devices and power state nodes, like:

```
power-states {
                idle: system_on_idle {
                        compatible = "zephyr,power-state";
                        power-state-name = "suspend-to-idle";
                        min-residency-us = <100>;
                };

                suspend_to_ram: s2ram {
                        compatible = "zephyr,power-state";
                        power-state-name = "suspend-to-ram";
                        min-residency-us = <10000>;
                        exit-latency-us = <1000>;
                };
};

a_peripheral_bus {
        uart0: uart@8000 {
                compatible = "nordic,nrf-uarte";
                reg = <0x8000 0x1000>;
                interrupts = <8 NRF_DEFAULT_IRQ_PRIORITY>;
                power-states-disabling-power = <&suspend_to_ram>;
                status = "disabled";
        };

        uart1: uart@9000 {
                compatible = "nordic,nrf-uarte";
                reg = <0x9000 0x1000>;
                interrupts = <9 NRF_DEFAULT_IRQ_PRIORITY>;
                power-states-disabling-power = <>;  // This instance is powered on in all power states
                status = "disabled";
        };
};
```

With this mapping, the driver code could call a new power management function implemented as:

```
void pm_policy_device_power_lock_get(const struct *device)
{
    enum pm_state state;
    uint8_t substate_id;

    for (POWER_STATES_DISABLING_POWER_FOR_DEVICE(device, state, substate_id)) {
        pm_policy_state_lock_get(state, substate_id);
    }
}

void pm_policy_device_power_lock_put(const struct *device)
{
    enum pm_state state;
    uint8_t substate_id;

    for (POWER_STATES_DISABLING_POWER_FOR_DEVICE(device, state, substate_id)) {
        pm_policy_state_lock_put(state, substate_id);
    }
}
```

This way the driver gets decoupled from its device environment and the power tree in the product. The power tree is handled in the module dedicated to power management.

**The alternatives**
I considered using existing APIs, but it leads to hard to maintain macros and unnecessary coupling between the device drivers and the power tree in the products integrating the devices like:

```
#ifdef CONFIG_SOC_NRF...
    pm_policy_state_lock_get(1, 0);
    pm_policy_state_lock_get(1, 2);
    pm_policy_state_lock_get(2, 0);
#elif CONFIG_SOC_STM32...
    pm_policy_state_lock_get(0, 0);
#endif
```

I was also thinking about using "power domains" in the device tree, but I'm not familiar enough with this concept to include it in the proposal.

ceolin, 2024-02-28:
That makes a lot of sense, was thinking about it other day as well. Will work on PoC and move from there.

ceolin, 2024-03-14:
@hubertmis https://github.com/zephyrproject-rtos/zephyr/pull/70111 FYI !

Would mind take a look on that ? Nomenclature-wise is still a problem imho, so any feedback is welcome. Feature-wise it seems complete.

closed 2024-06-07 by nashif