zephyrproject-rtos/zephyr #67492: ws2812(worldsemi,ws2812-spi) does not work with blackpill_f401cc/blackpill_f411ce boards

ku3mich, 2024-01-11:
**Describe the bug**

ws2812 based LED blinking but not have correct color

**overlay**
```
&spi1 {
	led_strip: ws2812@0 {
		compatible = "worldsemi,ws2812-spi";
		reg = <0>;
		spi-max-frequency = <8000000>;
		frame-format = <32768>; /* SPI_FRAME_FORMAT_TI */

		chain-length = <5>;
		spi-one-frame = <0x70>;
		spi-zero-frame = <0x40>;
		color-mapping = <LED_COLOR_ID_GREEN LED_COLOR_ID_RED LED_COLOR_ID_BLUE>;
	};
};
```

**source**
```
...<skipped>...

#define STRIP_NODE DT_NODELABEL(led_strip)
#define STRIP_NUM_PIXELS DT_PROP(STRIP_NODE, chain_length)

struct led_rgb pixels[STRIP_NUM_PIXELS];

static const struct device *const leds = DEVICE_DT_GET(STRIP_NODE);

int main(void)
{
    memset(pixels, 0x00, sizeof(pixels));
    int dir = 1;

    while (1)
    {
        int rc = led_strip_update_rgb(leds, pixels, STRIP_NUM_PIXELS);

        if (pixels[0].r == 0 || pixels[0].r == 250)
        {
            dir = -dir;

            if (pixels[0].r == 0)
                pixels[0].r = 250;
            else
                pixels[0].r = 0;
        }

        pixels[0].r += 10 * dir;

        k_msleep(50);
    }

    return 0;
}

**Expected behavior**
LED fading in-out using red color

**Impact**
ws2812 driver

thedjnK, 2024-01-11:
Works fine here, you probably have this set wrong:
```
		color-mapping = <LED_COLOR_ID_GREEN LED_COLOR_ID_RED LED_COLOR_ID_BLUE>;
```

ku3mich, 2024-01-11:
and what is right for your opinion @thedjnK ? it is identical from sample/drivers )

ku3mich, 2024-01-11:
@thedjnK it is impossible to run it on stm32f103 because of
```
[00:00:00.000,000] <err> spi_ll_stm32: Frame Format TI not supported
```

thedjnK, 2024-01-11:
```
&spi2 {
	led_strip: ws2812@0 {
		compatible = "worldsemi,ws2812-spi";
		reg = <0>;
		spi-max-frequency = <4000000>;

		/* WS2812 */
		chain-length = <42>;
		spi-one-frame = <0x70>;
		spi-zero-frame = <0x40>;
		reset-delay = <50>;

		color-mapping = <LED_COLOR_ID_GREEN
				 LED_COLOR_ID_RED
				 LED_COLOR_ID_BLUE>;
	};
};
```
I've had this working for years on an stm32f103

simonguinot, 2024-01-12:
Hi  @ku3mich 

> **Describe the bug**
> 
> ws2812 based LED blinking but does not have correct color

Is it blue or green instead of red ? If yes, as said @thedjnK, your color mapping is probably wrong. Note that this depends on the WS2812 device model you are using.

ndrs-pst, 2024-01-15:
@ku3mich, Based on the information in the [WS2812 datasheet](https://cdn-shop.adafruit.com/datasheets/WS2812.pdf), I recommend aligning `spi-one-frame` and `spi-zero-frame` with the specifications outlined in the datasheet. As an initial step, I suggest testing with `spi-max-frequency = <4000000>;`.

[image]

closed 2024-03-31 by github-actions[bot]