zephyrproject-rtos/zephyr #67515: intel_adsp: cavs25 tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack* fails

majunkier, 2024-01-12:
**Describe the bug**
Below tests fails on intel_adsp_cavs25
```
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.no_pool.alloc.no_user	 	 	 	  	 
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.no_pool.alloc.user	 	 	 	  	 
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.no_user	 	 	 	  	 
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user 
```

**To Reproduce**
twister -p intel_adsp_cavs25 --device-testing --hardware-map <path_to_cavs.map> -T tests/kernel/threads/dynamic_thread_stack/ -v

**Expected behavior**
Test should pass.

**Logs and console output**
One of error, they are similar
```
Running TESTSUITE dynamic_thread_stack
===================================================================
START - test_dynamic_thread_stack_alloc
ASSERTION FAIL [!arch_mem_coherent(stack)] @ WEST_TOPDIR/os.rtos.zephyr.zephyr/kernel/thread.c:635
E:  ** FATAL EXCEPTION
E:  ** CPU 0 EXCCAUSE 63 (zephyr exception)
E:  **  PC 0xbe011677 VADDR 0
E:  **  PS 0x60920
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:9 CALLINC:2)
E:  **  A0 0xbe011338  SP 0xbe01fdb0  A2 0x4  A3 0x9e024fe4
E:  **  A4 0  A5 0x60520  A6 0x1  A7 0x1
E:  **  A8 0x1  A9 0x1 A10 0x1 A11 0x9e024fe4
E:  ** A12 0x9e024b40 A13 0x9e025074 A14 0x3 A15 0x10
E:  ** LBEG 0xbe016f6c LEND 0xbe016f77 LCOUNT 0
E:  ** SAR 0x18
Backtrace:0xbe011674:0xbe01fdb0 0xbe011335:0xbe01fdc0 0xbe013594:0xbe01fde0 0xbe013675:0xbe01fe20 0xbe010afd:0xbe01fe50 0xbe01222b:0xbe01fe90 0xbe011327:0xbe01feb0 
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x9e024b40 (test_dynamic_thread_stack_alloc)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: Ubuntu
 - Toolchain: Zephyr SDK 0.16.4
 - Zephyr version: zephyr-v3.5.0-4020-g0ee1e28a2f5f

ceolin, 2024-01-16:
https://github.com/zephyrproject-rtos/zephyr/pull/67699

majunkier, 2024-02-06:
Tests are failing on below version:
OS: Ubuntu
Toolchain: Zephyr SDK 0.16.5
Zephyr version: v3.6.0-rc1
```
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.no_user  FAILED tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.no_pool.alloc.user  FAILED 
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.no_pool.alloc.no_user  FAILED
tests/kernel/threads/dynamic_thread_stack/kernel.threads.dynamic_thread.stack.pool.alloc.user  FAILED 
```
similar error output : 
`E:  ** FATAL EXCEPTION
E:  ** CPU 2 EXCCAUSE 13 (load/store PIF data error)
E:  **  PC 0xbe0137a3 VADDR 0x18
E:  **  PS 0x60120
E:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:1 CALLINC:2)
E:  **  A0 0xbe01636f  SP 0xbe01ad00  A2 0x14  A3 0x60125
E:  **  A4 0  A5 0x9e024ea8  A6 0x60725  A7 0x5
E:  **  A8 0xbe013cc1  A9 0xbe01ace0 A10 0 A11 0x60325
E:  ** A12 0x9e024b42 A13 0x9e0250a0 A14 0x60525 A15 0
E:  ** LBEG 0xbe0114c5 LEND 0xbe0114db LCOUNT 0
E:  ** SAR 0xa
E:  **  THREADPTR 0
Backtrace:0xbe0137a0:0xbe01ad00 0xbe01636c:0xbe01ad20 0xbe016137:0xbe01ad40 0xbe016110:0xbe01ad60 0xbe010b86:0xbe01ad90 0xbe01226b:0xbe01add0 0xbe01134b:0xbe01adf0 
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 2`

PR #68311 (2024-02-09): tests: dynamic_thread_stack: Fix multicore and incoherent cache
When running it in a multicore and with incoherent cache environment it
is possible that the thread allocating dynamic stacks is switched to a
different cpu. In this situation further access to that memory (like
when releasing resources) will be invalid.

Fixes #67515
closed 2024-02-09 by nashif