zephyrproject-rtos/zephyr #67539: RT1060: SMP: WDOG: Image upload when hw wdog is enabled causes an unexpected reset

kunoh, 2024-01-12:
**Describe the bug**
When the HW Watchdog is enabled and the timeout is set to less than the time it takes to finish the image upload, the MCU resets into recovery mode (Boot ROM).
The usb device **1fc9:0135 NXP Semiconductors SE Blank RT Family** shows up and communication can be done using using sdphost.
To actually recovery the device and boot it up in the user application, a manual reset of the soc have to be done by pressing the reset button.

- What target platform are you using?
MIMXTR1060

**To Reproduce**
Steps to reproduce the behavior:
1. Enable SMP Image Upload
2. Enable HW Watchdog
3. Timeout should be less than the time it takes to upload the image
4. Start uploading using a mcumgr client (Tested with https://github.com/vouch-opensource/mcumgr-client)
5. The reset should occur anytime now

**Expected behavior**
I expected the hwdog an the action of uploading an image to be able to coexist or in worst case the hw wdog would do a reset and boot the user application again.
I did not expect the mcu to boot up in recovery mode.

**Impact**
The mcu may enter and get stuck in recovery mode until a manual reset is done.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.3
 - Commit SHA or Version used: 3.5.0

dleach02, 2024-01-16:
I understand the expected behavior but what should be the watchdog value if the image upload doesn't complete either? Feels like the original application should set a timeout value that considers worse case scenarios of usage which in this case should be the worse case timing for image upload.

kunoh, 2024-01-17:
My concern is, even though the timeout is set considering the image upload - if for whatever reason the upload just takes a couple of seconds more than it usually does and the hw wdog triggers - we enter a very bad state.

Under normal circumstances the hw wdog just does a system reset and the application is rebooted, but when it happens during image upload we enter the recovery mode and is basically stuck there until one resets the mcu again.

DerekSnell, 2024-01-19:
Hi @kunoh ,
I feel you have two separate topics combined in this issue: one about how the SMP application is impacted by enabling a watchdog; the other about why the RT1060 enters serial downloader mode after reset in this scenario.

For the RT1060 reset, the ROM bootloader executes after reset, which manages the boot options.  You report after the watchdog reset, the bootloader enters serial downloader mode instead of rebooting your application.  I suspect the reason for this is because the flash was performing an erase/programming operation when the RT1060 resets.  The flash cannot be accessed during those operations.  Attempting to read from the flash during those operations will not succeed.  I believe the watchdog resets, the bootloader attempts to boot from flash, the flash reads do not succeed, the bootloader cannot find a valid image in flash, and as a fallback it enters serial downloader mode.  When you manually reset your board with the reset button, by that time the flash operation completed, and the bootloader is able to read the flash and boot as you expect.

For production the RT1060 bootloader can be configured with fuses to deal with this scenario.  One example, you could program the `WDOG_ENABLE` fuse.  That fuse enables a timeout during serial downloader mode.  If no downloader activity is detected before the watchdog times out, a software reset is triggered, and the bootloader will start over again.

As for your SMP topic, personally I feel you should convert this to an Enhancement request for the SMP application/service: to improve it to work better when a watchdog is enabled.  But we are still discussing the best approach for that enhancement.

Best regards

dleach02, 2024-01-19:
@nordicjm, @de-nordic, can you review the details in this issue? 

As @DerekSnell mentioned, we discussed this internally and are thinking that the service here probably needs to consider watchdog servicing when uploading a new image.

kunoh, 2024-01-22:
Hi @DerekSnell ,

Yes, you are right - this should probably be split into two different topics.

I will wait for @nordicjm and @de-nordic reply before opening a new Enhancement request.

DerekSnell, 2024-01-26:
Hi @kunoh ,
The boot behavior of the RT1060 is not an errata.  It is behaving as expected.  And as I shared earlier, you can configure the ROM bootloader with fuses to avoid the scenario you ran into.
Best regards

closed 2024-01-26 by kunoh