zephyrproject-rtos/zephyr #67546: Intermittent Data Loss in UART Async API on STM32H753ZI (Nucleo H753ZI) ARM

durgababu2010, 2024-01-12:
**Title:** _Intermittent Data Loss in UART Async API on STM32H753ZI (Nucleo H753ZI)_
**Issue Description:**
> I am working with the STM32H753ZI (Nucleo H753ZI) board and using the UART asynchronous API provided by Zephyr. While I am able to receive data from the host, I am encountering an intermittent issue where the received data is not correctly processed. Occasionally, when reading the buffer to extract the packet, I observe that the data contains zeros, indicating a loss or misinterpretation of the received data.

**Environment:**
Board: STM32H753ZI (Nucleo H753ZI)
Zephyr Version: [3.5.0]
Toolchain: [zephyr-sdk-0.16.3]

Followed Steps:

1. Set up UART to receive data asynchronously and enabling the seamless transfer
2. Enabling the callback
3. sent the data from the host pc to board
4. In the rx_rdy handler, copy the data to a ring buffer and signal a semaphore to notify a packet processing thread.
5. The packet processing thread then reads from this buffer.

**Expected Behavior:**
The data read from the buffer should be consistent and accurate, reflecting the data sent from the host.

**Observed Behavior:**
The received data intermittently contains zeros, suggesting data loss or corruption. This happens randomly; sometimes the data is received and processed successfully, but other times it fails. The issue persists even though I am using the volatile qualifier.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir wsp && cd wsp
2. Extract the uart_dma_issue_nucleo_h753zi.zip(see attachement) into wsp folder
3. cd uart_dma_issue_nucleo_h753zi
4. python3 ./initial_setup.py(create the virutal environment and install necessary packages prescribed by zephyr).
5. update the C_Cpp.default.compilerPath and gdb path on .code-workspace file if your are using visual studio code
6. west build -b nucleo_h753zi -p always src
7. west flash
8. the main thread prints "main" for every 5 seconds
9. open any serial port application on the pc and send the follwing packet in hex
10. "AACC33550000000C000000010000000000000001"
11. if its success it will print "CS" other wise what garbage data received it will print(attached in image)

> I am seeking assistance in understanding and resolving this intermittent data loss issue with the UART async API on the STM32H753ZI board. Any insights or suggestions to debug this problem would be greatly appreciated.
[uart_dma_issue_nucleo_h753zi.zip](https://github.com/zephyrproject-rtos/zephyr/files/13919209/uart_dma_issue_nucleo_h753zi.zip)
[image]

erwango, 2024-01-12:
@durgababu2010 Would you mind trying enabling uart fifo ?
See https://github.com/zephyrproject-rtos/zephyr/blob/37158ebf15c8c299a1bc349bee206916842609f5/dts/bindings/serial/st%2Cstm32-uart-base.yaml#L90

erwango, 2024-01-12:
Alternatively, you could try https://github.com/zephyrproject-rtos/zephyr/pull/60909

durgababu2010, 2024-01-13:
> @durgababu2010 Would you mind trying enabling uart fifo ? See
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/37158ebf15c8c299a1bc349bee206916842609f5/dts/bindings/serial/st%2Cstm32-uart-base.yaml#L90

When working with the main branch of Zephyr, as opposed to the specific tag ([zephyr-v3.5.0]), I'm not encountering any issues, even without enabling the "fifo-enable" property. Everything appears to be functioning correctly. I'll conduct thorough testing and provide an update soon.

In the [zephyr-v3.5.0] tag, within the st,stm32-uart-base.yaml file, there's no property for FIFO enablement. The issue persists in this release tag. I'm currently investigating what changes might have resolved this issue in the main branch, or if the issue might still be present but not yet evident.

Desvauxm-st, 2024-01-16:
hello
I can't use your code about host sending data
I try your code and did some modifications to see if all the data we sent on tx uart pin pb4 have been received on pin pb3 .
I do like that:
plug a wire beetween CN7 PB4 and PB3 see after the picture 

[image]

i modify your code  src   ( send 41 bytes each 500ms  and check if all data are received well )see after the code  in zip file:

[src.zip](https://github.com/zephyrproject-rtos/zephyr/files/13948972/src.zip)

and the result seem ok see image after
[image]

[image]

when we have a look on tx:ABCD.....
and rx:ABCD... we received all the data send in tx.
sometime the rx is split in 2 time due to the time of reception.

for me it is ok.

durgababu2010, 2024-01-17:
> hello I can't use your code about host sending data I try your code and did some modifications to see if all the data we sent on tx uart pin pb4 have been received on pin pb3 . I do like that: plug a wire beetween CN7 PB4 and PB3 see after the picture
> 
> ![IMG_20240116_114324](https://private-user-images.githubusercontent.com/119042859/297021405-bda39dbb-c873-4a08-b565-fd2f56d1f2df.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzcyODgsIm5iZiI6MTcwNTQ3Njk4OCwicGF0aCI6Ii8xMTkwNDI4NTkvMjk3MDIxNDA1LWJkYTM5ZGJiLWM4NzMtNGEwOC1iNTY1LWZkMmY1NmQxZjJkZi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QwNzM2MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNGFkZWNhNWIyZGY0YThjNjQ2MjM1ZWFhMmExYTY1OTUzNDYxZWIzNmU1NzQxNjQ2NmNiYjBjZDgxNGMzNWM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EYAH_u0b09GUBo_vu8c-rDLbQk_eMWoUPgKlZ9uew_8)
> 
> i modify your code src ( send 41 bytes each 500ms and check if all data are received well )see after the code in zip file:
> 
> [src.zip](https://github.com/zephyrproject-rtos/zephyr/files/13948972/src.zip)
> 
> and the result seem ok see image after ![image](https://private-user-images.githubusercontent.com/119042859/297023065-a75b97b3-5610-441b-b71b-d7819a3bd565.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzcyODgsIm5iZiI6MTcwNTQ3Njk4OCwicGF0aCI6Ii8xMTkwNDI4NTkvMjk3MDIzMDY1LWE3NWI5N2IzLTU2MTAtNDQxYi1iNzFiLWQ3ODE5YTNiZDU2NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QwNzM2MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jMTVjYjhhNTlhNWMyMzBhNDQ4ZDQyYTRiZWJiNDkzMDY3ZWU0NDUzMTkxZDgxZjE5OTBiMDU1YmYzMDBiZmJkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UgeBaXoi6-aAs2agV3BuxRmGPeIen9183fJZd-jqkbM)
> 
> ![image](https://private-user-images.githubusercontent.com/119042859/297023215-863d1a85-6848-403a-8975-2c9812bd2160.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU0NzcyODgsIm5iZiI6MTcwNTQ3Njk4OCwicGF0aCI6Ii8xMTkwNDI4NTkvMjk3MDIzMjE1LTg2M2QxYTg1LTY4NDgtNDAzYS04OTc1LTJjOTgxMmJkMjE2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTE3JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExN1QwNzM2MjhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05ODJmNmEzZmM1ZTYyODRlY2Q2YjAzYzhkZmY4NDQ2ZTcxMjJlNzkzOTY3YjMzMDRjNzA5NjU1ZDk3OTFhYTcyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-CyzHKp5aWtWBdp8yfhVNYE8KQnIIPfgZ1mzP-pXoa0)
> 
> when we have a look on tx:ABCD..... and rx:ABCD... we received all the data send in tx. sometime the rx is split in 2 time due to the time of reception.
> 
> for me it is ok.

hello @Desvauxm-st 
i slightly changed the src.zip that have been attched to reproduce the issue i am attaching the images and videos

**changes done to the code**
**in main.c**
 1)sleep for 5 seconds

**in uart_comm.c**
1)used global array to store the received data 
2)print the data immediately in the event handler
3)raised the semaphore

**In host_if.c**
1)extern the global array
2)print the contents in the stored array

**for reference**
video 1: setup video
video 2: data trasmission (issue video)
image 1 ,2,3,4  :[updated_src.zip](https://github.com/zephyrproject-rtos/zephyr/files/13960111/updated_src.zip) 
image 5 : west.yml

[image]
[image]
[image]
[image]
[image]
[dat_tramssion.webm](https://github.com/zephyrproject-rtos/zephyr/assets/77288169/42456b05-39a2-4299-9a50-6bb65a7db50e)
[updated_src.zip](https://github.com/zephyrproject-rtos/zephyr/files/13960117/updated_src.zip)

https://github.com/zephyrproject-rtos/zephyr/assets/77288169/fe5329e1-c10d-4c79-a644-8bea61391353

Desvauxm-st, 2024-01-17:
@durgababu2010

see after the code src modified

[src.zip](https://github.com/zephyrproject-rtos/zephyr/files/13962082/src.zip)

see after the result:
[image]

seem ok no data loss

in fact be carefull :  array must be clear each time we enter in uart_event_handler()
i also add current_index to manage the index of the rx string in input before calling k_sem_give(&host_rx_sem);

durgababu2010, 2024-01-17:
> @durgababu2010
>
> see after the code src modified
>
> [src.zip](https://github.com/zephyrproject-rtos/zephyr/files/13962082/src.zip)
>
> see after the result: [...]

I want to express my appreciation for your dedicated efforts in addressing the issue. At the same time, I'd like to bring to your attention that the problem persists despite the changes shown in the attached image.

I truly value your commitment to resolving this matter. I have noticed that you implemented a signal trigger after receiving a certain 
**change_1**
[image]
**change_2**
[image]

**image_1**
[image]
**image_2**
[image]

changes done:
1)every time mem_set the global buffer(array) with 0 on receiving the interrupt irrespective of index
2)immediately raising the semaphore to signal for thread processing
3)changed the uart transmit for proper printing(included \n in printing)

Desvauxm-st, 2024-01-17:
@durgababu2010

if the goal is to receive in the thread all the data we sent you must use the code using **current_index** as given 
otherwise i don't understand your issue

and of course if you remove this part:
else{
                my_memcpy(&array[current_index],&evt->data.rx.buf[evt->data.rx.offset],evt->data.rx.len);
                current_index=current_index+evt->data.rx.len;
 }

it is normal that you lose the second part of the frame we sent.

erwango, 2024-01-29:
@durgababu2010 Any feedback ?

durgababu2010, 2024-01-30:
> @durgababu2010 Any feedback ?

Upon working with the latest version of Zephyr, as opposed to the 3.5.0 release version, 

I am pleased to report that I have encountered no issues over a continuous one-week test period. 

It is noteworthy that no significant changes were made to the code, except for the enabling of FIFO.

Thank you once again for your assistance.

erwango, 2024-01-30:
Ok, thanks. I'm closing the issue then.

closed 2024-01-30 by erwango