zephyrproject-rtos/zephyr #67582: doc: Documentation issue in 'kconfig', dead link to CONFIG_PREEMPT_ENABLED

mimvdb, 2024-01-14:
**Describe the bug**
https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_MAIN_THREAD_PRIORITY

CONFIG_MAIN_THREAD_PRIORITY refers to CONFIG_PREEMPT_ENABLED in the defaults, but this is a dead link.

**Environment**

* Page: `kconfig`
* Version: main
* SHA-1: fc942ef7d30dea3a5b4e0d9644cc9f586385ecf0

kartben, 2024-01-15:
@gmarull Should we make it so that invisible symbols don't get rendered, or should we have "very" invisible symbols actually show up in docs (right now a symbol such as ``PREEMPT_ENABLED`` does not show up for it has neither help nor prompt)?

gmarull, 2024-01-15:
what could be done is that such symbols do not get rendered as a link?

closed 2024-03-31 by github-actions[bot]