zephyrproject-rtos/zephyr #67583: Generate interrupt vectors and tables from devicetree

bjarki-trackunit, 2024-01-14:
## Introduction
The entire interrupt layout, including interrupt controllers, their interrupt lines, which lines are shared, and which lines are used, can be determined from the devicetree alone. Using the ordinals of interrupt controllers, interrupt generating devices, and the interrupt line numbers, we can synthesize symbols for every IRQ, including the handler and data passed to the handler, similar to encoding `struct device` with `__device_dts_ord_<ord>`.

This RFC details exactly how we can create a completely link optimizable and safe interrupt controller subsystem, which can be entirely stored in ROM, while natively supporting shared interrupts and interrupt controller cascading (which may also be described as interrupt levels/aggregators in the current schema)

### Problem description
Consider the following devicetree snippet:
```
/ {
	/* ordinal 0 */
	soc {
		/* ordinal 1 */
		intc: foo_intc {
			interrupt-controller;
			interrupt-cells = <2>;
		};

		/* ordinal 2 */
		timer0: foo_timer0 {
			interrupts <0 0>;
			status = "enabled";
		};

		/* ordinal 3 */
		timer1: foo_timer1 {
			interrupts <0 0>;
			status = "enabled";
		};

		/* ordinal 4 */
		timer2: foo_timer2 {
			interrupts <1 0>;
			status = "disabled";
		};

		/* ordinal 5 */
		gpio0: foo_gpio {
			/* GPIO is also a general interrupt controller */
			gpio-controller;
			gpio-cells = <2>;
			interrupt-controller;
			interrupt-cells = <2>;

			/* Its interrupt parent is the <&intc> */
			interrupts <3 0>;
			status = "enabled";
		};

		/* ordinal 6 */
		i2c0: foo_i2c0 {
			interrupts <4 0>
				   <5 0>;
			status = "enabled";

				/* ordinal 7 */
				sensor0: foo_sensor {
					interrupt-parent = <&gpio0>;
					interrupts = <2 3>;
					status = "enabled";
				};
		};
	};
};

```
From this we can determine the following:
* We need 6 IRQs in our SW IRQ table
* 5 IRQs belong to `intc`
* 1 IRQ belongs to `gpio0`
* `timer0` and `timer1` share `intc` interrupt line number 0
[... 193 lines cut ...]

}
```
If the interrupt controller happened to have the compat `foo`, dts ord 4, and the interrupt number was 0, the macro would expand to:
```
INTC_DT_FOO_DIRECT_IRQ_HANDLER_DEFINE(__intc_irq_vector_4_0)
```

#### Dynamic interrupts
If we want to support dynamic interrupts (I really don't think we should, its a static devicetree after all...), we would create an additional table in RAM, indexed like the `struct inct_irq_spec __intc_irq_specs`, which we would then populate at runtime, and call into from the `__intc_irq_handler_*_*` handlers:
```
struct intc_irq {
		void (*handler)(const void *data);
		const void *data;
};

struct intc_irq __intc_irq_table[sizeof(__intc_irq_specs)][CONFIG_DYNAMIC_IRQ_NUM];

/* gpio0 */
static inline void __intc_irq_handler_1_3(void)
{
		__intc_irq_handler_1_3_5();
		intc_dynamic_handlers(INTC_IRQ_SPEC_1_3);
}
```

rakons, 2024-01-15:
Some input from my side:
1. I do not think we should get priority nor flags from DT. The reason for this is that for some specific applications both of them may change during runtime. The edge on which we wish to react may change while code is executing. The same about priority - we usually are not doing that, but in some applications it may be reasonable to temporary boost or lower some IRQ priorities. So the flags and priority should be rather software defined.
2. We need dynamic interrupts. Some of our clients are using, for example radio and are changing its functionality between Bluetooth and some of the proprietary communication. They wish to have a possibility to totally disconnect one and connect other implementation. For this kind of work dynamic interrupt implementation is very usefull and it would be hard to implemented in other way.

bjarki-trackunit, 2024-01-15:
@rakons
1. I do believe we should get the flags of an IRQ from the devicetree, given that the flags are not hardware agnostic. Priority is only supported by some interrupt controllers, and could be encoded into the flags as well, so not sure about that one, however, just as an initial value, I think it is still valuable to store it, the application can override it later.

2. I can see how it would be neat to have the kernel allow that, but I do believe it could also be solved on a driver level using a switch statement, like:
```
foo_irq_handler(const void *data)
{
	switch (radio_mode) {
		case BLE:
			foo_ble_isr(data);
			break;
		case WIFI:
			foo_wifi_isr(data);
			break;
	};
}
```
which is entirely static :)

rakons, 2024-01-30:
To be sure that we are using the same function for spurious vectors we could possibly create one function (like __sys_interrupt_spurious) and then generate weak aliases to that function
```
void __sys_interrupt_handler_1_2(void) __attribute__((weak, alias("__sys_interrupt_spurious"));
```
Note: static inline function cannot be weak. For the weak functions we have to have external linkage.

bjarki-trackunit, 2024-01-30:
@rakons

The static inline handlers are not weak, they are strongly defined, its only the vectors which are weak :)

I think one level of misdirection is clearer than the attribute
```
static inline void __sys_interrupt_spurious(void)
{
        z_fatal_error(K_ERR_SPURIOUS_IRQ, NULL);
}

static inline void __sys_interrupt_handler_1_2(void)
{
        __sys_interrupt_spurious();
}
```

carlescufi, 2024-01-30:
Architecture WG:

- @bjarki-trackunit goes over the concepts presented in this RFC
- Interrupts will use a global name that is synthesized, and hidden to the user
- Interrupt handlers will be defined via macro, which will invoke a handler passed as a parameter (see description above)