zephyrproject-rtos/zephyr #67601: zephyr shell : `net ping` doesn't print any ping duration if `CONFIG_FPU` is true but `CONFIG_NEWLIB_LIBC_FLOAT_PRINTF` isn't.

matt-leg-8798, 2024-01-15:
**The bug**
It seems that `net ping` use the float format string even when it's not supported by `snprintf`.

When calling `net ping` command with `CONFIG_FPU=y` we get the follwoing output : 
```
PING fe80:0:0:0:b062:a814:2bf5:e9ff
8 bytes from fe80::b062:a814:2bf5:e9ff to fe80::947f:ebce:7e66:909c: icmp_seq=1 ttl=64 rssi=-174 time= ms
8 bytes from fe80::b062:a814:2bf5:e9ff to fe80::947f:ebce:7e66:909c: icmp_seq=2 ttl=64 rssi=-174 time= ms
8 bytes from fe80::b062:a814:2bf5:e9ff to fe80::947f:ebce:7e66:909c: icmp_seq=3 ttl=64 rssi=-174 time= ms
```

Looking at the code in `zephyr\subsys\net\ip\net_shell.c`, we see that the float format string is gated behind the `CONFIG_FPU` config. 
But that config doesnt guarenty that float format strings are supported.

Adding 
```
CONFIG_NEWLIB_LIBC=y
CONFIG_NEWLIB_LIBC_FLOAT_PRINTF=y
```
to the config fix the problem but I'm unsure that `CONFIG_NEWLIB_LIBC_FLOAT_PRINTF` is the right way to check.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the nrf OpenThread CLI sample (or any other zephyr shell app)
2. Add `CONFIG_NET_SHELL=y` to the config
3. Flash it to a board
4. ping a valid ICMP target

**Expected behavior**
The ping command should display a value for time no matter the configuration.

**Impact**
Annoyance

**Environment**
 - OS: Windows
 - Toolchain : nRF Connect SDK v2.5.1
 - Zephyr 3.4

jukkar, 2024-01-16:
@matt-leg-8798 not sure if this will work in zephyr v3.4, but please try it

matt-leg-8798, 2024-01-16:
The change works on my nRF52840 but it sems that it would block some boards with no floating point support to use the net-shell. 
Wouldn't it be preferable to change the `#ifdef` in the net-shell.c to use `CONFIG_CBPRINTF_FP_SUPPORT` and `CONFIG_NEWLIB_LIBC_FLOAT_PRINTF`?

```diff
- #ifdef CONFIG_FPU
+ #if defined(CONFIG_CBPRINTF_FP_SUPPORT) || defined(CONFIG_NEWLIB_LIBC_FLOAT_PRINTF)
			 "time=%.2f ms",
			 (double)((uint32_t)k_cyc_to_ns_floor64(cycles) / 1000000.f)
#else
			 "time=%d ms",
			 ((uint32_t)k_cyc_to_ns_floor64(cycles) / 1000000)
#endif

rlubos, 2024-01-16:
IMO that doesn't scale well, the proposal seems to cover minimal libc and newlib, but what about others (picolib, external libc impelemtnations).

> The change works on my nRF52840 but it sems that it would block some boards with no floating point support to use the net-shell.

How exactly? If there's no FPU support on the board, software FP can be used.

But TBH, if we're concerned about FP usage here, I'd personally just remove that ifdef and stick to integer computations (we could even rewrite the integer variat to have this decimal part of ms for better accuracy).

closed 2024-01-17 by fabiobaltieri

keith-packard, 2024-01-17:
You can set `CONFIG_REQUIRES_FLOAT_PRINTF` to ensure that floating point is supported in printf, but there's no current libc independent mechanism to detect if it happens to be enabled. For picolibc, that doesn't actually use do floating point computation, so enabling that won't pull in the soft float emulation code.

PR #67653 (2024-01-17): net: shell: Require float printf support from libc
Various network shell commands like ping need floating point support from libc so select the CONFIG_REQUIRES_FLOAT_PRINTF option for it.

Fixes #67601
commit 3a37c5d9 (2024-01-17): net: shell: Require float printf support from libc

Various network shell commands like ping need floating
point support from libc so select the CONFIG_REQUIRES_FLOAT_PRINTF
option for it.

Fixes #67601