zephyrproject-rtos/zephyr #67602: kernel: sleep: kernel.common.timing.minimallibc fails

katgiadla, 2024-01-15:
**Describe the bug**
The test `kernel.common.timing.minimallibc` fails.

Observed for `nrf5340dk_nrf5340_cpunet`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk_nrf5340` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/kernel/sleep -p nrf5340dk_nrf5340_cpunet --device-testing --device-serial /dev/ttyACM1 -v --inline-logs`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-4014-g0d7d39d44172 ***
Running TESTSUITE sleep
===================================================================
START - test_sleep
PASS - test_sleep in 2.006 seconds
===================================================================
START - test_sleep_forever
PASS - test_sleep_forever in 0.001 seconds
===================================================================
START - test_usleep
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/sleep/src/usleep.c:93: sleep_test_usleep: (elapsed_ms <= UPPER_BOUND_MS is false)
overslept
FAIL - test_usleep in 30.411 seconds
===================================================================
TESTSUITE sleep failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  66.67% [sleep]: pass = 2, fail = 1, skip = 0, total = 3 duration = 32.418 seconds
- PASS - [sleep.test_sleep] duration = 2.006 seconds
- PASS - [sleep.test_sleep_forever] duration = 0.001 seconds
- FAIL - [sleep.test_usleep] duration = 30.411 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: bdc079553fb65c3b828ec69c58c3e5d4
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA or Version used: zephyr-v3.5.0-4014-g0d7d39d44172

PerMac, 2024-01-22:
no longer present since zephyr-v3.5.0-4395-ga1d33f

closed 2024-01-22 by PerMac

katgiadla, 2024-02-02:
The issue occurred once again on the toolchain Toolchain Zephyr SDK 0.16.5, on the version zephyr-v3.5.0-5140-g055ac615427f.