zephyrproject-rtos/zephyr #67608: Integrate Robotframework without Renode into Twister

MP-StefanKraus, 2024-01-15:
## Introduction

Incorporate a Robot Framework without Renode need into Twister for
enhanced interoperatibility and automation capabilities.

### Problem description

Robot Framework is a framework to specify actions and assertions
in natural language with specific keywords, making the expected behaviour
specifyable for non-technical persons.

Right now, the Robot Framework integration requires running the code
in Renode as a hard requirement. This prevents executing the Robot
Framework tests in emulators like qemu, or even on device itself.

To really foster the flexibility and use all capabilities of Robot Framework,
we therefore propose to integrate a way to execute Robot Framework files
without the need for renode.

### Proposed change

With the proposed change, there will a harnes called ``robotframework`` to run
the Robot Framework without Renode, but runs the application in an arbitrary
Emulator or on device, similar to ``pytest`` harness.

## Detailed RFC

### Proposed change (Detailed)

I already created a [PR](https://github.com/zephyrproject-rtos/zephyr/pull/67607) that implement the following changes.
There is also an [Issue](https://github.com/zephyrproject-rtos/zephyr/issues/64825) that requested this kind of feature.

The necessary steps are:

- Add another harness, ``Robotframework``, to twister. This shall work similar
    to ``Pytest`` harness right now. It cares about the proper command line parameters
    to call Robot Framework, including parameters for device configuration and selection.
- Share common code between pytest and robot in twister, as things like device
    selection and other options will be similar.
- Extend ``pytest-twister-harness`` with integration for robot framework.
    This includes e.g. refactoring of command line argument parsing, an
    entrypoint to start up robot similar to a pytest call, and reusing the TwisterHarnessConfig
    between the different points of entry. This way is chosen as a lot of code and fixture usage
    like Shell, McuMgr, and other abstractions can be reused, as they are not specific
    for Robot or Pytest itself.
- Look for proper Keywords and abstractions for Robot and place them into a ZephyrLibrary. This can
    be, for example, creating and flashing a device, and communicating on the Shell.
    Implement the Python code for the keywords in the library.
- Add a sample similar to pytest to show basic usage.
- Add documentation for ``robotframework``, similar to the Pytest section. Document the keywords
    in this section.

### Dependencies

The modifications affect two parts: The main twister code, and the ``pytest-twister-harness``
library. There are some modifications needed in both parts, but both do not include
breaking changes to existing functionality. The Zephyr RTOS main code
is not affected, these changes only extend the testing functionality.

### Concerns and Unresolved Questions

We have identified the following concerns:

- Integrating robot framework code into a module called ``pytest-twister-harness``
    is misleading, but a lot of shared code exists there. Is there a low effort
    solution that does not include breaking changes, without making the setup
    for the libraries more complex?
- The name ``robotframework`` for the harness can be confusing in regard to
    ``robot`` that is currently used by renode execution.

## Alternatives

We discussed the following options or alternatives:

- Re-use the existing ``robot`` harness of twister This harness right now directly
    calls the ``renode`` binary for execution. Replacing it with just robot is not
    possible, resulting in a change of behaviour and breaking current behaviour
- Add new python module only for robot framework: This would require more modifications
    to the PYTHONPATH than right now requires, and makes sharing the common code
    harder than necessary. We still recognize that naming of pytest-twister-harness
    is no misleading.
- Using robot without twister: While theoretically possible, testing zephyr with
    different configurations and boards is where twister shines, replicating this
    would be a lot of work that can't be kept up.

mateusz-holenko, 2024-01-16:
Hi @MP-StefanKraus, thanks for sharing your proposition!

As original authors of the Renode-RobotFramework integration in Zephyr we would like share our perspective.

While the current implementation of the `robot` harness indeed targets Renode directly, from the beginning a long-term plan was to make it generic enough to support different kind of targets (including hardware and other simulators as well).
It sounds like it's a good moment to discuss it with a broader audience.

We feel that supporting two separate harnesses, called `robot` and `robotframework` will most probably introduce a lot of confusion and create an unnecessary maintenance burden in the future, but we might resolve it a bit differently.

At the moment the robot harness calls `renode-test` command that is a wrapper script around robot itself. The main purpose of this layer is to setup the Renode simulator, configure necessary variables and aggregate results (when running multiple robot suites in one go). This is of course specific to Renode and not needed for other targets (although other targets might require other kinds of wrappers as well).

In this context we believe it should be possible to generalize the existing solution to allow users for selecting robot runners (one being `renode-run`, one being just standard robot entry point, or any other wrapper people may come up with).

In our opinion, one of the biggest challenges for creating generic tests is defining a universal set of keywords that could be used against different targets. Renode, for instance, has access to the full internal state of simulation making it possible to assert the content of memory, peripherals or CPU registers. It can observe output of any devices (e.g., LEDs, network cards, UARTs and others) and can drive them by e.g., pressing a button, writing a text on UART, feeding input to sensors or sending a network packet. It can also be used to inject faults (for more advanced testing scenarios).

While it might be challenging to implement some of this functionality on actual hardware or other simulators, maybe we could start smaller by targeting only UART input and output and integrating currently used Renode keywords with the proposed ZephyrLibrary.

nashif, 2024-02-12:
This duplicates https://github.com/zephyrproject-rtos/zephyr/issues/64825, lets track this in one place.

closed 2024-02-12 by nashif

MP-StefanKraus, 2024-03-14:
Sorry for the late response. 

@nashif Would it make more sense to close the other issue and reopen this one, as it contains more information and needs probably some discussion?