zephyrproject-rtos/zephyr #67621: Links to daily test result reports broken

dtrussel, 2024-01-15:
**Describe the bug**
Hyperlinks to the daily test result reports on https://testing.zephyrproject.org/daily_tests/index.html are broken and just return "Page not found."

**To Reproduce**
Steps to reproduce the behavior:
1. Open https://testing.zephyrproject.org
2. Click on any daily test result (e.g. zephyr-v3.5.0-4086-g0a8d03b95f84, Sun Jan 14 2024 22:53:14 GMT+0100 (Central European Standard Time))

**Expected behavior**
Redirected to test report when clicking a link.

**Impact**
Unable to view daily test reports.

**Logs and console output**
n/a

**Environment (please complete the following information):**
Tested on several browsers, but I don't think that this is relevant.

nashif, 2024-01-26:
it has been like this since it was started, links are created automatically and data is uploaded after the fact, some dates will not have any test results. The whole thing will be reworked and curently being worked on.

kartben, 2024-01-26:
@nashif I think the point of this issue is that *none* of the links seem to work - this doesn't sound right?

nashif, 2024-01-29:
> @nashif I think the point of this issue is that _none_ of the links seem to work - this doesn't sound right?

_none_ is not correct, take 

https://testing.zephyrproject.org/daily_tests/zephyr-v2.5.0-2637-ga3469a049785/report/index.html

 for example

nashif, 2024-03-19:
removed links leading to nowhere for now until we have a new setup

closed 2024-03-19 by nashif