zephyrproject-rtos/zephyr #67626: Add dynamic AT commands for modem chat

emillindq, 2024-01-15:
**Is your enhancement proposal related to a problem? Please describe.**
When configuring a modem_cellular instance, some AT commands needs to be dependent on runtime configurations, for instance APN settings. At the moment, this setting comes from Kconfig and there's no possibility to easily customize AT commands run-time.

**Describe the solution you'd like**
A new AT command macro, for instance `MODEM_CHAT_SCRIPT_CMD_RESP_DYNAMIC(_request_cb, _response_match)` where the callback is called to provide the required AT command. I can implement this, but before putting way too much time on this I'd like input from maintainer @bjarki-trackunit 

**Describe alternatives you've considered**
Inserting additional states like DIAL1, APN, DIAL2 and use `modem_chat_run_script` to send it in a state transition, however today all AT commands are static and the above suggestion would potentially solve other cases.

**Additional context**
[image]

bjarki-trackunit, 2024-01-15:
Hi, check out https://github.com/zephyrproject-rtos/zephyr/blob/6f6fc4e25c3175d5a2587b0e6f3d858b4c336e99/drivers/gnss/gnss_quectel_lcx6g.c#L696-L718 and https://github.com/zephyrproject-rtos/zephyr/blob/6f6fc4e25c3175d5a2587b0e6f3d858b4c336e99/drivers/gnss/gnss_quectel_lcx6g.c#L351-L403

bjarki-trackunit, 2024-01-15:
It is not the prettiest as I have not found a way to make it simpler than this, but it is fully flexible, you can create a complete script with as many matches and chats you want at runtime :)

emillindq, 2024-01-16:
@bjarki-trackunit Thanks for your input! I've been sitting all day pondering on how to implement the APN settings, adding additional states like dial1, apn and dial2 feels bloated. But modifying modem_chat to support semi-dynamic requests via macros is not trivial. One limitation I think it reasonable is to force the modem to be suspended when using the apn set API.

Would really appreciate your input into implementing this. Thanks!

closed 2024-08-28 by emillindq