zephyrproject-rtos/zephyr #67627: PPP does not shut-down the link properly

RobMeades, 2024-01-15:
Transferred here from https://discord.com/channels/720317445772017664/733037635194585148/1196453166590476439.

PR https://github.com/zephyrproject-rtos/zephyr/pull/61147 made a change to the PPP driver so that it calls `net_if_carrier_off()` inside `ppp_stop()`:

```
static int ppp_stop(const struct device *dev)
{
        struct ppp_driver_context *context = dev->data;

        net_if_carrier_off(context->iface);
        context->modem_init_done = false;
        return 0;
}
```

Unfortunately, this means the PPP shut-down process cannot work correctly, since `fsm.c` `ppp_send_pkt()` checks that the carrier is up in order to send packets, packets which need to be exchanged in order to run through the shut-down process with the PPP peer.

```
	if (!net_if_is_carrier_ok(iface)) {
		return -ENETDOWN;
	}
```

So after calling `net_if_down()` on a PPP link you end up with something like this (noting my debug prints preceded by `###`):

```
13:12:44 ### EVENT 0xd0010001
13:12:44 D: (net_mgmt): [0x20001790] phase RUNNING (4) => TERMINATE (5) (lcp_close():166)
13:12:44 D: (net_mgmt): [LCP/0x200017c8] Current state OPENED (9)
13:12:44 D: (net_mgmt): [IPCP/0x20001898] Current state OPENED (9)
13:12:45 D: (net_mgmt): [IPCP/0x20001898] state OPENED (9) => STARTING (1) (ppp_fsm_lower_down():282)
13:12:45 D: (net_mgmt): [0x20001790] Proto IPv4 (0x0021) down (0)
13:12:45 D: (net_mgmt): [IPCP/0x20001898] Current state STARTING (1)
13:12:45 D: (net_mgmt): [IPCP/0x20001898] state STARTING (1) => INITIAL (0) (ppp_fsm_close():240)
13:12:45 D: (net_mgmt): [PAP/0x20001970] Current state OPENED (9)
13:12:45 D: (net_mgmt): [PAP/0x20001970] state OPENED (9) => STARTING (1) (ppp_fsm_lower_down():282)
13:12:45 D: (net_mgmt): [0x20001790] phase TERMINATE (5) => DEAD (0) (ppp_link_down():136)
13:12:45 D: (net_mgmt): [0x20001790] phase DEAD (0) => ESTABLISH (1) (lcp_down():182)
13:12:45 ### [LCP/0x200017c8] fsm.c, entered ppp_send_pkt()
13:12:45 ### [LCP/0x200017c8] fsm.c, ppp_send_pkt() checks net_if_is_carrier_ok() and it returns false
13:12:45 D: (net_mgmt): [LCP/0x200017c8] state OPENED (9) => CLOSING (4) (terminate():215)
13:12:45 ### EVENT 0xd0010003
...
13:12:54 D: (sysworkq): [LCP/0x200017c8] Current state CLOSING (4)
13:12:54 D: (sysworkq): ### [LCP/0x200017c8]  fsm.c, entered ppp_send_pkt()
13:12:55 D: (sysworkq): ### [LCP/0x200017c8] fsm.c, ppp_send_pkt() checks net_if_is_carrier_ok() and it returns false
13:13:04 D: (sysworkq): [LCP/0x200017c8] Current state CLOSING (4)
13:13:04 D: (sysworkq): [LCP/0x200017c8] state CLOSING (4) => CLOSED (2) (ppp_fsm_timeout():136)
13:13:04 D: (sysworkq): [0x20001790] phase ESTABLISH (1) => DEAD (0) (ppp_link_terminated():123)
13:13:04 D: (sysworkq): [0x20001790] Link terminated
```

The link just times out and the PPP peer knows nothing about it.

If I comment out the call to `net_if_carrier_off()` in `ppp_stop(0` then a proper link termination, with the peer, is performed.

**To Reproduce**
- Make a PPP connection to something.
- Try to terminate that connection and watch the process happen (e.g. with debug on) so that you can tell the difference between a timed-out termination and a peer-to-peer termination.

**Impact**
The PPP peer is left hanging, which mean it won't start up again the next time you try to talk to it unless you've powered cycled or reset it.  In other words, once you have disconnected PPP you can never connect it again.

**Version**
- Zephry 3.4.99 (as arrives with nRFConnect 2.5.0), but headrev Zephyr seems actually worse, in that it now also calls `uart_rx_disable()` for the asynchronous UART case (which I am using)...!?

jukkar, 2024-01-15:
@bjarki-trackunit please also take a look if you have time.

bjarki-trackunit, 2024-01-15:
I don't believe the `net_if_down()` or `net_if_up()` should directly control the carrier, which is what is happening here (`ppp_stop` is called from `net_if_l2(iface)->enable(iface, false)` in `net_if.c`.

Maybe @jhirsi can expand on the issue?

bjarki-trackunit, 2024-01-16:
We seem to agree on the issue :) The solution should be to remove `net_if_carrier_on()` and `net_if_carrier_off()` calls from the `ppp_start()` and `ppp_stop()` functions, since the flag `NET_IF_LOWER_UP` (set/cleared with `net_if_carrier_on/off`) is set by default See https://github.com/zephyrproject-rtos/zephyr/blob/9074feabf6f827f6b37a69d77a5e13093799a905/include/zephyr/net/net_if.h#L2832 and https://github.com/zephyrproject-rtos/zephyr/blob/9074feabf6f827f6b37a69d77a5e13093799a905/include/zephyr/net/net_if.h#L2850

RobMeades, 2024-01-16:
That sounds great: while we're here, headrev Zephyr's `ppp_stop()` also has this line in it:

```
#if defined(CONFIG_NET_PPP_ASYNC_UART)
	uart_rx_disable(context->dev);
#endif
```

Since I am using the asynchronous UART interface this would cause the same problem: could it be removed also?

bjarki-trackunit, 2024-01-16:
@tomi-font Hi, can you help with this issue? I think you added the call to `uart_rx_disable(context->dev);` last month, it may be related to this issue :)

henrikbrixandersen, 2024-01-16:
@rlubos, @jukkar Please assign a priority to this issue.

carlescufi, 2024-01-17:
@bjarki-trackunit can you take a look at [Robert's comment](https://github.com/zephyrproject-rtos/zephyr/issues/67627#issuecomment-1893884405)?

bjarki-trackunit, 2024-01-17:
@carlescufi I'm working on a solution which makes everyone happy :) Give me a couple of hours ;)

bjarki-trackunit, 2024-01-18:
@rlubos

The PR #67731 should explain itself, but I will address your comment here for clarity :)

> For me the PPP driver does not seem to be the problem here (if the PPP L2 tells the driver to stop it's operation by calling stop() - what's wrong with disabling underlying UART peripheral?), but rather the fact that PPP L2 disables the driver, while still expecting from it to operate (to perform PPP teardown).

Agreed, when disabling the interface, the L2 PPP will now block until PPP is dead, before calling the `stop` function, making it safe to disable the UART.

>  It's probably even more confusing, given the PPP interface management logic is scattered between enable() ...

The management logic still uses both the call to `enable()` and the net_mgmt callback to manage the L2 PPP instance after #67731 the change here is that calling `net_if_down()` is now synchronous. We need the net mgmt callback to monitor if the carrier is lost, and when it comes back, to signal the L2 PPP interface (and the upper layers in propagation) that the network is down temporarily.

PR #67731 (2024-01-19): net: l2: ppp: Patch L2 PPP network interface behavior
### Overview

This PR adjusts the behavior of the L2 PPP subsystem as follows:
* Synchronize closing of protocols with call to net_if_down()
* Only close LCP if the carrier is present, and the network interface is down, otherwise simply reset protocols

These two changes ensure the following:
* The `struct ppp_api` `stop` API is called after L2 PPP module is completely `DEAD`
* The L2 PPP module will automatically recover a lost carrier while UP (`net_if_up()` has been called)

This fixes the following issues:
* Terminating the carrier from [... cut]
closed 2024-01-19 by henrikbrixandersen