zephyrproject-rtos/zephyr #67630: imx_pm_device: Builds fail when CONFIG_PM_DEVICE and CONFIG_NETWORKING are enabled for MIMXRT1170_EVK (CM7)

skumg, 2024-01-15:
**Describe the bug**
The build configuration fails with the following error

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? mimxrt1170_evk_cm7

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the following configurations in a sample program
   CONFIG_PM=y
   CONFIG_PM_DEVICE=y
   CONFIG_NETWORKING=y

3. Run the build for mimxrt1170_evk_cm7
4. Notice the following error: 

**Logs and console output**
   ```
Building C object zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_mcux.c.obj
FAILED: zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_mcux.c.obj 
ccache /opt/toolchains/zephyr-sdk-0.16.4/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DBOARD_FLASH_SIZE="CONFIG_FLASH_SIZE*1024" -DCACHE_MODE_WRITE_THROUGH -DCPU_MIMXRT1176DVMAA_cm7 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -DNDEBUG -DPICOLIBC_LONG_LONG_PRINTF_SCANF -DXIP_BOOT_HEADER_DCD_ENABLE=1 -DXIP_BOOT_HEADER_ENABLE=1 -DXIP_EXTERNAL_FLASH -D_POSIX_C_SOURCE=200809 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/workdir/deps/zephyr/include -I/workdir/build/zephyr/include/generated -I/workdir/deps/zephyr/soc/arm/nxp_imx/rt -I/workdir/deps/zephyr/subsys/net/l2 -I/workdir/deps/zephyr/subsys/net/lib/dns/. -I/workdir/deps/base/include -I/workdir/deps/modules/hal/cmsis/CMSIS/Core/Include -I/workdir/deps/zephyr/modules/cmsis/. -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176 -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/drivers -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/devices/MIMXRT1176/xip -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/common -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/caam -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/lpuart -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/gpt -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/enet -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/igpio -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/cache/armv7-m7 -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/drivers/gpc_3 -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/components/phy -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phyksz8081 -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phyar8031 -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phylan8720a -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/components/phy/device/phyrtl8211f -I/workdir/deps/modules/hal/nxp/mcux/mcux-sdk/components/phy/mdio/enet -I/workdir/deps/zephyr/modules/hal_nxp/. -I/workdir/deps/gin_utils/include/extensions -I/workdir/deps/gin_utils/include/lib -I/workdir/deps/gin_utils/include/types -I/workdir/deps/base/modules/gp/include -isystem /workdir/deps/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /workdir/build/zephyr/include/generated/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m7 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/opt/toolchains/zephyr-sdk-0.16.4/arm-zephyr-eabi/arm-zephyr-eabi -imacros /workdir/deps/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/workdir/deps/zephyr/samples/subsys/pm/device_pm=CMAKE_SOURCE_DIR -fmacro-prefix-map=/workdir/deps/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/workdir=WEST_TOPDIR -ffunction-sections -fdata-sections --specs=picolibc.specs -std=c99 -MD -MT zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_mcux.c.obj -MF zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_mcux.c.obj.d -o zephyr/drivers/ethernet/CMakeFiles/drivers__ethernet.dir/eth_mcux.c.obj -c /workdir/deps/zephyr/drivers/ethernet/eth_mcux.c
In file included from /workdir/deps/zephyr/include/zephyr/sys/util_macro.h:34,
                 from /workdir/deps/zephyr/include/zephyr/sys/util.h:17,
                 from /workdir/deps/zephyr/include/zephyr/sys/cbprintf_internal.h:15,
                 from /workdir/deps/zephyr/include/zephyr/sys/cbprintf.h:124,
                 from /workdir/deps/zephyr/include/zephyr/logging/log_msg.h:11,
                 from /workdir/deps/zephyr/include/zephyr/logging/log_core.h:9,
                 from /workdir/deps/zephyr/include/zephyr/logging/log.h:11,
                 from /workdir/deps/zephyr/drivers/ethernet/eth_mcux.c:23:
/workdir/deps/zephyr/include/zephyr/device.h:89:41: error: '__device_dts_ord_DT_N_S_soc_S_ethernet_40424000_P_clocks_IDX_0_PH_ORD' undeclared here (not in a function)
   89 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                         ^~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/devicetree.h:4112:9: note: in expansion of macro 'COND_CODE_1'
 4112 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT),   \
      |         ^~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:69:53: note: in expansion of macro '__DEBRACKET'
   69 | #define __GET_ARG2_DEBRACKET(ignore_this, val, ...) __DEBRACKET val
      |                                                     ^~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/sys/util_macro.h:180:9: note: in expansion of macro 'Z_COND_CODE_1'
  180 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/workdir/deps/zephyr/drivers/ethernet/eth_mcux.c:1498:9: note: in expansion of macro 'COND_CODE_1'
 1498 |         COND_CODE_1(CONFIG_NET_POWER_MANAGEMENT,                        \
      |         ^~~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/toolchain/common.h:138:23: note: in expansion of macro '_DO_CONCAT'
  138 | #define _CONCAT(x, y) _DO_CONCAT(x, y)
      |                       ^~~~~~~~~~
/workdir/deps/zephyr/include/zephyr/device.h:89:33: note: in expansion of macro '_CONCAT'
   89 | #define DEVICE_NAME_GET(dev_id) _CONCAT(__device_, dev_id)
      |                                 ^~~~~~~
/workdir/deps/zephyr/include/zephyr/device.h:216:37: note: in expansion of macro 'DEVICE_NAME_GET'
[... 16 lines cut ...]
      |                                                       ^~
/workdir/deps/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_nxp_kinetis_ethernet'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/workdir/deps/zephyr/drivers/ethernet/eth_mcux.c:1652:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
 1652 | DT_INST_FOREACH_STATUS_OKAY(ETH_MCUX_INIT)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[275/285] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/ipv6_nbr.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /workdir/build
```

**Expected behavior**
The build should be successful. 

**Impact**
Cannot proceed with the device runtime power management (showstopper)

**Environment (please complete the following information):**
    OS: Linux
    Toolchain: Zephyr SDK with West
    Version used: 3.5.99

**Additional context**
For the convenience of our project, the Zephyr SDK has been moved to /workdir/deps/zephyr/...

decsny, 2024-01-15:
The build error is because the RT series in the SOC DTSI does not put a clocks property on the enet node. I have to assume this means that net power management has never been tested or developed for the ethernet on the RT platforms before.

skumg, 2024-01-16:
@decsny Thank you for the response, Will it in that case be possible to mark this as an enhancement, mentioning the implementation of net power management for the Ethernet interfaces on the RT platforms, or create a new Issue:Enhancement altogether?

decsny, 2024-01-16:
> @decsny Thank you for the response, Will it in that case be possible to mark this as an enhancement, mentioning the implementation of net power management for the Ethernet interfaces on the RT platforms, or create a new Issue:Enhancement altogether?

Let's keep it as a bug for now because at the very least it shouldn't be confusing as to if it's supposed to be supported or not, so it needs to be addressed in some way

decsny, 2024-02-13:
@skumg apologies, I misunderstood the issue, now I see the build is breaking when trying to use device power management generally, not just for ethernet, when using ethernet, so I'll bump the priority of this up a bit. I have a PR which is a bit of a hotfix to bypass this build error for now. Please make an enhancement ticket for PM device in the ethernet driver on RT series to track this issue further

PR #68963 (2024-02-16): drivers: eth_mcux: Fix PM_DEVICE build failure
Enabling device power management while building for RT series platform while using ethernet currently breaks the build, RT series has not been tested with this net PM code anyways, so disable it for now except for kinetis family on which it has been tested

Fixes #67630
closed 2024-02-16 by henrikbrixandersen