zephyrproject-rtos/zephyr #67637: Telnet shell asserts if connection is closed

jukkar, 2024-01-15:
**Describe the bug**

Some extra discussion about the issue can be found in PR #67596, basically telnet shell asserts if telnet connection is closed.
If asserts are disabled no visible error was seen, but according to the code in
https://github.com/zephyrproject-rtos/zephyr/blob/11f7fa278200330fa4615e32489eefd174232c3f/subsys/shell/shell_ops.c#L425
the internal variables in the `z_shell_write()` could contain invalid values which could corrupt internal state of the shell.

**To Reproduce**
```
west build -p -b native_sim samples/net/telnet/ -d ../build/telnet \
  -t run -- -DCONFIG_ASSERT=y -DCONFIG_NATIVE_UART_0_ON_STDINOUT=y \
  -DCONFIG_LOG_BACKEND_UART=y -DCONFIG_SHELL_LOG_BACKEND=n
```
and then execute `echo x | telnet 192.0.2.1` from the host.

**Expected behavior**
The telnet connection is closed without any errors / crashes.

**Impact**
If the asserts are enabled, the crashing of the system is not necessary as we just closed the connection.
If the asserts are disabled, the internal state of the shell might cause corruption.

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v3.5.0-4132-gf37d150c1052 ***

uart:~$ ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/shell/shell_ops.c:438
@ WEST_TOPDIR/zephyr/lib/os/assert.c:43
Exiting due to fatal error
```

**Environment (please complete the following information):**
 - OS: Linux
 - Commit SHA or Version used: zephyr-v3.5.0-4132-gf37d150c1052

kevinior, 2024-09-24:
This issue is still present in the rewritten telnet server in 3.7.0:
`ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/shell/shell_ops.c:438`

As a result we're having to maintain a patched version of the telnet server that returns 0 when the connection closes, to avoid crashing our whole application.

kevinior, 2024-09-25:
> @kevinior if you have a patch for this, could you submit a PR for it?

We just patched the `telnet_write` function to always return 0 and log some diagnostics about errors. I don't know if that would be an acceptable fix for others.

jukkar, 2024-09-25:
> We just patched the `telnet_write` function to always return 0 and log some diagnostics about errors. I don't know if that would be an acceptable fix for others.

Yeah, that sounds a bit hackish to me. But you could try to send it still, the review could give ideas how to solve it better.

kevinior, 2024-09-25:
@jukkar OK, I've created a PR for the hackish fix.

kevinior, 2024-10-09:
@jukkar No responses to the PR so far

PR #78976 (2024-10-11): shell: backend: telnet: Don't assert if connection closed
The code in shell_ops.c that calls telnet_write will assert if it returns non-zero. For a telnet shell it's normal that the network might disconnect unexpectedly, so that should not trigger an assert.

Fixes #67637

Link: https://github.com/zephyrproject-rtos/zephyr/issues/67637
commit 3399e061 (2024-10-11): shell: backend: telnet: Don't assert if connection closed

The code in shell_ops.c that calls telnet_write will assert if it
returns non-zero. For a telnet shell it's normal that the
network might disconnect unexepectedly, so that should not
trigger an assert.

Fixes #67637

Link: https://github.com/zephyrproject-rtos/zephyr/issues/67637
commit 9cd8d8f4 (2025-07-21): shell: backend: telnet: Don't assert if connection closed

The code in shell_ops.c that calls telnet_write will assert if it
returns non-zero. For a telnet shell it's normal that the
network might disconnect unexepectedly, so that should not
trigger an assert.

Fixes #67637

Link: https://github.com/zephyrproject-rtos/zephyr/issues/67637

(cherry picked from commit 3399e0614adbc67350ceac85ac747daef32a0534)
closed 2024-10-11 by nashif