zephyrproject-rtos/zephyr #67682: Board / SoC porting progress

tejlmand, 2024-01-16:
Overview of current board / SoC porting effort:

Please update this description in the board / vendor section when starting board porting effort.

### Unsorted / no vendor
- [ ] Board: 96b_argonkey
- [ ] Board: 96b_neonkey
- [ ] Board: 96b_wistrio
- [ ] Board: acn52832
- [ ] Board: acrn
- [ ] Board: acrn_ehl_crb
- [ ] Board: adafruit_grand_central_m4_express
- [ ] Board: adafruit_kb2040
- [ ] Board: adafruit_qt_py_rp2040
- [ ] Board: arduino_uno_r4_minima
- [ ] Board: arty_a7_arm_designstart_m1
- [ ] Board: arty_a7_arm_designstart_m3
- [ ] Board: az3166_iotdevkit
- [ ] Board: bbc_microbit
- [ ] Board: bbc_microbit_v2
- [ ] Board: black_f407ve
- [ ] Board: black_f407zg_pro
- [ ] Board: circuitdojo_feather_nrf9160
- [ ] Board: circuitdojo_feather_nrf9160_ns
- [ ] Board: da14695_dk_usb
- [ ] Board: dragino_lsn50
- [ ] Board: dragino_nbsn95
- [ ] Board: ebyte_e73_tbb_nrf52832
- [ ] Board: esp32s2_lolin_mini
- [ ] Board: esp32s3_devkitm_appcpu
- [ ] Board: fvp_baser_aemv8r_aarch32
- [ ] Board: fvp_baser_aemv8r_aarch32_smp
- [ ] Board: heltec_wireless_stick_lite_v3
- [ ] Board: holyiot_yj16019
- [ ] Board: mps2_an521_ns
- [ ] Board: mps2_an521_remote
- [ ] Board: mps3
- [ ] Board: neorv32
- [ ] Board: npcx4m8f_evb
- [ ] Board: npcx7m6fb_evb
- [ ] Board: npcx9m6f_evb
- [ ] Board: nrf51_blenano
- [ ] Board: nrf51_vbluno51
- [ ] Board: nrf52832_mdk
- [ ] Board: nrf52840_mdk_usb_dongle
- [ ] Board: nrf52_blenano2
- [ ] Board: nrf52_bsim
- [ ] Board: nrf52_vbluno52
- [ ] Board: nrf5340bsim_nrf5340_cpuapp
- [ ] Board: nrf5340bsim_nrf5340_cpunet
- [ ] Board: nrf9131ek_nrf9131
- [ ] Board: nrf9131ek_nrf9131_ns
- [ ] Board: nrf9160_innblue21
- [ ] Board: nrf9160_innblue21_ns
- [ ] Board: nrf9160_innblue22
- [ ] Board: nrf9160_innblue22_ns
- [ ] Board: nsim_hs5x_smp_12cores
- [ ] Board: nsim_hs6x_smp_12cores
- [ ] Board: nucleo_f031k6
- [ ] Board: nucleo_f042k6
- [ ] Board: nucleo_l011k4
- [ ] Board: nucleo_u5a5zj_q
- [ ] Board: nxp_adsp_imx8ulp
- [ ] Board: pan1770_evb
- [ ] Board: pan1780_evb
- [ ] Board: pan1781_evb
- [ ] Board: pan1782_evb
- [ ] Board: pandora_stm32l475
- [ ] Board: qemu_cortex_m0
- [ ] Board: qemu_leon3
- [ ] Board: qemu_riscv32
- [ ] Board: qemu_riscv32_smp
- [ ] Board: qemu_riscv32_xip
- [ ] Board: qemu_riscv32e
- [ ] Board: qemu_riscv64
- [ ] Board: qemu_riscv64_smp
- [ ] Board: qemu_xtensa_mmu
- [ ] Board: raytac_mdbt50q_db_33_nrf52833
- [ ] Board: raytac_mdbt50q_db_40_nrf52840
- [ ] Board: raytac_mdbt53_db_40_nrf5340_cpuapp
- [ ] Board: raytac_mdbt53_db_40_nrf5340_cpuapp_ns
- [ ] Board: raytac_mdbt53_db_40_nrf5340_cpunet
- [ ] Board: raytac_mdbt53v_db_40_nrf5340_cpuapp
- [ ] Board: raytac_mdbt53v_db_40_nrf5340_cpuapp_ns
- [ ] Board: raytac_mdbt53v_db_40_nrf5340_cpunet
- [ ] Board: rcar_spider_cr52
- [ ] Board: riscv32_virtual
- [ ] Board: roc_rk3568_pc
- [ ] Board: roc_rk3568_pc_smp
- [ ] Board: ronoth_lodev
- [ ] Board: rpi_4b
- [ ] Board: rpi_pico
- [ ] Board: rpi_pico_w
- [ ] Board: scobc_module1
- [ ] Board: serpente
- [ ] Board: sparkfun_pro_micro_rp2040
- [ ] Board: stm32f030_demo
- [ ] Board: stm32f3_seco_d23
- [ ] Board: stm32f401_mini
- [ ] Board: up_squared
- [ ] Board: verdin_imx8mp_m7_ddr
- [ ] Board: verdin_imx8mp_m7_itcm
- [ ] Board: w5500_evb_pico
- [ ] Board: xt-sim
- [ ] 
### UP
- [ ] Board: up_squared_pro_7000

### actinus
- [ ] board: actinius_icarus
- [ ] board: actinius_icarus_bee
- [ ] board: actinius_icarus_bee_ns
- [ ] board: actinius_icarus_ns
- [ ] board: actinius_icarus_som
- [ ] board: actinius_icarus_som_dk
- [ ] board: actinius_icarus_som_dk_ns
- [ ] board: actinius_icarus_som_ns

### Adafruit 
- [ ] board: adafruit_feather_m0_basic_proto
- [ ] board: adafruit_feather_m0_lora
- [ ] board: adafruit_feather_nrf52840
- [ ] board: adafruit_feather_stm32f405
- [ ] board: adafruit_itsybitsy_m4_express
- [ ] board: adafruit_itsybitsy_nrf52840
- [ ] board: adafruit_trinket_m0
- [ ] board: nrf52_adafruit_feather

### altr
- [ ] board: altera_max10
- [ ] board: cyclonev_socdk

### ambiq
- [ ] board: apollo4p_blue_kxr_evb
- [ ] board: apollo4p_evb

### andestech
- [ ] board: adp_xc7k_ae350

### Arduino
- [ ] board: arduino_due
- [ ] board: arduino_giga_r1_m4
- [ ] board: arduino_giga_r1_m7
- [ ] board: arduino_mkrzero
- [ ] board: arduino_nano_33_ble
- [ ] board: arduino_nano_33_ble_sense
- [ ] board: arduino_nano_33_iot
- [ ] board: arduino_nicla_sense_me
- [ ] board: arduino_opta_m4
- [ ] board: arduino_portenta_h7_m4
- [ ] board: arduino_portenta_h7_m7
- [ ] board: arduino_zero

erwango, 2024-01-17:
@tejlmand See comment from discord: 
IMO, it would be clearer to list SoCs series above, vendors/boards being work output.
If we want to track elementary changes (PRs), listing series is easier for everyone to follow (one box, one series, one PR).

Additionally, it would help cleaning the "Unsorted / no vendor" section which is not accurate (likely due to missing "vendor" field in respective \<boards\>.yaml files.

tejlmand, 2024-01-17:
@erwango Thanks for the input.

> IMO, it would be clearer to list SoCs series above, vendors/boards being work output.

Sure it would, but it would require to go through all boards, fetch their SoC and series.
Can probably be scripted, question is mostly how much effort do we want to put into creating this list vs. porting.

but will take a second look.

> Additionally, it would help cleaning the "Unsorted / no vendor" section which is not accurate (likely due to missing "vendor" field in respective <boards>.yaml files.

Correct, those are exactly due to a missing vendor field in the `<boards>.yaml` file.
I preferred to get the list available first, and then fine tune it later by moving boards.

jfischer-no, 2024-01-17:
I have started porting atmel_sam0 based boards, but if someone has more time and wants to take over, please ping me.

tejlmand, 2024-01-24:
@erwango list updated according to suggestions.

nandojve, 2024-02-09:
> I have started porting atmel_sam0 based boards, but if someone has more time and wants to take over, please ping me.

Hi @jfischer-no ,

I'm ready to start migrate all SAM/SAM0 boards. Ping on Discord in PVT to sync about it : )

nandojve, 2024-02-10:
> > I have started porting atmel_sam0 based boards, but if someone has more time and wants to take over, please ping me.
> 
> Hi @jfischer-no ,
> 
> I'm ready to start migrate all SAM/SAM0 boards. Ping on Discord in PVT to sync about it : )

Hi @jfischer-no ,

I already migrate all SAM SoC and boards. I'll start with SAM0 and maybe there will be a PR for next week.
@tejlmand , you can set to me all sam* and all atsam* boards. Additionally, I migrate arduino_due (sam3x) and tdk_robokit1.

closed 2024-03-04 by nordicjm

SnappierSoap318, 2024-03-08:
Is there any chance this arduino Nano ESP32-S3 board might be added? https://docs.arduino.cc/hardware/nano-esp32/

nordicjm, 2024-03-08:
> Is there any chance this arduino Nano ESP32-S3 board might be added? https://docs.arduino.cc/hardware/nano-esp32/

That is unrelated to porting hwmv1 to hwmv2

SnappierSoap318, 2024-03-08:
> > Is there any chance this arduino Nano ESP32-S3 board might be added? https://docs.arduino.cc/hardware/nano-esp32/
> 
> That is unrelated to porting hwmv1 to hwmv2

Sorry, i saw an issue relating to Boards and i thought of putting it here.