zephyrproject-rtos/zephyr #67693: Tests: net: dumb_http_server_mt - setting different IPv4 address than one assigned by DHCP crashes.

JDuchniewicz, 2024-01-16:
**Describe the bug**
When running the sample `dumb_http_server_mt` on ESP32 WROOM I experience a crash occuring upon prompting the server to return data to the client - e.g HTTP request.

I found it only reproduces when I set the `CONFIG_NET_CONFIG_MY_IPV4_ADDR` to one that is not assigned by DHCP to the new Wi-Fi device upon running `wifi connect "<SSID>" <password>`. So if that variable in `prj.conf` is set to e.g. 192.168.0.3 and the DHCP server assigns 192.168.20.103 then the crash occurs, otherwise if I set it to the same IP that will be assigned by DHCP all is fine.

What is interesting is that all is fine with the single-thread sample - `dumb_http_server`. The IP indeed changes to the one set in the config and not assigned by DHCP.

**To Reproduce**
Steps to reproduce the behavior, From the root of zephyr project:
1. `cat CONFIG_NET_L2_WIFI_SHELL=y > samples/net/sockets/dumb_http_server_mt`
2. `cat CONFIG_ESP_HEAP_MEM_POOL_REGION_1_SIZE=1024 > samples/net/sockets/dumb_http_server_mt`
3. Enable the `wifi` in `dts` - `cp samples/net/wifi/boards/esp32_devkitc_wroom.conf samples/net/sockets/dumb_http_server_mt/`
4. `west build -p -b esp32_devkitc_wroom samples/net/sockets/dumb_http_server_mt`
5. `west flash`
6. `west espressif monitor`
7. `wifi connect "<SSID>" <password>`
8. See error

**Expected behavior**
Expected behavior is having no crash and ideally the interface adopting the IP set in the config file with precedence over the one assigned by DHCP.

**Impact**
Not big impact, apart from confusion - can take steps to debug and solve it once I finish the core task of my project.

**Logs and console output**
```
[00:01:17.174,000] <dbg> net_dumb_http_srv_mt_sample: process_tcp: [1] Connection #1 from 192.168.0.211
[00:01:22.604,000] <err> os:  **  PS 0x60320
[00:01:22.604,000] <err> os:  **    (INTLEVEL:0 EXCM: 0 UM:1 RING:0 WOE:1 OWB:3 CALLINC:2)
[00:01:22.604,000] <err> os:  **  A0 0x800e780c  SP 0x3ffe58f0  A2 0xffffffff  A3 0x1
[00:01:22.604,000] <err> os:  **  A4 0x3ffe59c8  A5 0x3ffcd0d0  A6 0x8  A7 0x1
[00:01:22.604,000] <err> os:  **  A8 0x800e863c  A9 0x3ffe58c0 A10 0x400e7bd4 A11 0x3ffcd0d0
0x400e7bd4: sock_ioctl_vmeth at /home/jduchniewicz/Projects/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:3465

[00:01:22.604,000] <err> os:  ** A12 0x3f401d34 A13 0x3ffe59b8 A14 0xeec90001 A15 0xd300a8c0
[00:01:22.604,000] <err> os:  ** LBEG 0x4000c2e0 LEND 0x4000c2f6 LCOUNT 0xffffffff
[00:01:22.604,000] <err> os:  ** SAR 0x4

Ba0x40085794:0x30x400e7809:0x30x400e82b1:0x30x40081199:0x30x400d0931:0x30x400d3ad0:0x30x40113095:0x30x401056c2:0x30x40106d95:0x30x40106e76:0x30x400fa3e1:0x30x400eb3d5:0x30x400ef1eb:0x30x00060f1d:0x3 |<-C
0x40085794: z_impl_k_mutex_unlock at /home/jduchniewicz/Projects/zephyrproject/zephyr/kernel/mutex.c:212

0x400e7809: k_mutex_unlock at /home/jduchniewicz/Projects/zephyrproject/zephyr/build/zephyr/include/generated/syscalls/kernel.h:1002

0x400e82b1: z_impl_zsock_accept at /home/jduchniewicz/Projects/zephyrproject/zephyr/subsys/net/lib/sockets/sockets.c:685 (discriminator 5)

0x40081199: malloc_internal_wrapper at /home/jduchniewicz/Projects/zephyrproject/modules/hal/espressif/zephyr/esp32/src/wifi/esp_wifi_adapter.c:432

0x400d0931: esf_buf_alloc_dynamic at ??:?

0x400d3ad0: esf_buf_alloc at ??:?

0x40113095: ic_ebuf_alloc at ??:?

0x401056c2: ieee80211_alloc_tx_buf at ??:?

0x40106d95: ieee80211_output_do at ??:?
0x40106e76: esp_wifi_internal_tx at ??:?

0x400fa3e1: esp32_wifi_send at /home/jduchniewicz/Projects/zephyrproject/zephyr/drivers/wifi/esp32/src/esp_wifi_drv.c:121

0x400eb3d5: net_l2_send at /home/jduchniewicz/Projects/zephyrproject/zephyr/include/zephyr/net/net_l2.h:167
 (inlined by) ethernet_send at /home/jduchniewicz/Projects/zephyrproject/zephyr/subsys/net/l2/ethernet/ethernet.c:719

0x400ef1eb: net_if_tx at /home/jduchniewicz/Projects/zephyrproject/zephyr/subsys/net/ip/net_if.c:258

[00:01:22.605,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:01:22.605,000] <err> os: Current thread: 0x3ffcb730 (tcp4_thread_id)
[00:01:22.776,000] <err> os: Halting system
```

**Environment (please complete the following information):**
I am running it on:
`Linux thinkie 6.6.10-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 05 Jan 2024 16:20:41 +0000 x86_64 GNU/Linux`

With Zephyr v3.5.0, commit: `b83b9bede3bdc787615073a99463e6a00932318d`
SDK version: `zephyr-sdk-0.16.0_linux-x86_64`

Running on board:
`ESP32 WROOM`

rlubos, 2024-01-24:
I'm afraid this could be something platform-specific - tried to reproduce on my `nucleo_h723zg` with no success, no issues with HTTP requests when trying with static address (before DHCP kicks in) or with DHCP-assigned one (different than sttaic address). I'm afraid I won't be able to help in this case.

sylvioalves, 2024-01-24:
@JDuchniewicz Is `CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y` enabled by default? I mean, if it is, I do think you should disabled it as it forces dhcp connection in driver level. Your application could handles this instead. Let me know.

JDuchniewicz, 2024-01-28:
I just tried reproducing it yet again and did not succeed. Weird because I had a 100% reproducibility rate before. Also yes, the flag `CONFIG_ESP32_WIFI_STA_AUTO_DHCPV4=y` is enabled in this particular build. Closing for now, if I encounter it again I will reopen. Thanks.

closed 2024-01-28 by JDuchniewicz