zephyrproject-rtos/zephyr #67695: posix: evolve pooled type pattern

cfriedt, 2024-01-16:
**Is your enhancement proposal related to a problem? Please describe.**
There is a pattern used in several areas in the POSIX API implementation (and elsewhere - e.g. `k_thread_stack_alloc()`), where a `bitarray` is used to mark whether a particular object has been reserved or not. 

There are a few issues with this approach:
1. While the bitarray is synchronized with a spinlock, it is only synchronized for bitarray operations (other operations that happen with the resource synchronize on `xxx_pool_spinlock` (where `xxx` is e.g. `posix_mutex`).
2. a fair bit of code is duplicated between sources (can we extract this pattern, give it a name, and make it a more generic solution for a common problem?)
3. while it's good to have a fixed number of things for systems where `malloc()` is a no-go, it's also nice to be able to use `malloc()`, when possible. Can the solution in 2 above be generalized to support 
  i. pool-only allocation
  ii. pool + heap allocation
  iii. heap-only allocation

**Describe the solution you'd like**
When considering the problem, and the configuration system we have available (Kconfig), likely the most concise way we could use to represent this functionality would be to use an `int`.

i. pool-only allocation: 5
ii. pool + heap allocation: -5
iii. heap-only allocation: 0

**Describe alternatives you've considered**
The current status-quo is to copy-paste from one `lib/posix` pooled ipc type to another, which isn't ideal. It would be great to not duplicate code.

We can likely get the same benefits of a bitarray for allocating / freeing individual by using a slab allocator. The slab allocator does not provide internal locking - but that's ok, because we already have a pool lock for each pooled resource.

The same pattern is used by `k_thread_stack_alloc()`. Theoretically, it could also be used for dynamic allocation of userspace objects.

**Additional context**

cfriedt, 2024-01-16:
Naming things is hard, but it can also be fun 😀 
* elastipool
* flexipool
* kiddie_pool
* splash_pad

I would assume Zephyr would probably only go for the non-fun names though :P

cfriedt, 2024-01-16:
cc: @ceolin

nashif, 2024-01-16:
@peter-mitsis  @andyross  FYI