zephyrproject-rtos/zephyr #67705: Add bluedroid SBC encoder/decoder to Zephyr as an external library

MarkWangChinese, 2024-01-17:
## Origin

The Android bluedroid SBC
https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/refs/heads/main/embdrv/sbc

## Purpose

The codes implement the encoder and decoder of SBC.
I am implementing the A2DP in Zephyr (https://github.com/zephyrproject-rtos/zephyr/pull/31583), A2DP has mandatory needs for SBC.

## Mode of integration

This would be a Zephyr module with `libsbc` as the name.
In the pr (https://github.com/zephyrproject-rtos/zephyr/pull/31583), I am using the name `libsbc` like `liblc3`. You can change it's name if needed.
I will do very limited codes changes and add one porting file based on bluedroid codes to pass building.

## Maintainership

Maintainers - @MarkWangChinese
I will add new people later after talking with community.

## Pull Request

I will wait it is approved and the repository is created, then I can push the codes and create PR.

## Description

In order to support A2DP in Zephyr, we need to have the SBC encoder and decoder because it is mandatory for A2DP. 
Android bluedroid maintain it, so Zephyr can take advantage of it. It's license is Apache2.0

## Dependencies

It is only used by A2DP now.
I am implementing one middle layer a2dp_codec/sbc/a2dp_codec_sbc.c (https://github.com/zephyrproject-rtos/zephyr/pull/31583), This component can be replaced if the middle layer API is re-implemented.

## Revision

wait the repository to be created.

## License

Apache License, Version 2.0
For example: the license of https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/refs/heads/main/embdrv/sbc/encoder/srce/sbc_encoder.c is as follow:
```
/******************************************************************************
 *
 *  Copyright (C) 1999-2012 Broadcom Corporation
 *
 *  Licensed under the Apache License, Version 2.0 (the "License");
 *  you may not use this file except in compliance with the License.
 *  You may obtain a copy of the License at:
 *
 *  http://www.apache.org/licenses/LICENSE-2.0
 *
 *  Unless required by applicable law or agreed to in writing, software
 *  distributed under the License is distributed on an "AS IS" BASIS,
 *  WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 *  See the License for the specific language governing permissions and
 *  limitations under the License.
 *
 ******************************************************************************/
```

nashif, 2024-01-31:
> I will wait it is approved and the repository is created, then I can push the codes and create PR.

It would help us make a decision if there is a draft PR that shows how this will be integrated.

MarkWangChinese, 2024-02-19:
I create one repository as https://github.com/MarkWangChinese/libsbc, please help to check. And one porting related file is put in zephyr repository (modules/libsbc/data_types.h) in pr https://github.com/zephyrproject-rtos/zephyr/pull/31583.

dleach02, 2024-04-10:
@nashif, see #70531 as a reference.

We would like to get things setup. It is unclear to me if we actually voted on this in the TSC or not but I do remember multiple discussions. What should the next steps be?

dleach02, 2024-06-04:
@nashif @carlescufi 

The TSC notes said we were waiting on a PR reference. @MarkWangChinese posted one so need to figure out what we need to do next to get this moving.

nashif, 2024-06-12:
> @nashif @carlescufi
> 
> The TSC notes said we were waiting on a PR reference. @MarkWangChinese posted one so need to figure out what we need to do next to get this moving.

the PR needs to be referenced i nthis issue, there is a section dedicated to that.

MarkWangChinese, 2024-06-13:
> > @nashif @carlescufi
> > The TSC notes said we were waiting on a PR reference. @MarkWangChinese posted one so need to figure out what we need to do next to get this moving.
> 
> the PR needs to be referenced i nthis issue, there is a section dedicated to that.

The issue description is updated, please help to check.

dleach02, 2024-08-20:
@nashif What do we need to do next on this?

mmahadevan108, 2024-10-28:
@dleach02 @nashif I have removed the Stale Label. Has this been discussed in the TSC

dleach02, 2024-12-12:
@MarkWangChinese we discussed this request in TSC today (Dec 11, 2024). Please reopen and refresh your example integration PR #70531

MarkWangChinese, 2024-12-13:
> @MarkWangChinese we discussed this request in TSC today (Dec 11, 2024). Please reopen and refresh your example integration PR #70531

Thanks @dleach02 I create one new pr https://github.com/zephyrproject-rtos/zephyr/pull/82942 because the old pr use discarded  nxp-zephyr repo.

dleach02, 2025-01-08:
@MarkWangChinese, TSC approved your request to add SBC to Zephyr Project. I'll work with the project on some of the logistics of getting this moving. The focus will shift to the actual PRs doing the integration now.

nashif, 2025-01-17:
approved

closed 2025-01-17 by nashif