zephyrproject-rtos/zephyr #67724: I2C property zephyr,concat-buf-size causes failure to initialize I2C device

Yaxit, 2024-01-17:
_Disclaimer: I would have initially filed this issue in the Nordic repository however it seems that issues have been disabled on their end. Correct me if I'm wrong._

**Describe the bug**
It seems that having the device tree node property `zephyr,concat-buf-size = <32>` enabled for the I2C device causes the I2C peripheral to fail initialization.

I just upgraded my firmware for a custom board based on the NRF5340 from NCS v2.3.0 to v2.5.1. I had some issues with my I2C device, and I realized that the I2C peripheral was not initializing properly at startup.
When the aforementioned property is set, my RTT logger shows this
```
[00:00:00.000,640] <err> i2c_nrfx_twim: Failed to initialize device: i2c@8000
*** Booting nRF Connect SDK v2.5.1 ***
... some stuff, then at the first I2C write to the device:
[00:00:03.679,382] <err> i2c_nrfx_twim: Error on I2C line occurred for message 0
ASSERTION FAIL [p_cb->state != NRFX_DRV_STATE_UNINITIALIZED] @ WEST_TOPDIR/modules/hal/nordic/nrfx/drivers/src/nrfx_twim.c:374
[00:00:03.679,443] <err> os: r0/a1:  0x00000004  r1/a2:  0x00000176  r2/a3:  0x2000bb6c
[00:00:03.679,473] <err> os: r3/a4:  0x00000176 r12/ip:  0x0000bf14 r14/lr:  0x0002d3910m
[00:00:03.679,504] <err> os:  xpsr:  0x69100000
[00:00:03.679,534] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x00046ae4  s[ 3]:  0x20016b94
[00:00:03.679,565] <err> os: s[ 4]:  0x00000000  s[ 5]:  0x0004e6ac  s[ 6]:  0x00000023  s[ 7]:  0x0003c42f
[00:00:03.679,595] <err> os: s[ 8]:  0x00000176  s[ 9]:  0x0003cb09  s[10]:  0x00000000  s[11]:  0x20016b94
[00:00:03.679,595] <err> os: s[12]:  0x20016bc0  s[13]:  0x0002d387  s[14]:  0x00046ae4  s[15]:  0x0004e4f8
[00:00:03.679,656] <err> os: fpscr:  0x0004e6ac
[00:00:03.679,687] <err> os: Faulting instruction address (r15/pc): 0x0003caf4
[00:00:03.679,718] <err> os: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
[00:00:03.679,779] <err> os: Current thread: 0x20003048 (main)
[00:00:04.129,791] <err> os: Halting system
```

However, if I remove the node property from the DTS file, there is no problem.
Here is the DTS file that I'm currently using.
```
&i2c0 {
	status = "okay";
	pinctrl-0 = <&i2c_mag_default>;
	pinctrl-1 = <&i2c_mag_sleep>;
	pinctrl-names = "default", "sleep";
        zephyr,concat-buf-size = <32>;
	clock-frequency = < I2C_BITRATE_FAST_PLUS >;

	mag: mag@1e {
		compatible = "st,iis2mdc";
		status = "okay";
		reg = < 0x1e >;
		drdy-gpios = <&gpio1 8 GPIO_ACTIVE_HIGH>;
	};

};
```

**Impact**
Hopefully only an annoyance, as there is no way to directly trace back to the issue from the error messages.
Potentially something worse might be happening

henrikbrixandersen, 2024-01-18:
Thank you for reporting this. However - unless you are able to reproduce this issue with upstream Zephyr `main` - please report issues with the nRF Connect SDK (NCS) on the [Nordic Semiconductor DevZone](https://devzone.nordicsemi.com/).

Also, please use our bug template when reporting bugs. If you can reproduce this with upstream Zephyr, please edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Yaxit, 2024-01-18:
I could replicate this on main Zephyr (9569e44104) with a nrf5340DK board, however it seems I initially did not correctly identified the problem, as I forgot to do a pristine build after changing the property in the DT.
The issue seems to be related to the speed rather than the concatenation buffer. When `clock-frequency` is set to `I2C_SPEED_FAST_PLUS` (1Mbps) which should be supported, the device fails to initialize.
Should I open a new issue or adapt this one?

henrikbrixandersen, 2024-01-18:
> Should I open a new issue or adapt this one?

Feel free to adapt this one.

anangl, 2024-05-27:
Fixed by #72586.

closed 2024-05-27 by anangl