zephyrproject-rtos/zephyr #67741: STM32F4 adc dma crashing

rijesha, 2024-01-17:
**Describe the bug**
I am trying to read 4 adc channels simultaneously with an STM32F411. The adc_read call only allows a single channel to be read with STM32 if not using DMA. This issue has been brought up a few times and the answer has generally been to be use DMA. There are a few issues related to it for example #19882 #34656 #32288 #39640

#52965 Is the pull request that provided STM32 ADC DMA support. None of the code examples I can find work for F4 chips. The test code in test/drivers/adc/adc_dma has code examples for H7, L4 and U5 chips, but not F4 chips. 

The F4 chips use the st,stm32f4-adc and the st,stm32-dma-v1 node.

I tried running the test code with a stm32f4 board, but the MCU always resets when the dma_config function calls api->config(dev, channel, config); (Line 345 of include/zephyr/drivers/dma.h) The adc code calls dma_config from drivers/adc/adc_stm32.c adc_stm32_dma_start function.

DTS File:
```
#include <st/f4/stm32f411Xe.dtsi>
#include <st/f4/stm32f411r(c-e)tx-pinctrl.dtsi>
#include <zephyr/dt-bindings/input/input-event-codes.h>
.
.
.

&adc1 {
	pinctrl-0 = <&adc1_in10_pc0 &adc1_in11_pc1
	&adc1_in12_pc2 &adc1_in13_pc3 &adc1_in1_pa1>;
	pinctrl-names = "default";
	st,adc-clock-source = <SYNC>;
	st,adc-prescaler = <2>;
	status = "okay";

	dmas = < &dma1 0 0 (STM32_DMA_PERIPH_TO_MEMORY |
	STM32_DMA_MEM_INC |  STM32_DMA_MEM_16BITS | STM32_DMA_PERIPH_16BITS) 0x3>;
	dma-names = "dma";

	#address-cells = <1>;
	#size-cells = <0>;

	channel@1 {
		reg = <1>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
	};

	channel@10 {
		reg = <10>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
	};

	channel@11 {
		reg = <11>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
	};

	channel@12 {
		reg = <12>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
	};

	channel@13 {
		reg = <13>;
		zephyr,gain = "ADC_GAIN_1";
		zephyr,reference = "ADC_REF_INTERNAL";
		zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
		zephyr,resolution = <12>;
	};
};

test_dma: &dma1 {
    status = "okay";
};
```

And I have the following in my config file:
```
CONFIG_ADC=y
CONFIG_DMA=y
CONFIG_ADC_STM32_DMA=y
CONFIG_ADC_ASYNC=y
```

**Environment (please complete the following information):**
 - OS: Windows
 - Zephyr 3.4.99

rijesha, 2024-01-17:
I should also mention that I do get this build warning:
```
unit address and first address in 'reg' (0xa) don't match for /soc/adc@40012000/channel@10
unit address and first address in 'reg' (0xb) don't match for /soc/adc@40012000/channel@11
unit address and first address in 'reg' (0xc) don't match for /soc/adc@40012000/channel@12
unit address and first address in 'reg' (0xd) don't match for /soc/adc@40012000/channel@13
```

rijesha, 2024-01-29:
Thanks. I have upgraded to v3.5.0 and am reading the ADC sequence with a separate thread. This seems to be working for now and Ill will probably change to DMA once v3.6.0 comes out.

rijesha, 2024-01-29:
Issue is and will be further resolved in future releases

rijesha, 2024-01-31:
I have done more testing and ran into issues.
The adc_read function with v3.5.0 ends up hanging after a while. I have the code in a while loop and randomly the loop will stall. With some experimenting it seems like the loop stall quicker when zephyr,acquisition-time is set to ADC_ACQ_TIME_DEFAULT rather than ADC_ACQ_TIME_MAX.

```
while (true)
{
	int ret = adc_read_dt(&adc_channels[0], &sequence);
	if (ret == 0)
	{
		got_new_data = true;
	}
	k_msleep(10);
}
```

I have also tried the latest main with DMA. That is not working either. The STM32F4 doesn't have the function LL_ADC_REG_StopConversion. This function is called on line 770 of drivers/adc/adc_stm32.c. 
I commented out the line. With the line commented out the signal is never set and the buffer doesnt fill with data.

gautierg-st, 2024-02-01:
> The adc_read function with v3.5.0 ends up hanging after a while. I have the code in a while loop and randomly the loop will stall. With some experimenting it seems like the loop stall quicker when zephyr,acquisition-time is set to ADC_ACQ_TIME_DEFAULT rather than ADC_ACQ_TIME_MAX.

Yes, if you look at the status register, I think you will see the overflow bit is set. That means the ADC has made another measurement while the data buffer was still full, i.e. the data is not read fast enough.
Dealing with the overflow error is a pain and I think the best way to avoid it is to set the timings correctly (acquisition time and the clock source/prescaler) to prevent it happening.

> I have also tried the latest main with DMA. That is not working either. The STM32F4 doesn't have the function LL_ADC_REG_StopConversion. This function is called on line 770 of [...]

Noted. I'll try to have a look next week.

gautierg-st, 2024-02-07:
Hi @rijesha,
I've just pushed a PR to add DMA support for STM32F4 ADC here: #68685
Can you have a look and tell me if that solves your problem?

rijesha, 2024-02-09:
Hi @gautierg-st. 
The PR works for me with an STM32F411 with 5 adc input channels. Thanks for your help in the matter.

erwango, 2024-02-09:
Re-opening the time #68685 is merged

PR #68685 (2024-02-12): STM32F4 ADC DMA support
This PR adds the support of the DMA for the STM32F4 ADC (valid also for STM32F2, F7 and L1).
Also adds an ADC DMA test for Nucleo F401RE.

Fixes #67741