zephyrproject-rtos/zephyr #67746: Nominate mrmarteng as zephyr contributor

MrMarteng, 2024-01-17:
# Background

The [TSC Project Roles] defines the main roles for the Zephyr Project, including
Maintainer, Collaborator, and Contributor.

By default anyone that contributes code or documentation is a Contributor, but
with the lowest [GitHub Permission Level] of Read. For example, Contributors
with Read permission do not have the permission to add reviewers to a pull
request.

Use this template to nominate a GitHub user for the Contributor role with
Triage permission level, which allows the user to add reviewers to a pull
request and be added as a reviewer by other users.

# Nomination

## GitHub User

Provide the following information about the GitHub user:

1. Full Name: Martin Kiepfer
1. GitHub username: MrMarteng
1. Organization (optional)

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/pull/61812
https://github.com/zephyrproject-rtos/zephyr/pull/58809
https://github.com/zephyrproject-rtos/zephyr/pull/59768
https://github.com/zephyrproject-rtos/zephyr/pull/67130
https://github.com/zephyrproject-rtos/zephyr/pull/67084
https://github.com/zephyrproject-rtos/zephyr/issues/67277
https://github.com/zephyrproject-rtos/zephyr/issues/61866

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

kartben, 2024-01-18:
Yes please! Thanks!

nashif, 2024-01-23:
Please check https://github.com/zephyrproject-rtos/zephyr/invitations

closed 2024-01-23 by nashif