zephyrproject-rtos/zephyr #67759: ESP32-S3 Code and Data Relocation Support

EricNRS, 2024-01-18:
**Describe the bug**
The code and data relocation feature (https://docs.zephyrproject.org/latest/kernel/code-relocation.html ) does not work for ESP32-S3 (and likely other Xtensa ESP chips) due to the `default.ld` segment names including an underscore ('_') and the lack of including `linker_relocate.ld` in the `default.ld` as is done for riscv, cortex_m, and cortex_a architectures.

For example, trying this on ESP32-S3:

prj.conf
`CONFIG_CODE_DATA_RELOCATION=y`

CMakeLists.txt
`zephyr_code_relocate(LIBRARY drivers__sensor LOCATION ext_ram_seg)`

Doing a build results in the error:
```sh
Memory region:  ext_ram_seg|COPY
Traceback (most recent call last):
  File "/zephyrproject/zephyr/scripts/build/gen_relocate_app.py", line 614, in <module>
    main()
  File "/zephyrproject/zephyr/scripts/build/gen_relocate_app.py", line 594, in main
    sections_by_category = assign_to_correct_mem_region(memory_type, full_list_of_sections)
  File "/zephyrproject/zephyr/scripts/build/gen_relocate_app.py", line 284, in assign_to_correct_mem_region
    mpu_align[memory_region] = int(align_size)
ValueError: invalid literal for int() with base 10: 'ram_seg|COPY'
ninja: build stopped: subcommand failed.
```

The root cause is that the tool has a naming convention of using an underscore '_' to separate the segment name from one or more section names.

https://docs.zephyrproject.org/latest/kernel/code-relocation.html

As a work-around, the relocation can be done manually by adding an extra linker descriptor file:

CMakeLists.txt
`zephyr_linker_sources(SECTIONS custom-sections.ld)`

custom-sections.ld
```sh
SECTION_DATA_PROLOGUE(_CUSTOM_SECTION_NAME2,,)
{
	. = ALIGN(4);
	__custom_section_start = .;
	KEEP(*(".custom_section.*"));
	__custom_section_end = ALIGN(4);
} GROUP_DATA_LINK_IN(RAMABLE_REGION, ROMABLE_REGION)
```

One user posted this workaround in Discord:  https://discord.com/channels/720317445772017664/883444902971727882/1173967050805346384

**Impact**
Failure of the code and data relocation support requiring work-arounds.

EricNRS, 2024-04-07:
The cleanup bot strikes again.  It would be good to get this cleaned up and into the v3.7 release, although it is lower priority than other Espressif tasks at the moment.

EricNRS, 2024-06-13:
Retested with 3.6.99 and this is still an issue.

EricNRS, 2024-06-14:
The external RAM (SPIRAM) is also currently broken and is being tracked as https://github.com/zephyrproject-rtos/zephyr/issues/74284

kartben, 2024-06-14:
re-opened as per @EricNRS request

EricNRS, 2024-08-14:
@sylvioalves - any chance we can get this issue resolved in the current cleanup efforts?

EricNRS, 2024-10-14:
Commenting to keep this issue open.

EricNRS, 2024-12-15:
Easy there stalebot, perfection takes time.  Let's keep this issue open.

EricNRS, 2025-02-17:
I haven't retested this on the latest tip revision, but I think it is still not completely fixed.  In the meantime, there are partial fixes and manual work-arounds here:

- https://github.com/zephyrproject-rtos/zephyr/issues/74284
- https://github.com/zephyrproject-rtos/zephyr/pull/83642

markand, 2025-02-26:
I may confirm that all chips seem to be affected. In fact, the current linkerscript are not even modified by devicetree/kconfig which means that adding any random value in both `&sram0`'s `reg` property and `SRAM_BASE_ADDRESS` / `SRAM_SIZE` don't have any effect on the linker script.

EricNRS, 2025-02-26:
> I may confirm that all chips seem to be affected. In fact, the current linker scripts are not even modified by devicetree/kconfig which means that adding any random value in both `&sram0`'s `reg` property and `SRAM_BASE_ADDRESS` / `SRAM_SIZE` don't have any effect on the linker script.

That is a different issue from this task and in reality, the SRAM configuration is fixed due to the architecture of the ESP Xtensa chips and cannot be made easily configurable.  For example, the ESP32-S3 has SRAM regions 0, 1, and 2 with SRAM0 being instruction bus only and SRAM2 being data bus only.  See table 15-3 from the ESP32-S3 TRM:

[image]

Because of these limitations and having to also place the instruction and data caches, there is little benefit to making the SRAM_* settings configurable.

closed 2025-05-12 by github-actions[bot]