zephyrproject-rtos/zephyr #67811: benchmarks: latency_measure:  significant increase user.latency on THREAD start.user.from metrics

golowanow, 2024-01-18:
**Describe the bug**
40% increase of  `THREAD start.user.from.kernel/user` metrics collected by `tests/benchmarks/latency_measure/benchmark.user.latency.*` on `npcx9m6f_evb` board
The change is observed comparing results using 92af1721 vs. 38010c1 (05--15/Jan/2024).
Possibly related to #67438 or has the same root cause.

**To Reproduce**
Steps to reproduce the behavior:
1. have `npcx9m6f_evb` board connected and described in `map.yaml` file
2. run `./scripts/twister -T tests/benchmarks/latency_measure -p npcx9m6f_evb  --device-testing -v --inline-logs --hafdware-map map.yaml`
3. check twister.log for `THREAD start.user.from.*` metrics collected.

**Expected behavior**
no latency degradation

**Impact**
slow performance or the `latency_measure` benchmark test issue.

**Logs and console output**

with 38010c1 (5/Jan)
```
npcx9m6f_evb/tests/benchmarks/latency_measure/benchmark.user.latency.timeslicing/zephyr
twister - DEBUG - DEVICE: THREAD start.user.from.kernel                       :    4992 cycles ,   332832 ns :
twister - DEBUG - DEVICE: THREAD start.user.from.user                         :    5230 cycles ,   348732 ns :

npcx9m6f_evb/tests/benchmarks/latency_measure/benchmark.user.latency.objcore.stats
twister - DEBUG - DEVICE: THREAD start.user.from.kernel                       :    5113 cycles ,   340899 ns :
twister - DEBUG - DEVICE: THREAD start.user.from.user                         :    5351 cycles ,   356798 ns :

npcx9m6f_evb/tests/benchmarks/latency_measure/benchmark.user.latency
twister - DEBUG - DEVICE: THREAD start.user.from.kernel                       :    4878 cycles ,   325232 ns :
twister - DEBUG - DEVICE: THREAD start.user.from.user                         :    5116 cycles ,   341132 ns :
```

with 92af1721 (15/Jan)
```
npcx9m6f_evb/tests/benchmarks/latency_measure/benchmark.user.latency.timeslicing/zephyr
twister - DEBUG - DEVICE: THREAD start.user.from.kernel                       :    8579 cycles ,   571965 ns :
twister - DEBUG - DEVICE: THREAD start.user.from.user                         :    8817 cycles ,   587865 ns :

npcx9m6f_evb/tests/benchmarks/latency_measure/benchmark.user.latency.objcore.stats
twister - DEBUG - DEVICE: THREAD start.user.from.kernel                       :    7692 cycles ,   512832 ns :
twister - DEBUG - DEVICE: THREAD start.user.from.user                         :    7930 cycles ,   528732 ns :

npcx9m6f_evb/tests/benchmarks/latency_measure/benchmark.user.latency
twister - DEBUG - DEVICE: THREAD start.user.from.kernel                       :    8465 cycles ,   564365 ns :
twister - DEBUG - DEVICE: THREAD start.user.from.user                         :    8703 cycles ,   580265 ns :
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.1
 - Commit SHA or Version used:  92af172 ^38010c1

**Additional context**
the issue was found analyzing benchmark recordings with additional metric name split using #67297

nashif, 2024-01-18:
would be nice if you could point to exact commit causing this.

nashif, 2024-01-18:
This is related to changes in the benchmark itself, not related to this specific platform and the increase is due to stack increase.

peter-mitsis, 2024-01-18:
In the commit that led to this observed behavior, events were added. This necessitated an increase to the stack size. This in turn causes an increase in the time to start a user thread because internally memset() is called to initialize the stack memory. Larger stack = more memory to initialize = longer start time.

closed 2024-01-18 by peter-mitsis

golowanow, 2024-01-18:
> would be nice if you could point to exact commit causing this.

looks like the issue is caused by a8914c5
few log snippets in a minute