zephyrproject-rtos/zephyr #67812: usb_descriptor: Fixing EP address: unnecessary

KozhinovAlexander, 2024-01-18:
**Describe the bug**
A clear and concise description of what the bug is.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
    - Dev OS: `Linux <my_vm_name_here> 6.5.0-14-generic #14~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 20 18:15:30 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
    - Zephyr RTOS: `*** Booting Zephyr OS build zephyr-v3.5.0-4378-g2794727c9724 ***` --> __NOTE__: Here githash may vary from that one of main branch! But this shall not change the core of the bug at the moment!
   
 - What have you tried to diagnose or workaround this issue?
     - Build sample project `samples/subsys/usb/testusb` and run it wit hdifferent USB Endpoint addresses
     - Notice no USB EP address change on my development machine
 
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
    - No. I do not. I need here help!

**To Reproduce**
Steps to reproduce current behavior:
1. Checkout curent `main` branch with `git checkout main`
2. (optional step) Add USB device to your board. In my case of nucelo_h745zi_q_m7 I've applided following patch:
```
diff --git a/boards/arm/nucleo_h745zi_q/nucleo_h745zi_q_m7.dts b/boards/arm/nucleo_h745zi_q/nucleo_h745zi_q_m7.dts
index 184232867b..c8ed5f41b3 100644
--- a/boards/arm/nucleo_h745zi_q/nucleo_h745zi_q_m7.dts
+++ b/boards/arm/nucleo_h745zi_q/nucleo_h745zi_q_m7.dts
@@ -26,6 +26,10 @@
 		zephyr,dtcm = &dtcm;
 		zephyr,sram = &sram0;
 		zephyr,flash = &flash0;
+		zephyr,usb-device = &usbd;
+	};
+
+	usbd: usbotg_fs {
 	};
 
 	pwmleds {
@@ -121,3 +125,12 @@
 &rng {
 	status = "okay";
 };
+
+&usbotg_fs {
+	pinctrl-0 = <&usb_otg_fs_sof_pa8
+				 &usb_otg_fs_vbus_pa9
+				 &usb_otg_fs_dm_pa11
+				 &usb_otg_fs_dp_pa12>;
+	pinctrl-names = "default";
+	status = "okay";
+};
```
3. Enable USB debug log. You may use here the patch below:
````
diff --git a/samples/subsys/usb/testusb/prj.conf b/samples/subsys/usb/testusb/prj.conf
index e310a67818..e34ca0104f 100644
--- a/samples/subsys/usb/testusb/prj.conf
+++ b/samples/subsys/usb/testusb/prj.conf
@@ -9,3 +9,6 @@ CONFIG_USB_DRIVER_LOG_LEVEL_ERR=y
 CONFIG_USB_DEVICE_LOOPBACK=y
 CONFIG_USB_DEVICE_LOG_LEVEL_ERR=y
 CONFIG_USB_DEVICE_INITIALIZE_AT_BOOT=n
+
+CONFIG_USB_DRIVER_LOG_LEVEL_DBG=y
+CONFIG_USB_DEVICE_LOG_LEVEL_DBG=y
````
5. Build application `samples/subsys/usb/testusb`
6. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

jfischer-no, 2024-01-19:
> The `usb_validate_ep_cfg_data` function in `subsys/usb/device/usb_descriptor.c` silently overwrites endpoint address!

This is intentional behavior and not a bug.

>Showstoper: Implementing device driver for already existing usb drivers in Linux Kernel for example may need additional endpoint entries in SoC, which may consume more RAM.

Looks like your host-side driver is not parsing function descriptors, here endpoint descriptors, but using fixed addresses. Please fix your broken host-side driver.

I do not accept this report. Closing.

KozhinovAlexander, 2024-01-19:
> This is intentional behavior and not a bug.
>
>
> Looks like your host-side driver is not parsing function descriptors, here endpoint descriptors, but using fixed addresses. Please fix your broken [...]

Can you please state the entry in USB standard, which defines this behaviour? Closing this issue with the reason `intentional behavior and not a bug` seems to me to be very subective reason.

I have a lot of consumer devices with exactly opposit behaviour as already implemented int Zephyr Stack!

1. Samsung Flash Drive:
````
T:  Bus=04 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  3 Spd=5000 MxCh= 0
D:  Ver= 3.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=090c ProdID=1000 Rev=11.00
S:  Manufacturer=Samsung
S:  Product=Flash Drive
S:  SerialNumber=0361415120013210
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=304mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=usb-storage
E:  Ad=01(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
````

2. ASRock LED Controller:
````
T:  Bus=03 Lev=02 Prnt=03 Port=00 Cnt=01 Dev#= 93 Spd=12   MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=26ce ProdID=01a2 Rev= 0.00
S:  Manufacturer=ASRock
S:  Product=LED Controller
S:  SerialNumber=A02019100900
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=02(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
````

3. Gamin Watch:
````
T:  Bus=03 Lev=02 Prnt=04 Port=00 Cnt=01 Dev#= 94 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=ff(vend.) Sub=ff Prot=ff MxPS=64 #Cfgs=  1
P:  Vendor=091e ProdID=0003 Rev= 0.01
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=82(I) Atr=03(Int.) MxPS=  64 Ivl=125us
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
````

4. Samsung Portable SSD T5:
````
T:  Bus=04 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  4 Spd=5000 MxCh= 0
D:  Ver= 3.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 9 #Cfgs=  1
P:  Vendor=04e8 ProdID=61f5 Rev= 1.00
S:  Manufacturer=Samsung
S:  Product=Portable SSD T5
S:  SerialNumber=1234567BDBA5
C:* #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=896mA
I:  If#= 0 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=uas
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
I:* If#= 0 Alt= 1 #EPs= 4 Cls=08(stor.) Sub=06 Prot=62 Driver=uas
E:  Ad=81(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=83(I) Atr=02(Bulk) MxPS=1024 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS=1024 Ivl=0ms
````

All above devices have NO bound between EP I/O addresses! What makes Zephyr RTOS here different? Why Zephyr RTOS have to follow this specific behaviour?

jfischer-no, 2024-01-19:
> All the above devices have NO bound between EP I/O addresses and theri indexes in the config fields! What makes Zephyr RTOS here different? Why Zephyr RTOS have to follow this specific behaviour?
> 
> Following your statement about broken host driver, would mean all the above devices having broken host drivers too. Ok. I would believe if we speak about Garmin Watch. But what about all the Samsung devices? Would you say, Samsung doing broken USB host drivers? Who decides it?
> 

My statement only refers to your original report, everything else you hallucinated is not part of it.

> Now I have even more questions. I reopen this issue in the hope all the other people with additional expertise in USB stack and USB standard take a look here. It looks to me know, tha Zephyr RTOS USB stack serving needs of one consumer only. Moreover I see this behavior as a desgin failure bloking it's flexibility.

Okay, still, if you want to discuss, open a discussion.

> @carlescufi Could you please add more peaople for review and discussion?

I am the maintainer of this area and do not accept this bug. Pinging my boss here will not help.

KozhinovAlexander, 2024-01-19:
@jfischer-no I didn't knew @carlescufi is your superior. I know him, as a person responsible for a lot of management in Zephyr RTOS. Thus I requested him to add more reviewers.

Since you're the maintainer of the USB stack, you shall be more familiar with USB standards than me. Thus I am asking your for a reference in the USB standard, where behavior I mentioned is defined. What I am getting instead is silent closing of my bug-report and mean (see the word "hallucinating" above). 

Opening a discussion would be also fine. But you didn't mentioned it in the first bug-closing.

I also do not like, how you behave towards me and my striving to make Zephyr RTOS better. Zephyr RTOS writes it follows respectful communication. In that case I have to notify responsible people.

3Nigma, 2024-01-25:
I too find the `if (!usb_dc_ep_check_cap(&ep_cfg))` handling at least semantically problematic to say the least. 

The [declaration of `usb_dc_ep_check_cap` reads](https://github.com/zephyrproject-rtos/zephyr/blob/0176a27474e029bb11e64c5235b4a33cc8bfdb96/include/zephyr/drivers/usb/usb_dc.h#L200):
```
 Function to check capabilities of an endpoint. usb_dc_ep_cfg_data structure
 provides the endpoint configuration parameters: endpoint address,
 endpoint maximum packet size and endpoint type.
 The driver should check endpoint capabilities and return 0 if the
 endpoint configuration is possible.
```
I would like to emphasise here the "return 0 if the **endpoint configuration is possible**" part.

Why would you do a "Fixing EP address" if the configuration is possible ?

I too lost quite a considerable amount of hours dealing with this inconsistency which, as @jfischer-no mentioned, silently changes the endpoint number.

jfischer-no, 2024-01-25:
> I too find the `if (!usb_dc_ep_check_cap(&ep_cfg))` handling at least semantically problematic to say the least.
>
> The [declaration of `usb_dc_ep_check_cap` [...]

Stack, not the driver API you are citing, fixes endpoint address. Additionally endpoint does not have to be occupied. If the endpoint is free and the desired configuration is possible, then reassign the address. Why? Because it makes absolute sense to distribute resources as efficiently as possible and anything else would be silly.

> I too lost quite a considerable amount of hours dealing with this inconsistency which, as @jfischer-no mentioned, silently changes the endpoint number.

You would not do this if your host function driver were checking the endpoint descriptor.

KozhinovAlexander, 2024-01-27:
> Stack, not the driver API you are citing, fixes endpoint address. Additionally endpoint does not have to be occupied. If the endpoint is free and the desired configuration is possible, then reassign [...]
I am still confused by this behavior.
I see Linux Kernel doing exactly opposite for years. Could you please tell why the're not doing endpoint address fix like Zephyr's stack?
Could you also please show any host endpoint driver doing the right way of addressing according to rules you describe?

> > I too lost quite a considerable amount of hours dealing with this inconsistency which, as @jfischer-no mentioned, silently changes the endpoint number.
> 
> You would not do this if your host function driver were checking the endpoint descriptor.

KozhinovAlexander, 2024-02-09:
I reopen it. cause there is still no solution to the problem and my questions, ideas etc. seemst to be not answered in a manner I am asking for.

jfischer-no, 2024-02-15:
> I reopen it, cause there is still no solution to the problem and my questions, ideas etc. seems to be not answered in a manner I am asking for. Moving information of this issue does not give any resonance from the community. Moreover - discussions seems to me to be not effective.

I have already explained to you that this is not a bug in Zephyr, but in your application. I also know that you are aware of the documentation enhancement that explains this in more detail. Stop spamming this bug report.

closed 2024-02-15 by KozhinovAlexander