zephyrproject-rtos/zephyr #67824: samples/bluetooth/broadcast_audio_source: FLASH overflow when built for nrf5340dk_nrf5340_cpuapp

gmarull, 2024-01-19:
**Describe the bug**

`samples/bluetooth/broadcast_audio_source` overflows FLASH region when built for `nrf5340dk_nrf5340_cpuapp`

**To Reproduce**

`west build -b nrf5340dk_nrf5340_cpuapp samples/bluetooth/broadcast_audio_source -T sample.bluetooth.broadcast_audio_source -p`

**Expected behavior**
Sample builds successfully

**Impact**
Sample broken, impacts CI for other PRs (see https://github.com/zephyrproject-rtos/zephyr/actions/runs/7574708555/job/20629782144?pr=67673).

**Logs and console output**

```
...
Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_INTEGER_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /home/gmarull/ws/nordic/zephyrproject/zephyr/build/hci_ipc/zephyr && /usr/bin/cmake -E true
/home/gmarull/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/zephyr_pre0.elf section `datas' will not fit in region `FLASH'
/home/gmarull/zephyr-sdk-0.16.1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: region `FLASH' overflowed by 300 bytes
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[13/16] Completed 'broadcast_audio_source'
FAILED: hci_ipc-prefix/src/hci_ipc-stamp/hci_ipc-build /home/gmarull/ws/nordic/zephyrproject/zephyr/build/hci_ipc-prefix/src/hci_ipc-stamp/hci_ipc-build 
cd /home/gmarull/ws/nordic/zephyrproject/zephyr/build/hci_ipc && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.1
 - Commit SHA or Version used b6e48d04f62d3aeb2afa40a2fab38baae3ea5709

**Additional context**
N/A

cvinayak, 2024-01-19:
Bisect points to: https://github.com/zephyrproject-rtos/zephyr/commit/4cc213bc8d17220a2569ff8bb7548d6ed03f7d2f
 
and has this in the PR: 
> New shim takes more ROM. Until it is fixed use legacy shim in memory constraint case.

Can we switch to legacy shim as default, else many samples need to be updated explicitly to not fail flash size when we update code.

Author has definitely stated that new shim needs improvement... default can be switch when the improvements are in place, right?

cvinayak, 2024-01-19:
@nordic-krch could you switch the defaults back to legacy shim?

PR #67828 (2024-01-19): drivers: serial: nrfx_uarte: Use legacy shim by default
Fixes #67824.
closed 2024-01-19 by aescolar