zephyrproject-rtos/zephyr #67833: it8xxx2_evb: kernel: tests/kernel/timer/timer_behavior/kernel.timer.timer fails.

gbarkadiusz, 2024-01-19:
**Describe the bug**
 _test_jitter_drift_timer_period_
```
do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
```
_test_jitter_drift_timer_startdelay_
```
do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
```

**To Reproduce**
1. cd <path/to>zephyrproject/zephyr
2. source zephyr-env.sh
3. twister --device-testing --hardware-map <path/to>/map.yaml -T tests/kernel/timer/timer_behavior/kernel.timer.timer -vv

**Expected behavior**
Test pass or is skipped

**Impact**
annoyance

**Logs and console output**
`<path_to>/twister-out/it8xxx2_evb/tests/kernel/timer/timer_behavior/kernel.timer.timer/handler.log`
```
***** delaying boot 500ms (per build configuration) *****
*** Booting Zephyr OS build zephyr-v3.5.0-4305-g95958f7e6db2 (delayed boot 500ms) ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 6 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096
expected period drift: 220.70312 us
period duration statistics for 5000 samples (0 rollovers):
expected: 1000 us,       	32.768000 cycles
min:      1098.632812 us, 	36 cycles
max:      1342.773438 us, 	44 cycles
mean:     1220.709229 us, 	40.000200 cycles
variance: 14899.857305 us, 	15.998600 cycles
stddev:   122.064972 us, 	3.999825 cycles
timer start cycle 17591, end cycle 217592,
total time 6103546.142578 us, expected time 5000000.000000 us,
expected time drift 1103515.625000 us, difference 30.517578 us
Assertion failed at <path_to>/tests/kernel/timer/timer_behavior/src/jitter_drift.c:274: do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
FAIL - test_jitter_drift_timer_period in 6.930 seconds
===================================================================
START - test_jitter_drift_timer_startdelay
periodic timer behavior test using explicit start with delay
collecting time samples for approx 6 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096
expected period drift: 220.70312 us
period duration statistics for 5000 samples (0 rollovers):
expected: 1000 us,       	32.768000 cycles
min:      1098.632812 us, 	36 cycles
max:      1342.773438 us, 	44 cycles
mean:     1220.660400 us, 	39.998600 cycles
variance: 14899.855517 us, 	15.998598 cycles
stddev:   122.064964 us, 	3.999825 cycles
timer start cycle 245130, end cycle 445123,
total time 6103302.001953 us, expected time 5000000.000000 us,
expected time drift 1103515.625000 us, difference -213.623047 us
Assertion failed at <path_to>/tests/kernel/timer/timer_behavior/src/jitter_drift.c:274: do_test_using: (min_us >= min_us_bound is false)
Shortest timer period too short (off by more than expected 10%)
FAIL - test_jitter_drift_timer_startdelay in 6.930 seconds
===================================================================
TESTSUITE timer_jitter_drift failed.
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 4096Hz, Expecting 10240 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 0
Faking busy work, remaining timers is 4, timer callbacks 263
Faking busy work, remaining timers is 4, timer callbacks 525
Faking busy work, remaining timers is 4, timer callbacks 787
Faking busy work, remaining timers is 4, timer callbacks 1049
Faking busy work, remaining timers is 4, timer callbacks 1312
Faking busy work, remaining timers is 4, timer callbacks 1574
Faking busy work, remaining timers is 4, timer callbacks 1836
Faking busy work, remaining timers is 4, timer callbacks 2098
Faking busy work, remaining timers is 4, timer callbacks 2361
Faking busy work, remaining timers is 4, timer callbacks 2623
Faking busy work, remaining timers is 4, timer callbacks 2886
Faking busy work, remaining timers is 4, timer callbacks 3148
Faking busy work, remaining timers is 4, timer callbacks 3410
Faking busy work, remaining timers is 4, timer callbacks 3673
Faking busy work, remaining timers is 4, timer callbacks 3935
Faking busy work, remaining timers is 4, timer callbacks 4197
Faking busy work, remaining timers is 4, timer callbacks 4460
Faking busy work, remaining timers is 4, timer callbacks 4722
Faking busy work, remaining timers is 4, timer callbacks 4984
Faking busy work, remaining timers is 4, timer callbacks 5247
Faking busy work, remaining timers is 4, timer callbacks 5509
Faking busy work, remaining timers is 4, timer callbacks 5772
Faking busy work, remaining timers is 4, timer callbacks 6034
Faking busy work, remaining timers is 4, timer callbacks 6296
Faking busy work, remaining timers is 4, timer callbacks 6559
Faking busy work, remaining timers is 4, timer callbacks 6821
Faking busy work, remaining timers is 4, timer callbacks 7083
Faking busy work, remaining timers is 4, timer callbacks 7346
Faking busy work, remaining timers is 4, timer callbacks 7609
Faking busy work, remaining timers is 4, timer callbacks 7871
Faking busy work, remaining timers is 4, timer callbacks 8133
Faking busy work, remaining timers is 4, timer callbacks 8395
Faking busy work, remaining timers is 4, timer callbacks 8658
Faking busy work, remaining timers is 4, timer callbacks 8920
Faking busy work, remaining timers is 4, timer callbacks 9183
Faking busy work, remaining timers is 4, timer callbacks 9445
Faking busy work, remaining timers is 4, timer callbacks 9707
Faking busy work, remaining timers is 4, timer callbacks 9970
Faking busy work, remaining timers is 4, timer callbacks 10232
One Tick Timer Train Done, took 10248 ms, busy loop ran 40 times
Perfect delta 32 cycles or 977 us
Timer 0 max delta 44 cycles or 1343 us, 41 late callbacks (0.0%)
Timer 1 max delta 45 cycles or 1373 us, 42 late callbacks (0.0%)
Timer 2 max delta 46 cycles or 1404 us, 38 late callbacks (0.0%)
Timer 3 max delta 46 cycles or 1404 us, 42 late callbacks (0.0%)
!! Some ticks were missed.
!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.
PASS - test_one_tick_timer_train in 10.298 seconds
===================================================================
TESTSUITE timer_tick_train succeeded
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [timer_jitter_drift]: pass = 0, fail = 2, skip = 0, total = 2 duration = 13.860 seconds
- FAIL - [timer_jitter_drift.test_jitter_drift_timer_period] duration = 6.930 seconds
- FAIL - [timer_jitter_drift.test_jitter_drift_timer_startdelay] duration = 6.930 seconds
SUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.298 seconds
- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.298 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 750de1ec4fd5c10be2577d16a3673549
PROJECT EXECUTION FAILED
```

**Environment:**
 - OS: Linux
 - Toolchain: zephyr-sdk-0.16.4
 - Commit SHA or Version used: Zephyr-v3.5.0-4305-g95958f7e6db2

Dino-Li, 2024-01-23:
@Chenhongren please take a look at this issue.

teburd, 2024-01-26:
It looks like the scheduled timers are shifted by a significant amount, the mean timer should be *very* close to the desired timeout which it is not.

The message saying the shorted timer is too short is incorrect here, its actually over 10% too long!

Dino-Li, 2024-02-16:
@RuibinChang , add Ruibin to this loop.

Chenhongren, 2024-03-18:
hi @gbarkadiusz
This issue is fixed by pr#69891. Could you please help verify it again? Thanks.

gbarkadiusz, 2024-03-18:
It has been fixed by https://github.com/zephyrproject-rtos/zephyr/pull/69891.
So, I'm closing it.

gbarkadiusz, 2024-03-19:
According to my investigation.
This bug was fixed with sdk ver. zephyr-sdk-0.16.4 but it returns with sdk ver. zephyr-sdk-0.16.5-1.
There are logs below.
**To reproduce**
```
$ export ZEPHYR_SDK_INSTALL_DIR=<path_to>/zephyr-sdk-0.16.5-1/
$ twister -p it8xxx2_evb --device-testing -T tests/kernel/timer/timer_behavior/ --west-flash=<path_to>loader3.sh --device-serial=<serial_port> -vv
```
```
cat <path_to>/zephyrproject/zephyr/twister-out/it8xxx2_evb/tests/kernel/timer/timer_behavior/kernel.timer.timer/handler.log
```
```
***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build v3.6.0-1133-g2f6b41a694ee (delayed boot 1ms) ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 6 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096
expected period drift: 220.70312 us
period duration statistics for 5000 samples (0 rollovers):
expected: 1000 us,              32.768000 cycles
min:      1159.667969 us,       38 cycles
max:      1220.703125 us,       40 cycles
mean:     1220.684814 us,       39.999400 cycles
variance: 0.930987 us,  0.001000 cycles
stddev:   0.964877 us,  0.031617 cycles
timer start cycle 1206, end cycle 201203,
total time 6103424.072266 us, expected time 5000000.000000 us,
expected time drift 1103515.625000 us, difference -91.552734 us
PASS - test_jitter_drift_timer_period in 6.898 seconds
===================================================================
START - test_jitter_drift_timer_startdelay
periodic timer behavior test using explicit start with delay
collecting time samples for approx 6 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096
expected period drift: 220.70312 us
period duration statistics for 5000 samples (0 rollovers):
expected: 1000 us,              32.768000 cycles
min:      1190.185547 us,       39 cycles
max:      1251.220703 us,       41 cycles
mean:     1220.703125 us,       40.000000 cycles
variance: 309.944153 us,        0.332800 cycles
stddev:   17.605230 us,         0.576888 cycles
timer start cycle 227700, end cycle 427700,
total time 6103515.625000 us, expected time 5000000.000000 us,
expected time drift 1103515.625000 us, difference 0.000000 us
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/timer/timer_behavior/src/jitter_drift.c:284: do_test_using: (stddev_us < (double)CONFIG_TIMER_TEST_MAX_STDDEV is false)
Standard deviation (in microseconds) outside expected bound
FAIL - test_jitter_drift_timer_startdelay in 6.919 seconds
===================================================================
TESTSUITE timer_jitter_drift failed.
Running TESTSUITE timer_tick_train
===================================================================
START - test_one_tick_timer_train
Initializing 4 Timers, Tick Rate 4096Hz, Expecting 10240 callbacks in 11000 ms
Starting Timers with Skews
Faking busy work, remaining timers is 4, timer callbacks 0
Faking busy work, remaining timers is 4, timer callbacks 265
Faking busy work, remaining timers is 4, timer callbacks 530
Faking busy work, remaining timers is 4, timer callbacks 794
Faking busy work, remaining timers is 4, timer callbacks 1060
Faking busy work, remaining timers is 4, timer callbacks 1325
Faking busy work, remaining timers is 4, timer callbacks 1590
Faking busy work, remaining timers is 4, timer callbacks 1856
Faking busy work, remaining timers is 4, timer callbacks 2121
Faking busy work, remaining timers is 4, timer callbacks 2386
[... 143 lines cut ...]
Faking busy work, remaining timers is 4, timer callbacks 9523
Faking busy work, remaining timers is 4, timer callbacks 9788
Faking busy work, remaining timers is 4, timer callbacks 10052
One Tick Timer Train Done, took 10075 ms, busy loop ran 39 times
Perfect delta 32 cycles or 977 us
Timer 0 max delta 42 cycles or 1282 us, 20 late callbacks (0.0%)
Timer 1 max delta 43 cycles or 1312 us, 23 late callbacks (0.0%)
Timer 2 max delta 42 cycles or 1282 us, 20 late callbacks (0.0%)
Timer 3 max delta 42 cycles or 1282 us, 22 late callbacks (0.0%)
!! Some ticks were missed.
!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.
PASS - test_one_tick_timer_train in 10.125 seconds
===================================================================
TESTSUITE timer_tick_train succeeded
------ TESTSUITE SUMMARY START ------
SUITE PASS - 100.00% [timer_jitter_drift]: pass = 2, fail = 0, skip = 0, total = 2 duration = 13.817 seconds
- PASS - [timer_jitter_drift.test_jitter_drift_timer_period] duration = 6.920 seconds
- PASS - [timer_jitter_drift.test_jitter_drift_timer_startdelay] duration = 6.897 seconds
SUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.125 seconds
- PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.125 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 2bf547326686414625ada2bddf7ccc4d
PROJECT EXECUTION SUCCESSFUL
```

gbarkadiusz, 2024-04-05:
Hi @Chenhongren 
I have got 100% reproduction rate.

**To Reproduce**
1) export ZEPHYR_SDK_INSTALL_DIR=<path_t0>/zephyr-sdk-0.16.5-1
2) west build -p always -b it8xxx2_evb <path_to>/tests/kernel/timer/timer_behavior/ -T kernel.timer.timer
3) sudo ./ite -f <path_to>zephyrproject/zephyr/build/zephyr/zephyr.bin

**Details**
SDK: zephyr-sdk-0.16.5-1
Zephyr_version: v3.6.0-1133-g2f6b41a694ee

`<path_to>/build.log`
```

```-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: <path_to>/zephyrproject/zephyr/tests/kernel/timer/timer_behavior
-- CMake version: 3.22.1
...
-- Zephyr version: 3.6.99  (<path_to>/zephyrproject/zephyr)
-- Found west (found suitable version "1.1.0", minimum required is "0.14.0")
-- Board: it8xxx2_evb, identifier: it81302bx
-- Found host-tools: zephyr 0.16.5 ( <path_to>/zephyr/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 ( <path_to>/zephyr/zephyr-sdk-0.16.5-1)
...
-- Found GnuLd: <path_to>/zephyr-sdk-0.16.5-1/riscv64-zephyr-elf/bin/../lib/gcc/riscv64-zephyr-elf/12.2.0/../../../../riscv64-zephyr-elf/bin/ld.bfd (found version "2.38") 
...
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: <path_to>/zephyr-sdk-0.16.5-1/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done
-- Generating done
-- Build files have been written to: <path_to>/build
-- west build: building application
[1/122] Preparing syscall dependency handling
...
[2/122] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (<path_to>/zephyr), build: v3.6.0-1133-g2f6b41a694ee
[121/122] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
             ROM:       76852 B         1 MB      7.33%
             RAM:         52 KB        60 KB     86.67%
        IDT_LIST:          0 GB         2 KB      0.00%
Generating files from <path_to>/build/zephyr/zephyr.elf for board: it8xxx2_evb

`<path_to>/handler.log`
```
***** delaying boot 1ms (per build configuration) *****
*** Booting Zephyr OS build v3.6.0-1133-g2f6b41a694ee (delayed boot 1ms) ***
Running TESTSUITE timer_jitter_drift
===================================================================
START - test_jitter_drift_timer_period
periodic timer behavior test using built-in restart mechanism
collecting time samples for approx 6 seconds
periodic timer samples gathered, calculating statistics
timer clock rate 32768, kernel tick rate 4096
expected period drift: 220.70312 us
period duration statistics for 5000 samples (0 rollovers):
  expected: 1000 us,            32.768000 cycles
  min:      1159.667969 us,     38 cycles
  max:      1251.220703 us,     41 cycles
  mean:     1220.690918 us,     39.999600 cycles
  variance: 621.750802 us,      0.667600 cycles
  stddev:   24.934931 us,       0.817068 cycles
timer start cycle 1206, end cycle 201204,
total time 6103454.589844 us, expected time 5000000.000000 us,
expected time drift 1103515.625000 us, difference -61.035156 us
[... 69 lines cut ...]
    Perfect delta 32 cycles or 977 us
Timer 0 max delta 43 cycles or 1312 us, 4 late callbacks (0.0%)
Timer 1 max delta 44 cycles or 1343 us, 10 late callbacks (0.0%)
Timer 2 max delta 45 cycles or 1373 us, 9 late callbacks (0.0%)
Timer 3 max delta 45 cycles or 1373 us, 6 late callbacks (0.0%)
!! Some ticks were missed.
!! Consider making CONFIG_SYS_CLOCK_TICKS_PER_SEC smaller.
 PASS - test_one_tick_timer_train in 10.236 seconds
===================================================================
TESTSUITE timer_tick_train succeeded

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  50.00% [timer_jitter_drift]: pass = 1, fail = 1, skip = 0, total = 2 duration = 13.840 seconds
 - FAIL - [timer_jitter_drift.test_jitter_drift_timer_period] duration = 6.906 seconds
 - PASS - [timer_jitter_drift.test_jitter_drift_timer_startdelay] duration = 6.934 seconds

SUITE PASS - 100.00% [timer_tick_train]: pass = 1, fail = 0, skip = 0, total = 1 duration = 10.236 seconds
 - PASS - [timer_tick_train.test_one_tick_timer_train] duration = 10.236 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION FAILED
```

Chenhongren, 2024-04-22:
Hi @gbarkadiusz 

Could I confirm one more thing?
Did you check the result after performing step#3 you listed?
  -->3. sudo ./ite -f <path_to>zephyrproject/zephyr/build/zephyr/zephyr.bin
Or, did you perform any reset between step#3 and testing?

teburd, 2024-04-22:
Hmmm

  stddev:   24.934931 us,       0.817068 cycles

If the stddev is less than a cycle I think you are doing ok, maybe this particular limit should be modified for this platform, or looked at with some more critical thinking.

There is a Kconfig for the test to allow for modifying the limits (noted in the assert!) and I'd recommend modifying that for this particular platform.

PR #71845 (2024-04-25): tests: timer_behavior: increase stdev tolerance for ite platform
There might be a deviation of 30.5 microseconds (1 cycle) due to the calculation deviation between free run and event timers. This commit increases stdev to 33 for ite platform.

Fix #67833
commit bfd24c4e (2024-04-25): tests: timer_behavior: increase stdev tolerance for ite platform

There might be a deviation of 30.5 microseconds (1 cycle) due to the
calculation deviation between free run and event timers. This commit
increases stdev to 33 for ite platform.

Fix #67833
closed 2024-04-25 by nashif