zephyrproject-rtos/zephyr #67836: Sam E70 Explained gPTP does not work

simontheu, 2024-01-19:
**Notes (delete this)**
Hi, this is a vanilla zephyr install from the getting started guide, with a vanilla gPTP project running on a Sam E70 Xplained board.

I have built the gPTP utility from the guide: https://docs.zephyrproject.org/latest/samples/net/gptp/README.html

The only change I have made is to add "CONFIG_NET_GPTP_LOG_LEVEL_DBG=y" into the prj.conf file.

**Describe the bug**
When trying to sync from the PC utility I get:
`~/gptp/build$ sudo ./gptp eno1 -F gptp_cfg.ini
INFO     : GPTP [11:30:23:798] gPTP starting
ERROR    : GPTP [11:30:23:798] Group ptp not found, will try root (0) instead
INFO     : GPTP [11:30:23:798] priority1 = 248
INFO     : GPTP [11:30:23:798] announceReceiptTimeout: 3
INFO     : GPTP [11:30:23:798] syncReceiptTimeout: 3
INFO     : GPTP [11:30:23:798] LINKSPEED_100MB - PHY delay
			TX: 1044 | RX: 2133
INFO     : GPTP [11:30:23:798] LINKSPEED_1G - PHY delay
			TX: 184 | RX: 382
INFO     : GPTP [11:30:23:798] neighborPropDelayThresh: 800
INFO     : GPTP [11:30:23:798] syncReceiptThreshold: 8
INFO     : GPTP [11:30:23:798] SyncFollowUp with negative correction field: forbidden
ERROR    : GPTP [11:30:23:799] Using clock device: /dev/ptp0
STATUS   : GPTP [11:30:23:799] Starting PDelay
STATUS   : GPTP [11:30:23:799] Link Speed: 100000 kb/sec
STATUS   : GPTP [11:30:26:799] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [11:30:26:799] New Grandmaster "E0:70:EA:FF:FE:CA:5B:6E" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [11:30:26:832] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:30:27:845] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:30:28:855] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:30:29:866] PDelay Response Receipt Timeout
EXCEPTION: GPTP [11:30:30:877] PDelay Response Receipt Timeout.........[repeats forever]`

And in the zephyr console as viewed from minicom, I get the following:
`:01:07.228,000] <inf> eth_sam: Link up
[00:01:07.329,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe12:11ba
[00:01:07.329,000] <inf> net_config: IPv6 address: fe80::fec2:3dff:fe12:11ba
[00:01:15.143,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 0 pkt 0x2040eed0
[00:01:15.143,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:15.143,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:16.155,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 1 pkt 0x2040eed0
[00:01:16.155,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:16.155,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:17.164,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 2 pkt 0x2040eed0
[00:01:17.164,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:17.164,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:18.174,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 3 pkt 0x2040eed0
[00:01:18.174,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:18.174,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:19.185,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 4 pkt 0x2040eed0
[00:01:19.185,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:19.185,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:20.198,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 5 pkt 0x2040eed0
[00:01:20.198,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:20.198,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:21.209,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 6 pkt 0x2040eed0
[00:01:21.209,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:21.209,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:22.216,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 7 pkt 0x2040eed0
[00:01:22.216,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:22.216,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:23.218,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 8 pkt 0x2040eed0
[00:01:23.218,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:23.218,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:24.229,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 9 pkt 0x2040eed0
[00:01:24.229,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:24.229,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:25.238,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 10 pkt 0x2040eed0
[00:01:25.238,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:25.238,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer
[00:01:26.252,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): Received PDELAY_REQ seq 11 pkt 0x2040eed0
[00:01:26.252,000] <dbg> net_gptp: gptp_handle_critical_msg: (rx_q[1]): No port found for gPTP buffer
[00:01:26.252,000] <dbg> net_gptp: gptp_handle_msg: (gptp): No port found for ptp buffer`

Please also mention any information which could help others to understand
the problem you're facing:
Sam E70 Xplained Board
I stripped back to this basic setup from a bigger project and have only change the debug level as described above.
I have not gone back to any previous commits yet, but happy to do so if someone can help point me to a known good one.

**To Reproduce**
Steps to reproduce the behavior:
Vanilla zephyr on current release.
`west build -p always -b sam_e70_xplained samples/net/gptp/
west flash`

**Expected behavior**
I expected to see on the linux utility as per the documentation for the gPTP project:
`STATUS   : GPTP [13:01:25:965] AsCapable: Enabled`

**Impact**
This project was for a PTP, so a bit of a showstopper for this one.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
Tried on Mac, and Windows

jukkar, 2024-01-19:
I could see the same issue with my Sam E70, the port was not found. This is not a gPTP issue but related to the board, I managed to flash the same gPTP sample to frdm_k64f board and it worked there just fine.
I recall that there are two variants of the sam_e70_xplained board, I have the first variant and the Ethernet driver seems to have been bitrotted. I do not have rev.B of the board so cannot test if gptp sample would work with it.

simontheu, 2024-01-19:
Tried also with sam_V71_xult board, exactly the same results

simontheu, 2024-01-19:
> I could see the same issue with my Sam E70, the port was not found. This is not a gPTP issue but related to the board, I managed to flash the same gPTP sample to frdm_k64f board and it worked there [...]

Ping and dumb_http_server seem to work ok on these boards, so core ethernet seems to work ok.

jukkar, 2024-01-19:
> Ping and dumb_http_server seem to work ok on these boards, so core ethernet seems to work ok.

Yes, DHCPv4 worked fine in my board. Perhaps the issue is related to ptp configuration in the driver.

nandojve, 2024-01-20:
Hi @jukkar , @simontheu ,

I don't have bandwidth to look on this. Besides, I never touched gPTP so my knowledge is zero.

@simontheu you can help to identify the commit that broke mainline using git bisect. With that maybe we can help to identify the issue.

talih0, 2024-01-21:
@simontheu, could you try this PR?

simontheu, 2024-01-21:
> @simontheu, could you try this PR? #67884
> 
> I tried to test on this board: https://docs.zephyrproject.org/latest/boards/arm/sam_v71_xult/doc/index.html, but for me the tx_completed interrupt is not firing for any transmitted packet.., so the code gets stuck in the eth_tx() function.

Can try this in about 12-15 hours.

simontheu, 2024-01-22:
I have tried this on 2 machines after a clean of the build folder, on both the e70 and v71 boards.

I am getting this warning in `west build ...`
`~/zephyrproject/zephyr/build/zephyr/zephyr.dts:466.27-479.5: Warning (unique_unit_address_if_enabled): /soc/ethernet@40050000: duplicate unit-address (also used in node /soc/mdio@40050000)`

Which could have something to do with the following:

I am getting this after a `west flash`
`Error: [atsamv71q21.cpu] clearing lockup after double fault
[atsamv71q21.cpu] halted due to debug-request, current mode: Handler HardFault
xPSR: 0xa1000003 pc: 0x0041fa0a msp: 0x2040aa98
shutdown command invoked`

simontheu, 2024-01-22:
Pulled again, and now it flashes and works for a while `AsCapable: Enabled` is shown.

It failed after some time with `Sync discontinuity` but this I think is now in my court.

simontheu, 2024-01-22:
I have intermittent building capability now, something is broken.

talih0, 2024-01-22:
> I have intermittent building capability now. And when I can flash I get the: `Error: [atsamv71q21.cpu] clearing lockup after double fault [atsamv71q21.cpu] halted due to debug-request, current mode: Handler HardFault xPSR: 0xa1000003 pc: 0x0041fa0a msp: 0x2040aa98 shutdown command invoked`
> 
> problem.

@simontheu, it's probably best to get a full log from west flash and to create a ticket. I saw this yesterday too, but cannot reproduce it now.

PR #67884 (2024-01-23): drivers: eth_sam_gmac: Use CONFIG_PTP_CLOCK_INIT_PRIORITY to init ptp
Use dedicated Kconfig symbol to set init priority.

Fixes #67836
closed 2024-01-23 by nashif