zephyrproject-rtos/zephyr #67868: Bluetooth: Controller: LL_ASSERT_OVERHEAD fails in lld_adv.c

mattrodgers31, 2024-01-20:
**Describe the bug**
I'm getting an assertion failure in `LL_ASSERT_OVERHEAD`, within the function `prepare_cb` in `lld_adv.c`. Specifically this always seems to happen when an advertising event occurs while an SPI transaction is in progress (updating a WS2812 LED strip). 

The problem is very reproducible, it always seems to happen within ~30 seconds of starting to update the LED strip. The application is doing an SPI transaction approximately every 500 ms.

I'm currently using the latest `main` branch of Zephyr (`429188723e4960415d4510bfeea3c0e2b749006b`) but can reproduce the problem with the `v3.5.0` and `v3.4.0` releases too.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
nrf52840 

 - What have you tried to diagnose or workaround this issue?
Tried releases `v3.5.0` and `v3.4.0`, same problem occurs. Issue does not happen when using nrf Connect SDK (I assume this uses the Nordic SoftDevice controller instead of the Zephyr controller?)

Tried increasing stack size (since call stack is quite long at the point of failure) but even 4kB does not resolve the issue.

 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
Have not gone further back than `v3.4.0` release so far to see if it still occurs.

**To Reproduce**
Steps to reproduce the behavior:
1. Perform SPI transaction with WS2812 LED strip approx every 500 ms
2. Start BLE advertising
3. Wait for assertion failure

**Expected behavior**
BLE advertising during an SPI transaction is possible without errors.

**Impact**
Zephyr controller not usable for this application. Can work around it using nrf Connect SDK instead, but I switched over to just using Zephyr since I want to use flash partitions from the device tree, and can't see any way to disable Nordic's partition manager (which ignores device tree partitions )when using nrf Connect SDK.

**Logs and console output**

Log of the problem occurring:
```
*** Booting Zephyr OS build zephyr-v3.5.0-4514-g429188723e49 ***
[00:00:01.752,593] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:01.752,624] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:01.752,655] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 3.5 Build 99
[00:00:01.753,295] <inf> bt_hci_core: Identity: DC:F8:D1:5F:AA:07 (random)
[00:00:01.753,326] <inf> bt_hci_core: HCI: version 5.4 (0x0d) revision 0x0000, manufacturer 0x05f1
[00:00:01.753,356] <inf> bt_hci_core: LMP: version 5.4 (0x0d) subver 0xffff
[00:00:18.781,433] <inf> bt_periph: Connected

[00:00:27.325,958] <dbg> light_pattern: light_pattern_execute: Setting light pattern of r 0 g 0 b 0
[00:00:31.275,207] <inf> bt_periph: Disconnected (reason 0x13)

[00:00:46.178,741] <dbg> light_pattern: light_pattern_execute: Setting light pattern of r 1 g 0 b 0
ASSERTION FAIL [0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:1044
        prepare_cb: Actual EVENT_OVERHEAD_START_US = 1281
[00:01:02.206,420] <err> os: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000002
[00:01:02.206,420] <err> os: r3/a4:  0x20000e30 r12/ip:  0x00000010 r14/lr:  0x00012199
[00:01:02.206,420] <err> os:  xpsr:  0x41000028
[00:01:02.206,451] <err> os: Faulting instruction address (r15/pc): 0x000121a4
[00:01:02.206,481] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:01:02.206,481] <err> os: Fault during interrupt handling

[00:01:02.206,512] <err> os: Current thread: 0x20000db0 (unknown)
[00:01:02.463,714] <err> os: Halting system
```

Call stack at the point of failure (it's always during the SPI transaction):
```
arch_system_halt@0x000323ae (zephyr/kernel/fatal.c:30)
k_sys_fatal_error_handler@0x0002c2ec (zephyr/kernel/fatal.c:44)
z_fatal_error@0x0002c3a0 (zephyr/kernel/fatal.c:118)
z_arm_fatal_error@0x00014774 (zephyr/arch/arm/core/fatal.c:76)
_oops@0x00014e62 (zephyr/arch/arm/core/cortex_m/swap_helper.S:448)
<signal handler called>@0xfffffff1 (Unknown Source:0)
prepare_cb@0x000121a4 (zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:1046)
lll_prepare_resolve@0x00026dec (zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:798)
lll_prepare@0x0001ebe4 (zephyr/subsys/bluetooth/controller/ll_sw/lll_common.c:65)
lll_adv_prepare@0x000275e4 (zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll_adv.c:689)
mayfly_run@0x0001c35e (zephyr/subsys/bluetooth/controller/util/mayfly.c:217)
_isr_wrapper@0x00015036 (zephyr/arch/arm/core/cortex_m/isr_wrapper.S:128)
<signal handler called>@0xfffffff1 (Unknown Source:0)
mayfly_pend@0x00028b30 (zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/mayfly.c:126)
mayfly_enqueue@0x0001c210 (zephyr/subsys/bluetooth/controller/util/mayfly.c:125)
ticker_cb@0x000205f6 (zephyr/subsys/bluetooth/controller/ll_sw/ull_adv.c:2370)
ticker_worker@0x0001c8c8 (zephyr/subsys/bluetooth/controller/ticker/ticker.c:1407)
mayfly_run@0x0001c35e (zephyr/subsys/bluetooth/controller/util/mayfly.c:217)
rtc0_nrf5_isr@0x00026470 (zephyr/subsys/bluetooth/controller/ll_sw/nordic/lll/lll.c:127)
_isr_wrapper@0x00015036 (zephyr/arch/arm/core/cortex_m/isr_wrapper.S:128)
<signal handler called>@0xfffffffd (Unknown Source:0)
spi_context_wait_for_completion@0x0002a5c4 (zephyr/drivers/spi/spi_context.h:160)
transceive@0x0002a5c4 (zephyr/drivers/spi/spi_nrfx_spi.c:256)
spi_nrfx_transceive@0x0002a5c4 (zephyr/drivers/spi/spi_nrfx_spi.c:292)
z_impl_spi_transceive@0x00031648 (zephyr/include/zephyr/drivers/spi.h:761)
spi_transceive@0x00031648 (app/build/zephyr/include/generated/syscalls/spi.h:38)
spi_write@0x00031648 (zephyr/include/zephyr/drivers/spi.h:850)
spi_write_dt@0x00031648 (zephyr/include/zephyr/drivers/spi.h:868)
ws2812_strip_update_rgb@0x00031648 (zephyr/drivers/led_strip/ws2812_spi.c:149)
led_strip_update_rgb@0x00012706 (zephyr/include/zephyr/drivers/led_strip.h:105)
light_partially_update_rgb@0x00012706 (app/src/light.c:154)
light_partially_update_rgb@0x000128fa (app/src/light.c:164)
light_pattern_execute@0x00012c32 (app/src/light_pattern.c:152)
light_pattern_thread_func@0x00012c32 (app/src/light_pattern.c:181)
z_thread_entry@0x0001364a (zephyr/lib/os/thread_entry.c:48)
??@0x765f89bc (Unknown Source:0)
```

**Environment (please complete the following information):**
 - OS: Built in docker, based off Ubuntu 22.04 (host OS is also ubuntu)
 - Toolchain: Zephyr SDK
 - Commit SHA or Version used: 429188723e4960415d4510bfeea3c0e2b749006b

**Additional context**
Can provide the entire application I'm using if that's useful

mattrodgers31, 2024-01-20:
Spotted this PR: https://github.com/zephyrproject-rtos/zephyr/pull/67725 which I thought might be relevant, but applying this doesn't seem to change the issue for me.

thedjnK, 2024-01-20:
Try setting `CONFIG_SOC_FLASH_NRF_TIMEOUT_MULTIPLIER=20`

mattrodgers31, 2024-01-20:
The issue still occurs with `CONFIG_SOC_FLASH_NRF_TIMEOUT_MULTIPLIER=20`.

Interestingly if I use Nordic's SPIM driver with DMA instead of the standard SPI driver (added `compatible = "nordic,nrf-spim";` to `spi0` in device tree overlay) then the problem doesn't occur any more. So at least I have a way to avoid the issue!

I think there must still be a problem somewhere - I wouldn't expect the SPI driver running in thread context to cause any timing issues for the bluetooth controller ISRs. But not sure whether the issue is SPI or Bluetooth-related now.

cvinayak, 2024-01-23:
@mattrodgers31 

Could you try use lower priorities for other SoC peripherals used in your application by having the below

`#define NRF_DEFAULT_IRQ_PRIORITY 5`

inside your application folder in `dts/arm/nordic/override.dtsi`  file.

MaureenHelm, 2024-01-23:
@cvinayak will you please assign a priority?

mattrodgers31, 2024-01-23:
@cvinayak using `#define NRF_DEFAULT_IRQ_PRIORITY 5` I don't see the issue any more.

cvinayak, 2024-01-23:
> @cvinayak using `#define NRF_DEFAULT_IRQ_PRIORITY 5` I don't see the issue any more.

Thank you for confirming, closing the issue. Feel free to reopen if you think otherwise.

closed 2024-01-23 by cvinayak

mattrodgers31, 2024-01-23:
For reference in case anyone runs into something similar in future, I had a bit more of a look into why the interrupts were able to block the BLE controller for so long (over 1 ms in my case). 

Each individual interrupt wasn't taking an excessive amount of time, but only 1 byte was being transferred per interrupt (I hadn't appreciated this initially, assumed there might be a FIFO so it could transfer more than 1 byte at a time). At an SPI frequency of 8 MHz this means the SPI ISR is called every 1 us, which is probably too frequent for the CPU to do anything else until the entire transfer is complete.

The `spim` driver with DMA works much better in this case.

cvinayak, 2024-01-24:
> For reference in case anyone runs into something similar in future, I had a bit more of a look into why the interrupts were able to block the BLE controller for so long (over 1 ms in my case).
>
> Each [...]

Bluetooth Controller uses SWI IRQ for bottom-half processing (scheduling etc.); by default all peripherals being configured to priority level 1 (at same level with SWI) but SPI IRQ having a natural higher priority due to its order in the peripheral instantiation in the SoC, it can starve the SWI IRQ from being serviced.