zephyrproject-rtos/zephyr #67878: compile native_sim error: cannot find -lgcc_s

git4Chub, 2024-01-21:
**Describe the bug**
recently, I want to try babblesim, however, I met some
compile error with board native_sim
cannot find -lgcc_s

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b native_sim ./samples/hello_world/ -p

**Expected behavior**
compile successfully to get zephyr.exe

**Logs and console output**
```
-- west build: making build dir /home/railphone/zephyrproject/upstream/zephyr/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /home/railphone/zephyrproject/upstream/zephyr/samples/hello_world
-- CMake version: 3.23.2
-- Found Python3: /home/railphone/miniconda3/envs/zephyr/bin/python (found suitable version "3.8.13", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/railphone/.cache/zephyr
-- Zephyr version: 3.5.99 (/home/railphone/zephyrproject/upstream/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: native_sim
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.16.3 (/home/railphone/zephyr-sdk-0.16.3)
-- Found toolchain: host (gcc/ld)
-- Found Dtc: /home/railphone/zephyr-sdk-0.16.3/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/railphone/zephyrproject/upstream/zephyr/boards/posix/native_sim/native_sim.dts
-- Generated zephyr.dts: /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/dts.cmake
Parsing /home/railphone/zephyrproject/upstream/zephyr/Kconfig
Loaded configuration '/home/railphone/zephyrproject/upstream/zephyr/boards/posix/native_sim/native_sim_defconfig'
Merged configuration '/home/railphone/zephyrproject/upstream/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/railphone/zephyrproject/upstream/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/railphone/zephyrproject/upstream/zephyr/build/zephyr/include/generated/autoconf.h'
-- Found GnuLd: /usr/bin/ld.bfd (found version "2.30") 
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Using ccache: /usr/bin/ccache
-- Configuring done
-- Generating done
-- Build files have been written to: /home/railphone/zephyrproject/upstream/zephyr/build
-- west build: building application
[1/81] Preparing syscall dependency handling

[2/81] Generating include/generated/version.h
-- Zephyr version: 3.5.99 (/home/railphone/zephyrproject/upstream/zephyr), build: 7fedc81666f3
[79/81] Linking C executable zephyr/zephyr.elf
FAILED: zephyr/zephyr.elf zephyr/zephyr_pre0.map zephyr/zephyr.map zephyr/zephyr.stat /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr_pre0.map /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.map /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.stat 
: && ccache /usr/bin/gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr.elf  zephyr/CMakeFiles/offsets.dir/./arch/posix/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/posix/core/libarch__posix__core.a  zephyr/soc/soc/posix/inf_clock/libsoc__posix__inf_clock.a  zephyr/boards/posix/native_sim/libboards__posix__native_sim.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/home/railphone/zephyrproject/upstream/zephyr/build/zephyr  -r  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -m32  -Wl,--unresolved-symbols=ignore-all && cd /home/railphone/zephyrproject/upstream/zephyr/build/zephyr && /usr/bin/cmake -E true && cd /home/railphone/zephyrproject/upstream/zephyr/build/zephyr && /usr/bin/cmake -E echo Generating\ files\ from\ /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.elf\ for\ board:\ native_sim && /usr/bin/cmake -E copy zephyr_pre0.map zephyr.map && /usr/bin/readelf -e zephyr.elf > zephyr.stat
/usr/bin/ld.bfd: cannot find -lgcc_s
/usr/bin/ld.bfd: cannot find -lgcc_s
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

**Environment (please complete the following information):**
 - OS: ubuntu1~18.04
 - Toolchain ：zephyr 0.16.3
 - Commit ：7fedc81666f34d5948a05adcacc6f644fb85332f
 
**Additional context**
# infomation
## libgcc_s lib I get in my host
```
libgcc_s.so.1 (libc6,x32) => /usr/libx32/libgcc_s.so.1
libgcc_s.so.1 (libc6,x86-64) => /lib/x86_64-linux-gnu/libgcc_s.so.1
libgcc_s.so.1 (libc6) => /lib/i386-linux-gnu/libgcc_s.so.1
libgcc_s.so.1 (libc6) => /usr/lib32/libgcc_s.so.1
```

# analysis
## related to static library link
I found a similar problem(probably same reason) below
> https://stackoverflow.com/questions/51998912/when-statically-linking-a-library-getting-linker-error-cannot-find-lgcc-s
I suspect libgcc_s is considered as  a static library, however there is only libgcc_s in shared library format.
Additionally, I mannually touch an empty file called libgcc_s.a in /usr/lib/, the linker can find it eventually
```
[79/81] Linking C executable zephyr/zephyr.elf
FAILED: zephyr/zephyr.elf zephyr/zephyr_pre0.map zephyr/zephyr.map zephyr/zephyr.stat /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr_pre0.map /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.map /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.stat 
: && ccache /usr/bin/gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr.elf  zephyr/CMakeFiles/offsets.dir/./arch/posix/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/arch/posix/core/libarch__posix__core.a  zephyr/soc/soc/posix/inf_clock/libsoc__posix__inf_clock.a  zephyr/boards/posix/native_sim/libboards__posix__native_sim.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L/home/railphone/zephyrproject/upstream/zephyr/build/zephyr  -r  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -m32  -Wl,--unresolved-symbols=ignore-all && cd /home/railphone/zephyrproject/upstream/zephyr/build/zephyr && /usr/bin/cmake -E true && cd /home/railphone/zephyrproject/upstream/zephyr/build/zephyr && /usr/bin/cmake -E echo Generating\ files\ from\ /home/railphone/zephyrproject/upstream/zephyr/build/zephyr/zephyr.elf\ for\ board:\ native_sim && /usr/bin/cmake -E copy zephyr_pre0.map zephyr.map && /usr/bin/readelf -e zephyr.elf > zephyr.stat
/usr/lib/gcc/x86_64-linux-gnu/7/../../../libgcc_s.a: file not recognized: File truncated
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

git4Chub, 2024-01-23:
> @git4Chub From your report it seems you are building on the 18.04 Ubuntu release. That is almost 6 years old now, and out of standard support by Ubuntu. You are also building with gcc 7.5 which is quite old. There has been 2 new Ubuntu LTS releases since then. I would encourage you to upgrade to 22.04 or beyond.
> 
> It is difficult for me to guess what is broken there, but I don't think anything in the Zephyr build will add the `-lgcc_s` option, and it does not seem to be in that gcc invocation that fails. So my immediate guess is that your host toolchain installation is broken to some degree.
> 
> So at this point, I don't think it is a good use of resources for the project to dig into this issue, unless others are finding this issue in newer Linux releases.
> 
> Prioritizing it as low, as so far it has only been reported in a too old Linux release.

thanks for detailed reply @aescolar.  I'll try newer ubuntu LTS later and check if my host toolchain .

aescolar, 2024-03-15:
Closing this issue as it seems like a issue with the user environment. If more users report it we can reopen it.

closed 2024-03-15 by aescolar